bperez77/xilinx_axidma #4: Move Contiguous Memory Region Specification to the Device Tree

bperez77, 2016-07-21:
To allocate large regions of physically-contiguous memory, the driver uses the CMA (contiguous memory allocator). At boot time, the user specifies the `cma=<size>M` parameter to reserve a region of `size` megabytes for the CMA. 

It would be much nicer if the user could instead specify this in the device tree, as this requires less manual user intervention, since they don't need to update the command line via U-Boot or the device tree.

This can be done via a reserved memory region device tree node. The user specifies a device tree node that describes the contiguous memory region (size, alignment, type, etc.). Then the user would put the **memory-region** property in the driver's device tree node. This is simply a phandle to the reserved memory region device tree node. See the [reserved-memory](https://github.com/Xilinx/linux-xlnx/blob/master/Documentation/devicetree/bindings/reserved-memory/reserved-memory.txt) documentation for more details.

jannikmahr, 2018-12-07:
This DMA Buffer driver already has support for "reserved-memory":
[https://github.com/ikwzm/udmabuf#memory-region](https://github.com/ikwzm/udmabuf#memory-region)

You probably already know this project, but if not it might contain some useful code.

maikonadams, 2019-02-04:
I am using petalinux, programming over JTAG right now, and I do not know where to set to configure CMA, when I run the benchmark I get :

cma: cma_alloc: alloc failed, req-size: 2025 pages, ret: -12
axidma: axidma_chrdev.c: axidma_mmap: 285: Unable to allocate contiguous DMA memory region of size 8294400.
axidma: axidma_chrdev.c: axidma_mmap: 287: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.

I tried to set it in the system-user.dtsi 
chosen {
		bootargs = "console=ttyPS0,115200 CMA=28MB earlyprintk";
		stdout-path = "serial0:115200n8";
	};

guerricmeurice, 2019-02-04:
Hello @maikonadams,

For CMA configuration using petalinux (tested with 2017.4):

Go to kernel features
Make sure the following settings are met:
o Contiguous Memory Allocator
 True
o CMA debug messages (DEVELOPMENT) (if you want...)
 True

Then enable and set the CMA value:

Go to Device Drivers / Generic Driver options
Make sure the following settings are met:
o DMA Contiguous Memory Allocator
 True
o Size in Mega Bytes
 25 (for example)

keegandent, 2026-01-28:
This issue appears closed