bperez77/xilinx_axidma #13: Support of SG mode

GOOD-Stuff, 2017-02-06:
Do I understand correctly that this the character driver and the library does not support scatter gather mode?

bperez77, 2017-02-07:
It actually does support scatter-gather. Though from the driver/library's point of view, it is agnostic as to whether or not scatter-gather is being used.

Is that unclear in the README? Perhaps I should update it; I guess the support of CMA could be seen as a solution to not supporting scatter-gather.

GOOD-Stuff, 2017-02-07:
Thank you for your reply. 

> the support of CMA could be seen as a solution to not supporting scatter-gather.
I thought so, when I found mention about CMA, but mention about scatter gather in code misled me. 

P.S. 
Sorry for my english, it's not my native language.

GOOD-Stuff, 2017-02-07:
> it is agnostic as to whether or not scatter-gather is being used.

But what do you mean by an agnostic? The driver does not matter work with SG or not, is it?

bperez77, 2017-02-07:
Yep, that's what I mean. From the driver's point of view, it doesn't matter whether or not you're using SG.

In Linux, there's a whole DMA interface that abstracts away the fine details of setting up DMA transfers. The backend Xilinx driver handles that.

closed 2017-02-09 by GOOD-Stuff

yaobaishen, 2017-04-09:
Interesting that I have a try to enable the scatter-gather mode in AXI DMA IP settings, the all the examples application would lead to "Segmental Fault". I have enabled CMA in .config, and set 'cma=25M ' in bootargs. Does anybody have tried this with SG enabled?

bperez77, 2017-04-09:
I've used in SG mode before, it should work. I'd need to see your `dmesg` output after the segfault to help you more.

yaobaishen, 2017-04-19:
Clarify: the driver and library could work with SG mode, verified with kernel 4.0.