bperez77/xilinx_axidma #15: only one (tx) vdma channel recognized

jamsoftJirka, 2017-02-27:
Hello,

 

please, help to solve the problem...

I have a Zynq board (7020) running Linux (Linaro 4.4).  

My design is using Xilinx VDMA IP Core configured in a loopback mode.

 If I try insmod axidma.ko driver (and dmesg | grep dma)  then only one VDMA channel (tx) is recognized. The driver reads the info about channels from the device-tree but it does not see the rx channel.  Instead it finds the not configured (and non-existing) DMA receive channel...

Here are my VDMA and driver nodes in devicetree.dts:

 axi_vdma_0: axivdma@40030000 {
    compatible = "xlnx,axi-vdma", "xlnx,axi-vdma-6.2", "xlnx,axi-vdma-1.00.a";
    #dma-cells = <1>;
    reg = < 0x40030000 0x10000 >;
    dma-ranges = <0x00000000 0x00000000 0x40000000>;
    xlnx,num-fstores = <0x8>;
    xlnx,flush-fsync = <0x1>;
    xlnx,addrwidth = <0x20>;
    clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
    clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk",
              "m_axis_mm2s_aclk", "s_axis_s2mm_aclk";
    dma-channel@40030000 {
        compatible = "xlnx,axi-vdma-mm2s-channel";
        interrupts = < 0 54 4 >;
        xlnx,datawidth = <0x40>;
    } ;
    dma-channel@40030030 {
        compatible = "xlnx,axi-vdma-s2mm-channel";
        interrupts = < 0 53 4 >;
        xlnx,datawidth = <0x40>;
    } ;
} ;
        
    axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};

 
The messages from the axidma driver after insmod:

axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 0 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 1 transmit channels and 0 receive channels.

I have no entry for AXI DMA in the device tree nor there is no AXI DMA in the design. Only AXI VDMA is present in the design and devicetree...

The Linux (Xilinx) VDMA driver says:

xilinx-vdma 43000000.axivdma: Xilinx AXI VDMA Engine Driver Probed!!

Am I missing something or there is a problem with the driver itself, please?

Thank you for your help!

bperez77, 2017-02-27:
Interesting, I haven't seen this issue before. I'll need to run through this myself before I can say for sure what the issue is.

In the meantime, can you attach the full output from `dmesg`?

jamsoftJirka, 2017-02-28:
Hello,

thank you for your interest. Attached are 2 files - outputs from dmesg. The first one is without the axidma.ko driver loaded and the other one is after insmod axidma.ko. 

[dmesg1.txt](https://github.com/bperez77/xilinx_axidma/files/809203/dmesg1.txt)
[dmesg2.txt](https://github.com/bperez77/xilinx_axidma/files/809204/dmesg2.txt)

Also I checked the /proc/device-tree entries - they looks like OK. The VDMA branch contains two channels...
Thank you once more for any help...

bperez77, 2017-03-09:
Sorry, I've been busy for the past week, so I haven't had much time to look at this. Are you on the most recent commit for this repository? I noticed that your device tree is missing the `xlnx,device-id` property for the VDMA channels. The driver requires this property to be in each channel, so it should be throwing an error when you try to load it into the kernel.

jamsoftJirka, 2017-03-15:
Hello, sorry for the late answer - I am very busy too. I am also sorry, I've copied a bad version of the device-tree VDMA entry in my first post.

The current one I am using is the following:

	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_vdma_0: dma@43000000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-vdma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 31 4 0 32 4>;
			reg = <0x43000000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,flush-fsync = <0x1>;
			xlnx,num-fstores = <0x1>;
			dma-channel@43000000 {
				compatible = "xlnx,axi-vdma-mm2s-channel";
				interrupts = <0 31 4>;
				xlnx,datawidth = <0x10>;
				xlnx,device-id = <0x0>;
				xlnx,genlock-mode ;
			};
			dma-channel@43000030 {
				compatible = "xlnx,axi-vdma-s2mm-channel";
				interrupts = <0 32 4>;
				xlnx,datawidth = <0x10>;
				xlnx,device-id = <0x1>;
				xlnx,genlock-mode ;
			};
		};
	};

Anyway, the result is the same. Only one channel is found. And yes, I am using the most recent version of your driver.
Thank you for your effort to solve this issue... I will wait and try your following suggestions.

rogumag, 2017-03-17:
hi bperez77, i´m using your driver with a vdma IP in a loopback mode. i´m using your driver and i have the same issue of jamsoftJirka. my device-tree is very similar. I pass the console output when i execute `dmesg | grep dma.`

> [0.286072] xilinx-vdma 43000000.dma: Xilinx AXI VDMA Engine Driver Probed!!
[   52.733412] axidma: loading out-of-tree module taints kernel.
[   52.734515] axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 0 transmit channels and 1 receive channels.
[   52.734526] axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 1 transmit channels and 0 receive channels.

Thank you for your time and efford.

jamsoftJirka, 2017-03-19:
Hello,

I've built another linux kernel for my board - the Petalinux 2016.3 and SDK, 4.6.0-xilinx (instead of Linaro) and the result is the same. Only one channel (transmit) was found... 

Do you have any suggestion, please? Thank you. Jirka

bperez77, 2017-03-20:
@jamsoftJirka, @rogumag  I'm looking at this right now. I'm working off the branch `xilinx-v2016.3` branch of the main Xilinx kernel (https://github.com/Xilinx/linux-xlnx). In any case, this is likely a problem across branches, but let me know what's the specific branch you're both on.

bperez77, 2017-03-21:
Wow this was quite a dumb error on my part, I was simply incrementing the wrong value. This should be fixed by 46c24a5b10e29e599998ba6c19f5c1ad6e7bf848. Try pulling, recompling your code, and let me know if it works.

rogumag, 2017-03-21:
Hi brandon, i have proved your fixed code and works very well. i have one transmit channel and one receive chanel. Thank you for tour time and efford.
Best Regards

jamsoftJirka, 2017-03-21:
Hello Brandon, thank you for your effort and the problem solution! I will give it a try this afternoon...

Jirka

jamsoftJirka, 2017-03-21:
So, as Rogumag, I have 1 tx and 1 transmit after changing the code :) Great work! Also I have the same question - the example codes work with the dma only? If I do not make mistake, there is only axidma_init function in the library and axivdma_init  is missing. What to change to init struct for dma_init function with vdma channels, please?  Thank you, Jirka

bperez77, 2017-03-21:
At the moment, all of the code for VDMA is there in the driver, but I never did get the chance to integrate it into the userspace library. I should be able to add it into the library pretty quickly, but you also have the option of using the driver's IOCTL inteface directly.

bperez77, 2017-03-22:
There's a couple changes I need to make to the driver, but I think I should be able to get VDMA transactions up and running. Most of the code is already there, there's just no code path to it currently, since I didn't need it when I frist wrote the driver.

jamsoftJirka, 2017-03-22:
Thank you, Brandon. It could be great if you can modify the code to make VDMA transactions up and running. For me personaly it could be a great help. Thank you and have a nice day!

bperez77, 2017-03-25:
I've added support for VDMA to driver with commit 4642323a81490b9ba7f416fdc33cf9119cf82d12. 

Unfortunately, I haven't been able to get even the VDMA properly working in a loopback configuration. I've even tried running their AXI VDMA test driver ([drivers/dma/xilinx/vdmatest.c](https://github.com/Xilinx/linux-xlnx/blob/xilinx-v2016.3/drivers/dma/xilinx/vdmatest.c)), and that didn't work for me either. 

So, I think either the issue is that my device tree is incorrect, or the Xilinx DMA driver doesn't support frame counter interrupts. I believe is the latter, because, when I look over their driver ([drivers/dma/xilinx/xilinx_dma.c](https://github.com/Xilinx/linux-xlnx/blob/xilinx-v2016.3/drivers/dma/xilinx/xilinx_dma.c)), I don't see them enabling frame count interrupts anywhere. Without this, the only time the VDMA IP generates an interrupt is when an error occurs. Here's the device tree I'm using:
```
&amba {
    axi_vdma_0: vdma@43000000 {
        #dma-cells = <1>;
        compatible = "xlnx,axi-vdma-1.00.a";
        reg = <0x43000000 0x10000>;
        dma-ranges = <0x00000000 0x00000000 0x40000000>;
        clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_s2mm_aclk";
        clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;

        xlnx,addrwidth = <32>;
        xlnx,num-fstores = <1>;
        xlnx,include-sg;

        vdma-channel@43000000 {
            compatible = "xlnx,axi-vdma-mm2s-channel";
            interrupt-parent = <&intc>;
            interrupts = <0 29 4>;
            xlnx,datawidth = <32>;
            xlnx,device-id = <0x0>;
            xlnx,include-dre;
        };

        vdma-channel@43000030 {
            compatible = "xlnx,axi-vdma-s2mm-channel";
            interrupt-parent = <&intc>;
            interrupts = <0 30 4>;
            xlnx,datawidth = <32>;
            xlnx,device-id = <0x1>;
            xlnx,include-dre;
        };
    };
};
```

closed 2017-03-27 by bperez77