bperez77/xilinx_axidma #16: axidma_chrdev.c: axidma_mmap: 276: Unable to allocate contiguous DMA memory region of size 41472000.

raffaelesury, 2017-03-09:
Hi Brandon, 
I'm trying to use your driver on a zedboard but the error in the title comes up, I tried increasing to cma=45M in the hope that it would fix the issue but it doesn't. 

After booting, in the **dmesg** I can see the following lines relevant to the issue: 
**[    0.000000] cma: Reserved 48 MiB at 0x1b800000**
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c0938ec0, node_mem_map dfbe0000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 12 pages/cpu @dfbbd000 s19776 r8192 d21184 u49152
[    0.000000] pcpu-alloc: s19776 r8192 d21184 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] **Kernel command line**: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 **cma=45M** rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] **Memory**: 461696K/524288K available (5286K kernel code, 231K rwdata, 1848K rodata, 1024K init, 222K bss, 13440K reserved, **49152K cma-reserved**, 0K highmem)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc07f78d8   (8127 kB)
      .init : 0xc0800000 - 0xc0900000   (1024 kB)
      .data : 0xc0900000 - 0xc0939c20   ( 232 kB)
       .bss : 0xc0939c20 - 0xc0971624   ( 223 kB)

U-boot reads the uEnv.txt file from the BOOT partition and uses those bootargs instead of the ones in the device tree, here's my uEnv.txt file: 

```
uenvcmd=run adi_sdboot
adi_sdboot=echo Copying Linux from SD to RAM... && fatload mmc 0 ${kernel_load_address} ${kernel_image} && fatload mmc 0 ${devicetree_load_address} ${devicetree_image} && if fatload mmc 0 ${ramdisk_load_address} ${ramdisk_image}; then bootm ${kernel_load_address} ${ramdisk_load_address} ${devicetree_load_address}; else bootm ${kernel_load_address} - ${devicetree_load_address}; fi
bootargs=console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 cma=45M rootwait
```

This is working properly as shown by the dmesg log. 

I'm using u-boot and linux cloned from the xilinx repositories, I checked out the tag xilinx-v2016.3 and made a branch with those before compiling them. 
I'm using the rootfs I got from linaro-vivid-developer-20151215-714.tar.gz from the linaro website. 
 Linux version 4.6.0-xilinx (root@osboxes) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) 

My device tree is the one generated by hsi 2016.3 which I then edited to give a unique names to the channels and add the node for your driver.
Here is the relevant .dtsi file 

```
/ {

	axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
	    dma-names = "tx_channel", "rx_channel";
	};

	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>; //NB 20h = 32b
			dma-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>; 
				xlnx,device-id = <0x0>;
			};
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>; 
				xlnx,device-id = <0x1>;
			};
		};
	};
};

```
Thanks for any help you can offer.

bperez77, 2017-03-09:
Thanks for starting with all your info, that's helpful. I haven't seen this error before, but I've never tried to allocate a region larger than 8 MB personally. Are there any other relevant messages in the kernel log buffer near where you see the error from my driver?

The region you're trying to allocate is pretty large (39.5 MB). Do you need this as one giant bufffer? In any case, do you get this same error if you try a smaller allocation size, say around 8 MB or so?

raffaelesury, 2017-03-09:
>Are there any other relevant messages in the kernel log buffer near where you see the error from my driver?
I don't think so but I'm going to post the messages around the error as soon as I get my hands on the zedboard. 

I forgot to mention: 
I set up an AXI DMA core with default settings with the stream channels in a loop, in the same fashion as the outdated xilinx example that is floating around everywhere on the internet, and I wanted to run your benchmark example application, so I haven't identified any needs at this point except that I have data appearing in the PL (less than 8 MB for sure) and I need it available for a userspace application. 

I started with CMA=25M because that's what the readme said, and then switched to 45M because the error message said the module was trying to allocate a region of size 41472000 (no idea why the driver would want this).

I've been struggling to get the DMA working in linux for weeks in various fashions, the xilinx dma engine looks daunting and I'm a newbie to the kernel so I've been digging for drivers and yours looks very promising since it's recent, documented and doesn't seem to overcomplicate stuff.

I'm going to test CMA=8M tomorrow and post the results, thanks for helping.

bperez77, 2017-03-09:
I see, which C application are you using. Are you using your own or are you using `examples/axidma_benchmark.c`? If you're using a custom one, post it here if you can.

The CMA parameter just controls how big the region of memory is for the contiguous memory allocator to use, so that doesn't directly affect the memory you're trying to allcoate, it just takes it from that region. The way memory allocation works is that an application calls libaxidma's `axidma_malloc` with a specified size, then this invokes the driver's `mmap` method, which in turn goes and gets the memory region from the CMA.

I feel your pain. Unfortunately, Xilinx is pretty terrible when it comes to documentation in general, but the whole DMA system is particularily bad. I ended up having to write this driver so I could actually write user-level applications that could use the DMA engine.

Keep the CMA=25M, that should work fine. I'm still not sure what's causing the issue, but it looks like a bad size parameter is being passed to the `mmap` method.

raffaelesury, 2017-03-10:
I'm running axidma_benchmark. 

Here's the full transcript with CMA=25M

```
root@linaro-developer:/opt/outputs# ls
axidma.ko  axidma_benchmark  axidma_display_image  axidma_transfer  libaxidma.so
root@linaro-developer:/opt/outputs# insmod axidma.ko
root@linaro-developer:/opt/outputs# ./axidma_benchmark 
AXI DMA Benchmark Parameters:
	Transmit Buffer Size: 7.91 Mb
	Receive Buffer Size: 7.91 Mb
	Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Unable to start remainder transactions.
Segmentation fault
root@linaro-developer:/opt/outputs# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.6.0-xilinx (root@osboxes) (gcc version 5.2.1 20151005 (Linaro GCC 5.2-2015.11-2) ) #1 SMP PREEMPT Thu Mar 9 12:31:01 CET 2017
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: esempiolauriblog
[    0.000000] cma: Reserved 28 MiB at 0x1cc00000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c0938ec0, node_mem_map dfbea000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 12 pages/cpu @dfbc7000 s19776 r8192 d21184 u49152
[    0.000000] pcpu-alloc: s19776 r8192 d21184 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 cma=25M rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 482216K/524288K available (5286K kernel code, 231K rwdata, 1848K rodata, 1024K init, 222K bss, 13400K reserved, 28672K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc07f78d8   (8127 kB)
      .init : 0xc0800000 - 0xc0900000   (1024 kB)
      .data : 0xc0900000 - 0xc0939c20   ( 232 kB)
       .bss : 0xc0939c20 - 0xc0971624   ( 223 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] efuse mapped to e0800000
[    0.000000] slcr mapped to e0802000
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] zynq_clock_init: clkc starts at e0802100
[    0.000000] Zynq clock init
[    0.000010] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.000029] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[... 99 lines cut ...]
[    1.051564] can: raw protocol (rev 20120528)
[    1.054563] can: broadcast manager protocol (rev 20120528 t)
[    1.058874] can: netlink gateway (rev 20130117) max_hops=1
[    1.063403] Registering SWP/SWPB emulation handler
[    1.078132] mmc0: new high speed SDHC card at address aaaa
[    1.078251] hctosys: unable to open rtc device (rtc0)
[    1.078549] ALSA device list:
[    1.078552]   No soundcards found.
[    1.090219] mmcblk0: mmc0:aaaa SU08G 7.40 GiB 
[    1.094783]  mmcblk0: p1 p2
[    1.125257] EXT4-fs (mmcblk0p2): recovery complete
[    1.128668] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    1.135488] VFS: Mounted root (ext4 filesystem) on device 179:2.
[    1.148455] devtmpfs: mounted
[    1.151410] Freeing unused kernel memory: 1024K (c0800000 - c0900000)
[    1.390173] systemd[1]: Failed to insert module 'autofs4'
[    1.424149] random: systemd urandom read with 6 bits of entropy available
[    2.599743] systemd-journald[847]: Received request to flush runtime journal from PID 1
[    8.882442] macb e000b000.ethernet eth0: link up (1000/Full)
[   64.532404] axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 1 transmit channels and 1 receive channels.
[   64.532423] axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 0 transmit channels and 0 receive channels.
[   87.714006] axidma: axidma_chrdev.c: axidma_mmap: 276: Unable to allocate contiguous DMA memory region of size 41472000.
[   87.723509] axidma: axidma_chrdev.c: axidma_mmap: 278: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.

```

bperez77, 2017-03-10:
Unforutnately, I won't have access to a Zedboard for a few more days, so I can't look at this directly. 

However, I might still be able to debug this. Can you first pull the newest code (I made a minor fix to a return value check)? Then, apply the patch below to your code, recompile everything (driver, library, and example) from scratch, run it again, and post the same output as you did above.

```
diff --git a/driver/axidma_chrdev.c b/driver/axidma_chrdev.c
index 413242f..7756c3b 100644
--- a/driver/axidma_chrdev.c
+++ b/driver/axidma_chrdev.c
@@ -264,6 +264,8 @@ static int axidma_mmap(struct file *file, struct vm_area_struct *vma)
     }
 
     // Set the user virtual address and the size
+    axidma_err("vma->vm_end=%08lx, vma->vm_start=0x%08lx, dma_alloc->size=%zu\n",
+            vma->vm_end, vma->vm_start, dma_alloc->size);
     dma_alloc->size = vma->vm_end - vma->vm_start;
     dma_alloc->user_addr = (void *)vma->vm_start;
 
diff --git a/examples/axidma_benchmark.c b/examples/axidma_benchmark.c
index 788f90f..70f939b 100644
--- a/examples/axidma_benchmark.c
+++ b/examples/axidma_benchmark.c
@@ -454,12 +454,14 @@ int main(int argc, char **argv)
     }
 
     // Map memory regions for the transmit and receive buffers
+    printf("axidma_benchmark: tx_size=%zu\n", tx_size);
     tx_buf = axidma_malloc(axidma_dev, tx_size);
     if (tx_buf == NULL) {
         perror("Unable to allocate transmit buffer from the AXI DMA device.");
         rc = -1;
         goto destroy_axidma;
     }
+    printf("axidma_benchmark: tx_size=%zu\n", rx_size);
     rx_buf = axidma_malloc(axidma_dev, rx_size);
     if (rx_buf == NULL) {
         perror("Unable to allocate receive buffer from the AXI DMA device");
diff --git a/library/libaxidma.c b/library/libaxidma.c
index cbecb3c..175a714 100644
--- a/library/libaxidma.c
+++ b/library/libaxidma.c
@@ -142,6 +142,7 @@ void *axidma_malloc(axidma_dev_t dev, size_t size)
     void *addr;
 
     // Call the device's mmap method to allocate the memory region
+    printf("libaxidma: mmap size=%zu\n", size);
     addr = mmap(NULL, size, PROT_READ|PROT_WRITE, MAP_SHARED, dev->fd, 0);
     if (addr == MAP_FAILED) {
         return NULL;
```

raffaelesury, 2017-03-13:
Hi, I did as instructed, here's the console output

```
root@linaro-developer:/opt/outputs# ./axidma_benchmark 
AXI DMA Benchmark Parameters:
	Transmit Buffer Size: 7.91 Mb
	Receive Buffer Size: 7.91 Mb
	Number of DMA Transfers: 1000 transfers

axidma_benchmark: tx_size=8294400
libaxidma: mmap size=8294400
axidma_benchmark: tx_size=8294400
libaxidma: mmap size=8294400
Using transmit channel 0 and receive channel 1.
libaxidma: mmap size=41472000
Unable to start remainder transactions.
Segmentation fault
root@linaro-developer:/opt/outputs# dmesg
```

and here's the last part of dmesg 
```
[    1.099389] Freeing unused kernel memory: 1024K (c0800000 - c0900000)
[    1.345395] systemd[1]: Failed to insert module 'autofs4'
[    1.379183] random: systemd urandom read with 6 bits of entropy available
[    2.649443] systemd-journald[1186]: Received request to flush runtime journal from PID 1
[    8.840976] macb e000b000.ethernet eth0: link up (1000/Full)
[   68.350973] axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 1 transmit channels and 1 receive channels.
[   68.350990] axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 0 transmit channels and 0 receive channels.
[   92.661015] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb6e58000, vma->vm_start=0xb666f000, dma_alloc->size=116692
[   92.691396] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb666f000, vma->vm_start=0xb5e86000, dma_alloc->size=511092934
[   92.740630] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb5e86000, vma->vm_start=0xb36f9000, dma_alloc->size=0
[   92.750312] axidma: axidma_chrdev.c: axidma_mmap: 278: Unable to allocate contiguous DMA memory region of size 41472000.
[   92.760288] axidma: axidma_chrdev.c: axidma_mmap: 280: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
```

At boot it said "cma: Reserved 28 MiB at 0x1cc00000"

raffaelesury, 2017-03-13:
I noticed that the patched code calls 
```
+    axidma_err("vma->vm_end=0x%08lx, vma->vm_start=0x%08lx, dma_alloc->size=%zu\n",
+            vma->vm_end, vma->vm_start, dma_alloc->size)
```
right before the value is changed by
`dma_alloc->size = vma->vm_end - vma->vm_start;`

so I added another axidma_err print     
```
axidma_err("Next I call dma_alloc_coherent with dma_alloc->size=%zu\n",
              dma_alloc->size);
```
right before
```
    dma_alloc->kern_addr = dma_alloc_coherent(NULL, dma_alloc->size,
                                              &dma_alloc->dma_addr, GFP_KERNEL);
```

the results are
```
[  200.035990] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb6ea6000, vma->vm_start=0xb66bd000, dma_alloc->size=116692
[  200.046148] axidma: axidma_chrdev.c: axidma_mmap: 275: Next I call dma_alloc_coherent with dma_alloc->size=8294400
[  200.075480] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb66bd000, vma->vm_start=0xb5ed4000, dma_alloc->size=2512895394
[  200.085943] axidma: axidma_chrdev.c: axidma_mmap: 275: Next I call dma_alloc_coherent with dma_alloc->size=8294400
[  200.133917] axidma: axidma_chrdev.c: axidma_mmap: 268: vma->vm_end=0xb5ed4000, vma->vm_start=0xb3747000, dma_alloc->size=511092934
[  200.144285] axidma: axidma_chrdev.c: axidma_mmap: 275: Next I call dma_alloc_coherent with dma_alloc->size=41472000
[  200.153820] axidma: axidma_chrdev.c: axidma_mmap: 280: Unable to allocate contiguous DMA memory region of size 41472000.
[  200.163519] axidma: axidma_chrdev.c: axidma_mmap: 282: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.

```

bperez77, 2017-03-20:
Sorry about the missed print in the patch, I wasn't able to run the code on my end.

Sorry about the trouble with regards to this to, because this was because of a change I made to application that I forgot about it. Basically, I tried to extend the AXI DMA benchmark application so it could handle arbitrary logic on the PL. The application is trying to handle the case when you have more than two DMA channels that your PL hardware uses. So it's trying to start "remainder" transactions on these DMA channels, so that the hardware on the PL fabric doesn't hang.

Thus, it needs to allocate a 3rd DMA buffer for these transactions, and it allocates one that is 5 times larger (hard-coded), which is why you see that third allocation. I've decided to remove this code because it leads to confusion (as demonstrated here), and it doesn't really address the problem well anyway. I've pushed a new version of the applications with commit 64a7680. Try pulling, recompiling, and running, that should do the trick.

raffaelesury, 2017-03-21:
Ah so that's what that stuff was about.  

The new commit works fine and the axidma_transfer app is working as intended so I'm closing the issue, thanks for your help. 

FYI there is now this dmesg message in red every time a transfer is done
`xilinx-vdma 40400000.dma: Channel df5bbf10 has errors 10, cdr 0 tdr 0`

closed 2017-03-21 by raffaelesury

bperez77, 2017-03-22:
That's a `DMAIntErr` (internal error, see `DMASR` register description). I can't tell if the S2MM or MM2S based on that message. This error only occurs for MM2S if the scatter-gather descriptor length is 0. On the S2MM side, this error also occurs in the case of an early stop, where the amount of received data is less than was specified.

Are you still using the AXI DMA in the loopback configuration here? If not, then that may be the source of the error (which can be safely ignored if it's an early stop).

raffaelesury, 2017-03-22:
thanks for the tip! I couldn't find anything in google linking that error to the IP core registers. 
I'm using the core without scatter gather and indeed playing with custom AXI-Stream IP now, adding what you told me I reached a diagnosis of exclusion: 
S2MM `DMAIntErr`
```
When Scatter Gather
is disabled, this error is flagged if any error occurs during
Memory write or if the incoming packet is bigger than what is
specified in the DMA length register
```

the second issue is indeed what could have happened since I'm trying random stuff. 

I'm going to keep this in mind as I work on my project.

bperez77, 2017-03-22:
Yeah that's probably it then. The DMASR register description is in the AXI DMA IP manual. You can change the receive transfer size using the `-o` or `-s` options to the program. See `./axidma_benchmark -h` for the full list of options. 

Eventually, you'll probably need to write an application of your own, but the benchmark program serves as a good starting point.

yaobaishen, 2017-04-12:
Hi Raffaele,

I am using zedboard too. I have read your issue carefully and find that my develop environment is almost the same as yours: I am using kernel/uboot/device-tree v2016.3 too. The "axidma_transfer" can works well, but I run into the same error as yours when run "axidma_benchmark":

>root@zedboard-zynq7:~/xilinx_axidma# ./axidma_benchmark
>AXI DMA Benchmark Parameters:
>Transmit Buffer Size: 7.91 Mb
>Receive Buffer Size: 7.91 Mb
>Number of DMA Transfers: 1000 transfers

>Using transmit channel 0 and receive channel 1.
>Warning: 99.95% of the receive buffer matches the initialization pattern.
>This may mean that the receive buffer was not properly updated.
>Single transfer test successfully completed!
>Beginning performance analysis of the DMA engine.

>DMA Timing Statistics:
>Elapsed Time: 0.08 s
>Transmit Throughput: 104316.99 Mb/s
>Receive Throughput: 104316.99 Mb/s
>Total Throughput: 208633.97 Mb/s

As Brandon has fixed the 3rd buffer allocating issue, the main buffer allocating issue discussed in this thread has been resolved:

> Thus, it needs to allocate a 3rd DMA buffer for these transactions, and it allocates one that is 5 times larger (hard-coded), which is why you see that third allocation. I've decided to remove this [...]

I have checked out the latest xilinx_axidma code, So I am wondering what's the status if you run axidma_benchmark now? Do you still see the "Warning: 99.95% of the receive buffer matches the initialization pattern" log?

gdw439, 2019-03-15:
sorry, I also have no idea about the problem as you when I try to apply dma driver to debian linux, I guess I don't understand the result about how to deal with the problem. so could you talk the method to handle the problem in detai? I am really a new guy in linux driver. Thank you so much.