bperez77/xilinx_axidma #17: axidma: axidma_dma.h: axidma_request_channels: 621: Unable to get slave channel 0: tx_channel.

nick1chao, 2017-03-20:
Hi Brandon,
I'm trying to use your driver on a zedboard, petalinux system.When i run insmod axidma.ko command to load driver,the error in the title comes up.

This is the error log:
 `root@dmaTest:/lib/modules/4.6.0-xilinx/extra# insmod axidma.ko `
`axidma: axidma_dma.h: axidma_request_channels: 621: Unable to get slave channel 0: tx_channel.`
`axidma: probe of axidma_chrdev@0 failed with error -38`

And my device tree of dma part is:
 

     axidma_chrdev: axidma_chrdev@0 {
    		compatible = "xlnx,axidma-chrdev";
    		dmas = <&axi_dma_0 0 &axi_dma_0 1>;
    		dma-names = "tx_channel", "rx_channel";
	};
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		
		
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,include-sg ;
			dma-mm2s-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
			dma-s2mm-channel@40400000 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
		};
	};

I didn't change anything for you driver,but where there is a problem.
Thanks for any help you can offer.

bperez77, 2017-03-21:
I wasn't able to reproduce the problem on my end.  I'm working off the branch xilinx-v2016.3 branch of the main Xilinx kernel (https://github.com/Xilinx/linux-xlnx).

Can you let me know what repository, branch, and specific commit you're using for your kernel?

nick1chao, 2017-03-21:
Thank you for your answer.
I use the petalinux tool to generate uboot kernel and device tree.The kernel version is 4.6.0.[](url)https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html.

I'm going to try to use the kernel version you provided.

raffaelesury, 2017-03-21:
Hi Nick, 
I'm using a device tree generated by HSI through petalinux 2016.3 (even though I'm not using petalinux anymore as I'm using the linaro rootfs on SD card, but still using the xilinx-v2016.3 kernel nevertheless) and then adapted the pl.dtsi file to work with the driver, and I notice that you have the same address for the two channels while I have dma-s2mm-channel@40400030, I think this may be the issue. The rest looks the same.
Here's my pl.dtsi: 
```
/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version: HSI 2016.3
 * Today is: Wed Mar  8 15:25:35 2017
*/

/ {

	axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
	    dma-names = "tx_channel", "rx_channel";
	};

	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>; //NB 20h = 32b
			dma-mm2s-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>; 
				xlnx,device-id = <0x0>;
			};
			dma-s2mm-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>; 
				xlnx,device-id = <0x1>;
			};
		};
	};
};
```

bperez77, 2017-03-22:
Those tags after the device tree nodes are just names that are used to refer to them, so that shouldn't matter. 

Can you attach the full `dmesg` output? I think the issue is that the `clock-names` property is incorrect in your device tree. The example on the main README hasn't been updated in a while, so that's likely the issue. Change your clock names to following:
```
clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
```

Which matches the entry in @raffaelesury's device tree.

nick1chao, 2017-03-23:
Hi
Thank you for your help.
When I use the xilinx-v2016.3 kernel,it can work. I don't know why?
I locate this problem here:   /driver/dma/of_dma.c

the function:

/**
 * of_dma_find_controller - Get a DMA controller in DT DMA helpers list
 * @dma_spec:	pointer to DMA specifier as found in the device tree
 *
 * Finds a DMA controller with matching device node and number for dma cells
 * in a list of registered DMA controllers. If a match is found a valid pointer
 * to the DMA data stored is retuned. A NULL pointer is returned if no match is
 * found.
 */
static struct of_dma *of_dma_find_controller(struct of_phandle_args *dma_spec)
{
	struct of_dma *ofdma;

	### list_for_each_entry(ofdma, &of_dma_list, of_dma_controllers)
		if (ofdma->of_node == dma_spec->np)
			return ofdma;

	pr_debug("%s: can't find DMA controller %s\n", __func__,
		 dma_spec->np->full_name);

	return NULL;
}

bperez77, 2017-03-23:
Hmm that's odd. Can you attach the full output from `dmesg`? The error is probably somewhere in the kenel log.

nick1chao, 2017-03-27:
Hi Brandon,
These days I use the drivers and test programs. When s2mm and mm2s connection,it can work.
But when I send the data to the PL and deal with these data and then back to PS, cannot receive data. What reason is this

bperez77, 2017-03-27:
I'm going to need more information to answer your first question. What's the nature of the error, is it hanging or crashing? What's the design you're using on the PL? What's the application that you're running? Can you attach the program output and `dmesg` output? The error is probably in there.

As for the interrupts, each AXI DMA/VDMA will have an interrupt port (`s2mm_introut` and `mm2s_introut`). You'll need to concatenate these signals together with a `Concat` IP block, and pass it into the `IRQ_F2P` port of the processing system IP. Bits 0 - 7 get assigned to interrupts 29 - 36 and bits 8 - 15 get assigned to interrupts 52 - 59. Then, you need to fill in the `interrupts` property of the device nodes appropiately.

closed 2017-03-27 by bperez77

nick1chao, 2017-03-28:
Hi,
The PS can send data to PL. In order to test PS  can receive  data from PL. I changed the design of PL. 
Counter count has been to write data to fifo, PS to run the application  to read 12 data. I modify your program, realized the function of a DMA read only.

The application read data failed:

> root@linaro-ubuntu-desktop:# ./change_read                        
AXI DMA File Transfer Info:                                                     
        Transmit Channel: 0                                                     
        Receive Channel: 1                                                      
        Transmit size: 48                                                       
        Receive size: 48                                                                                         
output= 48 49 50 51 52 53 54 55 56 57 58 59          //Initialize the data of my read buffer                                
...read_success_callback...               //read callback                                            
transfor time= 0.000149                                                         
recv= 41337014 49 50 51 52 53 54 55 56 57 58 59            //The read function after the data buffer

The dmesg:

> Booting Linux on physical CPU 0x0
Linux version 4.9.0-xilinx (lzrobot@ubuntu) (gcc version 4.6.1 (Sourcery CodeBench Lite 2011.09-50) ) #2 7
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Zynq Zed Development Board
cma: Reserved 16 MiB at 0x1f000000
Memory policy: Data cache writealloc
On node 0 totalpages: 131072
free_area_init_node: node 0, pgdat c0931200, node_mem_map debf0000
  Normal zone: 1024 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 131072 pages, LIFO batch:31
percpu: Embedded 14 pages/cpu @debc9000 s25984 r8192 d23168 u57344
pcpu-alloc: s25984 r8192 d23168 u57344 alloc=14*4096
pcpu-alloc: [0] 0 [0] 1 
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 494532K/524288K available (5120K kernel code, 199K rwdata, 1352K rodata, 1024K init, 230K bss, 13)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0600000   (6112 kB)
      .init : 0xc0800000 - 0xc0900000   (1024 kB)
      .data : 0xc0900000 - 0xc0931e00   ( 200 kB)
       .bss : 0xc0931e00 - 0xc096b75c   ( 231 kB)
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
        RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=2
NR_IRQS:16 nr_irqs:16 16
efuse mapped to e0800000
slcr mapped to e0802000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
zynq_clock_init: clkc starts at e0802100
Zynq clock init
sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[... 148 lines cut ...]
init: bluetooth main process ended, respawning
init: bluetooth main process (1587) terminated with status 1
init: bluetooth main process ended, respawning
init: lightdm main process (1470) terminated with status 1
init: bluetooth main process (1623) terminated with status 1
init: bluetooth main process ended, respawning
init: bluetooth main process (1660) terminated with status 1
init: bluetooth main process ended, respawning
init: bluetooth main process (1681) terminated with status 1
init: bluetooth main process ended, respawning
init: bluetooth main process (1697) terminated with status 1
init: bluetooth main process ended, respawning
init: bluetooth main process (1715) terminated with status 1
init: bluetooth respawning too fast, stopped
init: failsafe-x main process (1611) terminated with status 1
random: crng init done
axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 0 transmit channels and 0 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 674: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 676: VDMA: Found 0 transmit channels and 0 receive channels.

As for the interrupts, i mean do you have any interruption function of the driver? Or do you have any information on how to write the DMA interrupt driver.
For the first time I use the DMA  and  do not know much about the DMA is how to transmit data at the bottom.
Thanks for you help.

bperez77, 2017-03-28:
So aren't you receiving data back then? You have a print statement of the receive buffer after the DMA transfer. I'm not sure what the issue is there.

As for the interrupts, the driver supports interrupts; you don't need to make any modifications to it. You just need to follow the instructions as I listed in the last post.

For example, take the AXI DMA loopback example. In this case, you have one transmit and one receive channel, which each have an interrupt dedicated to them. This leads to device tree nodes for the AXI DMA nodes like follows:
```
axi_dma_0: axidma0@40400000 {
    #dma-cells = <1>;
    compatible = "xlnx,axi-dma", "xlnx,axi-dma-6.03.a", "xlnx,axi-dma-1.00.a";
    reg = <0x40400000 0x10000>;
    clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
    clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";    
    xlnx,include-sg;    
    xlnx,addrwidth = <32>;

    dma-mm2s-channel@40400000 {
        compatible = "xlnx,axi-dma-mm2s-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <0>;
        interrupt-parent = <&intc>;
        interrupts = <0 29 4>;
    };
    
    dma-s2mm-channel@40400000 {
        compatible = "xlnx,axi-dma-s2mm-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <1>;
        interrupt-parent = <&intc>;
        interrupts = <0 30 4>;
    };
};
```

Note the `interrupts` property. The second number corresponds to the interrupt number for the channel. Above, I'm assuming the MM2S interrupt is the lowest bit, and the S2MM interrupt is the next bit, leading to the interrupt values 29 and 30, respectively.

egaznep, 2019-05-28:
> So aren't you receiving data back then? You have a print statement of the receive buffer after the DMA transfer. I'm not sure what the issue is there.
>
> As for the interrupts, the driver supports [...]

I am asking to confirm, the driver does not support DMAs with single channel, right? Currently on my design I have S2MM interface only as I just require that, and I am experiencing the documented problem. If necessary I can re-configure the core so that it has both S2MM and MM2S.

tony1121, 2019-06-11:
using kernel version 4.19.0：
have the same error  ：
axidma: axidma_dma.h: axidma_request_channels: 621: Unable to get slave channel 0: tx_channel.
axidma: probe of axidma_chrdev@0 failed with error -38

solusion：
because not attach the clock info in dts file。now attatch。
&axi_dma_0{
    clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
    clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
            dma-channel@40400000 {
                xlnx,device-id = <0x0>;
                dma-channels = <1>;
            };
            dma-channel@40400030 {
                xlnx,device-id = <0x1>;
                dma-channels = <1>;
            };
};