bperez77/xilinx_axidma #19: cant find driver at /dev/axidma after insmod

xuchendev, 2017-04-05:
hey brandon, i didnt add your cma args into my bootargs, then i run the insmod and look up the dmesg, it shows that 
[  113.487763] random: nonblocking pool is initialized, i think if it was the bootargs problem ?

this is my boot args 
    console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait devtmpfs.mount=0
using the zynq 7020

bperez77, 2017-04-05:
That message about the nonblocking pool is a standard boot message; that just deals with the pool used to generate pseudorandom numbers. If you don't specify `cma=<size>` on the command line, it will default to some size, I think it typically is around 10 MB, so that shouldn't be the issue either.

If you're looking for warnings/errors related to the driver, you should look for messages with "axidma" in them. I'm going to need more information to address this issue. Can you attach the full output from the `dmesg` command, the kernel log buffer?

xuchendev, 2017-04-05:
sure, this is my dmesg log 

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.6.0-xilinx (xuchen@xuchen-dev) (gcc version 4.9.2 (Sourcery CodeBench Lite 2015.05-17) ) #1 SMP PREEMPT Wed Mar 29 11:05:39 CST 2017
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: xlnx,zynq-7000
[    0.000000] cma: Reserved 16 MiB at 0x0f000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c0938dc0, node_mem_map cedf0000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 12 pages/cpu @cedcf000 s19776 r8192 d21184 u49152
[    0.000000] pcpu-alloc: s19776 r8192 d21184 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait devtmpfs.mount=0
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 234632K/262144K available (5303K kernel code, 230K rwdata, 1844K rodata, 1024K init, 222K bss, 11128K reserved, 16384K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
                   lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc07fafe4   (8140 kB)
                     .init : 0xc0800000 - 0xc0900000   (1024 kB)
                     .data : 0xc0900000 - 0xc0939b20   ( 231 kB)
                      .bss : 0xc0939b20 - 0xc09714a8   ( 223 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] efuse mapped to d0800000
[    0.000000] slcr mapped to d0802000
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] zynq_clock_init: clkc starts at d0802100
[    0.000000] Zynq clock init
[    0.000010] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.000029] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[    0.000052] Switching to timer-based delay loop, resolution 3ns
[    0.000134] clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
[    0.000160] timer #0 at d080a000, irq=17
[    0.000482] Console: colour dummy device 80x30
[    0.000497] Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[    0.000511] pid_max: default: 32768 minimum: 301
[    0.000644] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000653] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001250] CPU: Testing write buffer coherency: ok
[    0.001445] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001502] Setting up static identity map for 0x100000 - 0x100058
[    0.100640] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.100739] Brought up 2 CPUs
[    0.100752] SMP: Total of 2 processors activated (1333.33 BogoMIPS).
[    0.100758] CPU: All CPU(s) started in SVC mode.
[    0.101579] devtmpfs: initialized
[... 107 lines cut ...]
[    2.298667] systemd[1]: Listening on /dev/initctl Compatibility Named Pipe.
[    2.309173] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    2.320887] systemd[1]: Reached target Remote File Systems (Pre).
[    2.331275] systemd[1]: Reached target Remote File Systems.
[    2.341255] systemd[1]: Reached target Swap.
[    2.349816] systemd[1]: Listening on udev Kernel Socket.
[    2.375831] systemd[1]: Created slice User and Session Slice.
[    2.380584] systemd[1]: Created slice System Slice.
[    2.421217] systemd[1]: Starting Remount Root and Kernel File Systems...
[    2.430449] systemd[1]: Created slice system-serial\x2dgetty.slice.
[    2.439032] systemd[1]: Starting Journal Service...
[    2.470211] systemd[1]: Starting Load Kernel Modules...
[... cut ...]
[    2.591659] systemd[1]: Starting Apply Kernel Variables...
[    2.611227] systemd[1]: Starting udev Coldplug all Devices...
[    2.618894] systemd[1]: Starting Load/Save Random Seed...
[    2.653703] systemd[1]: Started Journal Service.
[    2.901228] systemd-journald[686]: Received request to flush runtime journal from PID 1
[    4.299087] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.820897] macb e000b000.ethernet eth0: link up (1000/Full)
[    7.820925] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

xuchendev, 2017-04-05:
this is my memory distribution

memory{
    device_type = "memory"
    reg = <0x0 0x10000000>
}

i distribute 256MB memory to PS, 256 MB memory to PL

bperez77, 2017-04-05:
Can you show the `dmesg` output after you run `insmod`? The driver should produce some error messages. Also, can  you show the output from when you run the program (I assume it's `axidma_benchmark`)?

xuchendev, 2017-04-05:
i got nothing changed in **dmesg** command after i run the **insmod** axidma.ko = = , 
run **axidma_benchmark** 
-bash: ./axidma_benchmark: No such file or directory

**lsmod** got the following output 
Module                  Size  Used by
axidma                 19157  0

but i cat the **/dev/devices** not found the axidma device

bperez77, 2017-04-05:
Can you show me your device tree? I suspect that you're missing the device tree node for the driver.

xuchendev, 2017-04-05:
sorry to reply u late, i find some problems in my project and fixing it ,

this is my pl devicetree

/ {
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		FullConvNetTop_0: FullConvNetTop@43c00000 {
			compatible = "xlnx,FullConvNetTop-1.0";
			interrupt-parent = <&intc>;
			interrupts = <0 30 4>;
			reg = <0x43c00000 0x10000>;
			xlnx,s-axi-axilites-addr-width = <0x6>;
			xlnx,s-axi-axilites-data-width = <0x20>;
		};
		axi_vdma_0: dma@43000000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_mm2s_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-vdma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4>;
			reg = <0x43000000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,flush-fsync = <0x1>;
			xlnx,num-fstores = <0x3>;
			dma-channel@43000000 {
				compatible = "xlnx,axi-vdma-mm2s-channel";
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
		};
	};
};

bperez77, 2017-04-05:
Yep, that's the problem. So, you also need a device tree node for my driver that tells it which DMA channels it is allowed to use. See the [README](https://github.com/bperez77/xilinx_axidma/blob/master/README.md) for the information on the format for that node.

xuchendev, 2017-04-05:
OK, thank u brandon , i'll look up the doc : ]

closed 2017-04-05 by bperez77

xuchendev, 2017-04-05:
hey brandon , i add the axi_dma and char_device node to device tree, but unfortunately i cant start the kernel ....my linux version is 4.6.0-xilinx which is the xilinx-v2016.4 branch of linux-xilinx

xuchendev, 2017-04-05:
my question is , i didnt add `DMA` ip into my vivado project ,if i should add the device tree node which u described in the README.md ?

bperez77, 2017-04-05:
You shouldn't need to add the 'axi_dma' node, you already had that in the device tree before. You only needed to add the 'char_device' node.

You don't have the DMA in your Vivado project? As in there's no AXI DMA instantiated in your block design. If you don't have an AXI DMA IP, then the driver is useless, because it will only interact with an AXI DMA IP that is on the PL fabric.

The kernel not starting usually means that you have the `reg` property incorrect. Make sure the address matches the one in the 'Address Editor' in your blockd design.

xuchendev, 2017-04-06:
yes , currently i only use the `VDMA` IP in my IP Desgin which address is 0x43000000, now i use the linux system call `mmap` to use the VDMA, how can i use your driver : ]

bperez77, 2017-04-06:
Yeah, so the driver supports VDMA as well. You just need to create a device tree node for the character device that is like the one in the README. In your case, it would look like:
```
axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};
```

Once you have that, you should be able to insert the module correctly. To use the driver, I would start with the [axidma_benchmark](https://github.com/bperez77/xilinx_axidma/blob/master/examples/axidma_benchmark.c) application. Run that, and make sure it works. Then, you can use that as a template to write your own application that uses the driver.

xuchendev, 2017-04-07:
i am not familiar with the device tree syntax , this is my `system.dtsi` file, is that ok?

`/dts-v1/;
/include/ "zynq-7000.dtsi"
/include/ "pl.dtsi"
/ {
	cpus {
		cpu@0 {
			operating-points = <666666 1000000 333333 1000000>;
		};
	};
	chosen {
		bootargs = "console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait devtmpfs.mount=0";
	};
	aliases {
		ethernet0 = &gem0;
		serial0 = &uart1;
		spi0 = &qspi;
	};
	memory {
		device_type = "memory";
		reg = <0x0 0x10000000>;
	};
};
&gem0 {
    status = "okay";
    phy-mode = "rgmii-id";
    phy-handle = <&ethernet_phy>;
    ethernet_phy: ethernet-phy@0{
        reg = <0>;
    };
};
&intc {
	num_cpus = <2>;
	num_interrupts = <96>;
};
&qspi {
	is-dual = <0>;
	num-cs = <1>;
	status = "okay";
};
&sdhci0 {
	status = "okay";
	xlnx,has-cd = <0x1>;
	xlnx,has-power = <0x0>;
	xlnx,has-wp = <0x1>;
};
&sdhci1 {
	status = "okay";
	xlnx,has-cd = <0x1>;
	xlnx,has-power = <0x0>;
	xlnx,has-wp = <0x1>;
};
&uart1 {
	device_type = "serial";
	port-number = <0>;
	status = "okay";
};
&usb0 {
	dr_mode = "host";
	phy_type = "ulpi";
	status = "okay";
};
&clkc {
	fclk-enable = <0x3>;
	ps-clk-frequency = <33333333>;
};
axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};
`

bperez77, 2017-04-07:
Yeah that looks fine, except I don't see your AXI VDMA device tree nodes anywhere. Are they in a `zynq-zed.dts`?

xuchendev, 2017-04-07:
i find it was incorrect when use dtc to  compile it....

xuchendev, 2017-04-07:
/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version:  
 * Today is: Mon Mar 13 16:00:14 2017
*/

/ {
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		FullConvNetTop_0: FullConvNetTop@43c00000 {
			compatible = "xlnx,FullConvNetTop-1.0";
			interrupt-parent = <&intc>;
			interrupts = <0 30 4>;
			reg = <0x43c00000 0x10000>;
			xlnx,s-axi-axilites-addr-width = <0x6>;
			xlnx,s-axi-axilites-data-width = <0x20>;
		};
		axi_vdma_0: dma@43000000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_mm2s_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-vdma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4>;
			reg = <0x43000000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,flush-fsync = <0x1>;
			xlnx,num-fstores = <0x3>;
			dma-channel@43000000 {
				compatible = "xlnx,axi-vdma-mm2s-channel";
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
		};
	};
};

xuchendev, 2017-04-07:
/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version:  
 * Today is: Mon Mar 13 16:00:14 2017
*/

/dts-v1/;
/include/ "zynq-7000.dtsi"
/include/ "pl.dtsi"
/ {
	cpus {
		cpu@0 {
			operating-points = <666666 1000000 333333 1000000>;
		};
	};
	chosen {
		bootargs = "console=ttyPS0,115200 root=/dev/mmcblk0p2 rw earlyprintk rootfstype=ext4 rootwait devtmpfs.mount=0";
	};
	aliases {
		ethernet0 = &gem0;
		serial0 = &uart1;
		spi0 = &qspi;
	};
	memory {
		device_type = "memory";
		reg = <0x0 0x10000000>;
	};
};
&gem0 {
    status = "okay";
    phy-mode = "rgmii-id";
    phy-handle = <&ethernet_phy>;
    ethernet_phy: ethernet-phy@0{
        reg = <0>;
    };
};
&intc {
	num_cpus = <2>;
	num_interrupts = <96>;
};
&qspi {
	is-dual = <0>;
	num-cs = <1>;
	status = "okay";
};
&sdhci0 {
	status = "okay";
	xlnx,has-cd = <0x1>;
	xlnx,has-power = <0x0>;
	xlnx,has-wp = <0x1>;
};
&sdhci1 {
	status = "okay";
	xlnx,has-cd = <0x1>;
	xlnx,has-power = <0x0>;
	xlnx,has-wp = <0x1>;
};
&uart1 {
	device_type = "serial";
	port-number = <0>;
	status = "okay";
};
&usb0 {
	dr_mode = "host";
	phy_type = "ulpi";
	status = "okay";
};
&clkc {
	fclk-enable = <0x3>;
	ps-clk-frequency = <33333333>;
};

xuchendev, 2017-04-07:
/*
 *  Copyright (C) 2011 - 2014 Xilinx
 *
 * This software is licensed under the terms of the GNU General Public
 * License version 2, as published by the Free Software Foundation, and
 * may be copied, distributed, and modified under those terms.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 */
/include/ "skeleton.dtsi"

/ {
	compatible = "xlnx,zynq-7000";

	cpus {
		#address-cells = <1>;
		#size-cells = <0>;

		cpu0: cpu@0 {
			compatible = "arm,cortex-a9";
			device_type = "cpu";
			reg = <0>;
			clocks = <&clkc 3>;
			clock-latency = <1000>;
			cpu0-supply = <&regulator_vccpint>;
			operating-points = <
				/* kHz    uV */
				666667  1000000
				333334  1000000
			>;
		};

		cpu1: cpu@1 {
			compatible = "arm,cortex-a9";
			device_type = "cpu";
			reg = <1>;
			clocks = <&clkc 3>;
		};
	};

	pmu {
		compatible = "arm,cortex-a9-pmu";
		interrupts = <0 5 4>, <0 6 4>;
		interrupt-parent = <&intc>;
		reg = < 0xf8891000 0x1000 0xf8893000 0x1000 >;
	};

	regulator_vccpint: fixedregulator@0 {
		compatible = "regulator-fixed";
		regulator-name = "VCCPINT";
		regulator-min-microvolt = <1000000>;
		regulator-max-microvolt = <1000000>;
		regulator-boot-on;
		regulator-always-on;
	};

	amba: amba {
		compatible = "simple-bus";
		#address-cells = <1>;
		#size-cells = <1>;
		interrupt-parent = <&intc>;
		ranges;

		adc: adc@f8007100 {
			compatible = "xlnx,zynq-xadc-1.00.a";
			reg = <0xf8007100 0x20>;
			interrupts = <0 7 4>;
[... 325 lines cut ...]
			interrupts = <0 21 4>;
			reg = <0xe0002000 0x1000>;
			phy_type = "ulpi";
		};

		usb1: usb@e0003000 {
			compatible = "xlnx,zynq-usb-2.20a", "chipidea,usb2";
			status = "disabled";
			clocks = <&clkc 29>;
			interrupt-parent = <&intc>;
			interrupts = <0 44 4>;
			reg = <0xe0003000 0x1000>;
			phy_type = "ulpi";
		};

		watchdog0: watchdog@f8005000 {
			clocks = <&clkc 45>;
			compatible = "cdns,wdt-r1p2";
			interrupt-parent = <&intc>;
			interrupts = <0 9 1>;
			reg = <0xf8005000 0x1000>;
			timeout-sec = <10>;
		};
	};
};

xuchendev, 2017-04-07:
this is my tree dts file ,where should i put your char_node  = = ,thank u = =

bperez77, 2017-04-07:
The char node can go anywhere in the device tree. Typically it's placed at the top-level where the `chosen` node and others are located.

Are you getting a compilation error when running the device tree compiler? I'm going to need more information to debug that.

xuchendev, 2017-04-07:
i put the char node under the root node and parallel to the chosen node , it compiles successfully !

xuchendev, 2017-04-07:
now the linux system start up successfully~
but i `insmod` get the error message `segmentation fault`,
this is the `dmesg`

`[   26.684890] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   26.691597] pgd = cd55c000
[   26.693022] [00000000] *pgd=00000000
[   26.695238] Internal error: Oops - BUG: 5 [#1] PREEMPT SMP ARM
[   26.699748] Modules linked in: axidma(O+)
[   26.702445] CPU: 1 PID: 1357 Comm: insmod Tainted: G           O    4.6.0-xilinx #1
[   26.708775] Hardware name: Xilinx Zynq Platform
[   26.711989] task: ce73e7c0 ti: cd454000 task.ti: cd454000
[   26.716094] PC is at axidma_of_parse_dma_nodes+0x1cc/0x48c [axidma]
[   26.721023] LR is at of_find_property+0x40/0x4c
[   26.724230] pc : [<bf0020e8>]    lr : [<c04b6c90>]    psr: 600e0013
               sp : cd455cc0  ip : 00000000  fp : cedefa04
[   26.733081] r10: cdf98c14  r9 : cdf98c00  r8 : cedefeb0
[   26.736987] r7 : 00000014  r6 : 00000000  r5 : cd5669c0  r4 : 00000001
[   26.742195] r3 : 00000000  r2 : 00000000  r1 : 600e0013  r0 : 00000000
[   26.747404] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   26.753219] Control: 18c5387d  Table: 0d55c04a  DAC: 00000051
[   26.757645] Process insmod (pid: 1357, stack limit = 0xcd454210)
[   26.762332] Stack: (0xcd455cc0 to 0xcd456000)
[   26.765376] 5cc0: cd455cd8 00000006 00000001 00000001 00000001 00000000 cedefa04 00000001
[   26.772234] 5ce0: 00000001 c04b7f14 c0963fbc 00000001 00000010 00000001 cedefeb0 cedefeb0
[   26.779090] 5d00: 00000002 bf004098 c0962060 ce550200 00000002 cd477064 cd477048 c04b8124
[   26.785948] 5d20: ffffffff cd5669c0 00000014 bf004098 c0962060 ce550200 00000002 cd477064
[   26.792805] 5d40: cd477048 bf001c10 ce550200 cd5669c0 cd5669c0 ce550200 bf004098 c0962060
[   26.799662] 5d60: 00000000 bf000074 fffffdfb bf000028 ce550210 c0384718 ce550210 c0962058
[   26.806519] 5d80: bf004098 c03830c4 00000000 ce550210 ce550244 bf004098 c0918480 c09063d0
[   26.813376] 5da0: 00000001 c03832a8 00000000 bf004098 c0383228 c0381828 ce497f5c ce54d7b4
[   26.820233] 5dc0: bf004098 00000000 ce607180 c03827c0 bf00288c bf00288d 00000000 bf004098
[   26.827091] 5de0: bf006000 00000000 c09063d0 c03839d0 c03841f8 cdf98dc0 bf006000 c0101700
[   26.833947] 5e00: c093aa58 cdf98e00 cefac700 00000000 00000001 cefac720 00000000 00000001
[   26.840805] 5e20: c073fff3 c0190b14 c0906284 cefac700 00000000 00000001 cedf0000 cd477040
[   26.847662] 5e40: 00000000 00000000 cd477048 c01929a8 00000000 bf004100 00000001 cdef9900
[   26.854519] 5e60: bf004148 cd477040 00000001 c018afb4 bf004100 0000def9 cd455f54 00000001
[   26.861376] 5e80: bf004100 c0175428 bf00410c 00007fff bf004100 c0172b3c 00000000 c0172a64
[   26.868234] 5ea0: bf0042a4 000000ec 00000000 00000000 d09d294c 7f618c10 00000000 c01cbee4
[   26.875090] 5ec0: 00007d79 00000000 7fffffff 00000000 00000000 00000000 00000000 00000000
[   26.881947] 5ee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   26.888805] 5f00: 00000000 00000000 7fffffff 00000000 7f618c10 00000003 0000017b c0106f84
[   26.895662] 5f20: cd454000 00000000 00000000 c0175a78 7fffffff 00000000 00000002 00007d79
[   26.902520] 5f40: 00000000 d09cd000 00007d79 00000000 00000000 d09cd000 00007d79 d09d2384
[   26.909377] 5f60: d09d21f0 d09d4424 000042c8 000047c8 00000000 00000000 00000000 0000182c
[   26.916234] 5f80: 00000025 00000026 0000001a 0000001e 00000010 00000000 1e8f8500 00000000
[   26.923092] 5fa0: 7f62b000 c0106dc0 1e8f8500 00000000 00000003 7f618c10 00000000 bedb54f0
[   26.929948] 5fc0: 1e8f8500 00000000 7f62b000 0000017b 00000000 00000000 7f62a9bc 00000000
[   26.936806] 5fe0: bedb5498 bedb5488 7f60f4a3 b6edd7b2 800e0030 00000003 00000000 00000000
[   26.943700] [<bf0020e8>] (axidma_of_parse_dma_nodes [axidma]) from [<bf001c10>] (axidma_dma_init+0x98/0x1bc [axidma
])
[   26.952979] [<bf001c10>] (axidma_dma_init [axidma]) from [<bf000074>] (axidma_probe+0x4c/0xc4 [axidma])
[   26.961041] [<bf000074>] (axidma_probe [axidma]) from [<c0384718>] (platform_drv_probe+0x50/0xa0)
[   26.968588] [<c0384718>] (platform_drv_probe) from [<c03830c4>] (driver_probe_device+0x118/0x27c)
[   26.976133] [<c03830c4>] (driver_probe_device) from [<c03832a8>] (__driver_attach+0x80/0xa4)
[   26.983248] [<c03832a8>] (__driver_attach) from [<c0381828>] (bus_for_each_dev+0x6c/0x90)
[   26.990105] [<c0381828>] (bus_for_each_dev) from [<c03827c0>] (bus_add_driver+0xc8/0x1e4)
[   26.996960] [<c03827c0>] (bus_add_driver) from [<c03839d0>] (driver_register+0x9c/0xe0)
[   27.003646] [<c03839d0>] (driver_register) from [<c0101700>] (do_one_initcall+0x100/0x184)
[   27.010591] [<c0101700>] (do_one_initcall) from [<c018afb4>] (do_init_module+0x54/0x1b4)
[   27.017363] [<c018afb4>] (do_init_module) from [<c0175428>] (load_module+0x1730/0x1bc4)
[   27.024043] [<c0175428>] (load_module) from [<c0175a78>] (SyS_finit_module+0x88/0x90)
[   27.030555] [<c0175a78>] (SyS_finit_module) from [<c0106dc0>] (ret_fast_syscall+0x0/0x3c)
[   27.037408] Code: e3a02000 eb52d2db e3500000 1a000006 (e5963000) 
[   27.042495] ---[ end trace ea29b3f80d327928 ]---
`

i didnt enable the read channel in `VDMA`, is that problem ?

xuchendev, 2017-04-07:
also you can provide us an example  to let us run the test : ]

bperez77, 2017-04-09:
If you didn't enable the read channel, then you shouldn't have a device tree entry for it. For the examples, you can see `axidma_benchmark` under the `examples` directory. Just run `./axidma_benchmark -h` to see how to run it.

xuchendev, 2017-04-10:
thank u brandon ,finally i run the `insmod` successfully, the dmesg shows that

xuchendev, 2017-04-10:
i also tried VDMA,
`
[18.324202] axidma: axidma_dma.c: axidma_dma_init: 705: DMA: Found 0 transmit channels and 0 receive channels.
[18.324217] axidma: axidma_dma.c: axidma_dma_init: 707: VDMA: Found 1 transmit channels and 0 receive channels.
`

but still 

`-bash: ./axidma_benchmark: No such file or directory`

xuchendev, 2017-04-10:
OK, i comiple it on the zynq device instead of CROSS_COMPILE on the pc ,it will run!

bperez77, 2017-04-10:
You have to copy the `axidma_benchmark` binary to the root filesystem on the SD card (assuming you're doing an SD boot). Otherwise, you won't have that executable to run. It's not a built-in executable, it's compiled on the host side (or you can do it on the board like you did).

Did VDMA also work for you?

xuchendev, 2017-04-11:
not yet, i meet the **DMA receive transaction timed out** issue which yaobaishen asked,

bperez77, 2017-04-11:
So just to clarify, the DMA works, but the VDMA does not? I haven't been able to get VDMA working either. 

For the display app, the image should be in some 32-bit format. I used it with an RGBA image, but the code doesn't rely on that fact other than the 32-bit size. Also, be careful with that example, as I haven't updated it in a while, so I'm not sure how reliable it is.

qudevin521314, 2018-11-10:
Hello, guy, I'm running DMA in linux on zedboard,now i run the ` insmod axidma.ko` successfully, the dmesg shows that 
 `axidma:loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit channels and 0 receive channels.`
but when i run ./axidma_benchmark,the output is 
`AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 MiB
        Receive Buffer Size: 7.91 MiB
        Number of DMA Transfers: 1000 transfers
Error opening AXI DMA device: No such file or directory
Expected the AXI DMA device at the path `/dev/axidma`
Failed to initialize the AXI DMA device.`
i am looking forward to your answer!

bperez77, 2018-11-10:
That's odd, it looks like the module is being loaded correctly. You're not removing the module or restarting the processor between those lines, correct?

Can you send the full output of `dmesg` after running the benchmark program. Please attach the output as a file to the thread.

qudevin521314, 2018-11-10:
[image]

qudevin521314, 2018-11-10:
>
 zynq> dmesg
Booting Linux on physical CPU 0x0
Linux version 4.9.0-xilinx (root@devin-VirtualBox) (gcc version 4.6.1 (Sourcery CodeBench Lite 2011.09-50) ) #1 SMP PREEMPT Fri Nov 9 22:13:05 CST 2018
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: xlnx,zynq-7000
earlycon: cdns0 at MMIO 0xe0001000 (options '115200n8')
bootconsole [cdns0] enabled
cma: Reserved 28 MiB at 0x1c800000
Memory policy: Data cache writealloc
On node 0 totalpages: 131072
free_area_init_node: node 0, pgdat c0a316c0, node_mem_map dfbea000
  Normal zone: 1024 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 131072 pages, LIFO batch:31
percpu: Embedded 14 pages/cpu @dfbc5000 s25984 r8192 d23168 u57344
pcpu-alloc: s25984 r8192 d23168 u57344 alloc=14*4096
pcpu-alloc: [0] 0 [0] 1
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
Kernel command line: console=ttyPS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4 earlyprintk rootwait earlycon cma=25M
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
Memory: 475536K/524288K available (6144K kernel code, 200K rwdata, 1472K rodata, 1024K init, 230K bss, 20080K reserved, 28672K cma-reserved, 0K highmem)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
    lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
    pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
    modules : 0xbf000000 - 0xbfe00000   (  14 MB)
      .text : 0xc0008000 - 0xc0700000   (7136 kB)
      .init : 0xc0900000 - 0xc0a00000   (1024 kB)
      .data : 0xc0a00000 - 0xc0a322c0   ( 201 kB)
       .bss : 0xc0a322c0 - 0xc0a6be5c   ( 231 kB)
Preemptible hierarchical RCU implementation.
        Build-time adjustment of leaf fanout to 32.
        RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=2
NR_IRQS:16 nr_irqs:16 16
efuse mapped to e0800000
slcr mapped to e0802000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
zynq_clock_init: clkc starts at e0802100
Zynq clock init
sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
Switching to timer-based delay loop, resolution 3ns
clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
timer #0 at e080a000, irq=17
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Setting up static identity map for 0x100000 - 0x100058
CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
Brought up 2 CPUs
SMP: Total of 2 processors activated (1333.33 BogoMIPS).
[... 84 lines cut ...]
NET: Registered protocol family 17
can: controller area network core (rev 20120528 abi 9)
NET: Registered protocol family 29
can: raw protocol (rev 20120528)
can: broadcast manager protocol (rev 20161123 t)
can: netlink gateway (rev 20130117) max_hops=1
mmc0: new high speed SDHC card at address e624
Registering SWP/SWPB emulation handler
hctosys: unable to open rtc device (rtc0)
mmcblk0: mmc0:e624 SC16G 14.8 GiB
of_cfs_init
 mmcblk0: p1
of_cfs_init: OK
ALSA device list:
  No soundcards found.
RAMDISK: gzip image found at block 0
EXT4-fs (ram0): mounted filesystem without journal. Opts: (null)
VFS: Mounted root (ext4 filesystem) on device 1:0.
random: sshd: uninitialized urandom read (32 bytes read)
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit channels and 0 receive channels.
zynq>
>