bperez77/xilinx_axidma #21: Unknown symbol xilinx_vdma_channel_set_config

GOOD-Stuff, 2017-04-24:
Hello, I'm work with Linux kernel version of:
``` 
# uname-r
4.6.0-xilinx-22223-gcda28de-dirty
```
I'm allocate CMA of 64M and this is my [dmesg](http://bit.ly/2q6dKL0) output.
[This is](http://bit.ly/2oCH8Z7) my part of device tree.
And when I try load axidma.ko, I get:
```
# insmod axidma.ko 
insmod: can't insert 'axidma.ko': Invalid argument
# dmesg | tail
xilinx-vdma 80400000.dma: Xilinx AXI VDMA Engine Driver Probed!!
axidma: no symbol version for xilinx_vdma_channel_set_config
axidma: Unknown symbol xilinx_vdma_channel_set_config (err -22)
```
But by on Linux kernel of 3.17 this module is loading.
Firstly, I thought that this because Xilinx DMA driver doesn't load with Linux kernel and System.map doesn't have line about ``` xilinx_vdma_channel_set_config ```. But if upload Xilinx DMA driver manually, the kallsyms will be have this line. 
If work with kernel of 3.17, when start benchmark, I get:
```
# ./axidma_benchmark 
AXI DMA Benchmark Parameters:
	Transmit Buffer Size: 7.91 Mb
	Receive Buffer Size: 7.91 Mb
	Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Segmentation fault
```
and if transfer (file kek.txt has some text):
```
# ./axidma_transfer kek.txt lel.txt 
AXI DMA File Transfer Info:
	Transmit Channel: 0
	Receive Channel: 1
	Input File Size: 0.00 Mb
	Output File Size: 0.00 Mb

Segmentation fault
```
Output of [dmesg](http://bit.ly/2pcwSdf) after this execution

GOOD-Stuff, 2017-04-24:
An error occurs when called the function ```dmaengine_prep_slave_sg``` on 208 line

bperez77, 2017-06-24:
Sorry for the late reply, did this issue ever end up being resolved? I suspect it's related to issue #14, so there really isn't a good way to solve it other than switching kernel versions.

Support for 3.x Linux versions was recently removed, because it was too much of a pain to continue supporting, so I'd recommend upgrading.

closed 2017-06-24 by bperez77

ajrodgon, 2017-12-19:
Hi, I don't know if this issue is already solved, but I had just face the same problem. The point was that the module for VMDA in kernel (at least in Petalinux) needs to be included. The route for include this module in the kernel configuration menu is as follows:

->Device Drivers
    -> DMA Engine Support
        -> Xilinx AXI DMAS Engine

Hope this heps!

bperez77, 2017-12-20:
This is somewhat related, but different issue. The problem here was that there was some inconsistency between 3.x and 4.x Linux kernel versions about the name of the configuration setting function. 

This is mentioned in the README under setting up the driver, but I think the actual configuration options are out of date. I should probably change the instructions to go via `make menuconfig` instead.