bperez77/xilinx_axidma #29: SG mode doesn`t work well

sploet, 2017-10-23:
Hi there!

I found out that this software (Scatter Gather mode) will not work properly if transfer data size more than 2^23 (8 Mb).
Yesterday I saw the sources and found out  that in SG mode only one pice of data is used.
So it`s impossible to send data more than 2^23 because of IP core limitation.

I think it is possible to divide transfer data size to pieces of 8Mb and create SG list with more than one element, am I right?

bperez77, 2017-10-29:
This is not possible if you're using AXI DMA at the moment. If you're using AXI VDMA, then this is supported with the `axidma_video_transfer` function, which takes in a list of DMA buffers. 

What's your use case here? I can see about adding support for that.

sploet, 2017-11-03:
Actually, I need a very big buffer to decrease a interrupt overhead.
I get about 4 GB data per axi dma stream, so for blocks of the 8MB, i am going to get 4*1024*1024*1024/8192 interrupts per second. It too much...

Am I right?

bperez77, 2017-11-16:
Sorry for the late reply, I've been traveling recently. Yeah that's correct for 4 GB of data you're going to be getting hundreds of interrupt per transfer.

To be honest, I don't think AXI DMA (I'm referring to IP block in this case), and by extension the AXI4-Stream protocol is designed for this use case. Even if you use SG mode, you'll still need setup and use hundreds of SG descriptors for each transfer.

Are you actually streaming out this 4 GB of data, or are you just using the memory as a scratchpad? I would look into foregoing using the AXI4-Stream protocol and instead look into using the AXI4 protocol directly in your hardware IP.

bperez77, 2017-12-22:
There hasn't been any recent activity on this thread, so I'm going to close it.

closed 2017-12-22 by bperez77