bperez77/xilinx_axidma #30: axidma_oneway_transfer: 2 tlast signals?

casalebrunet, 2017-11-02:
Hello @bperez77 first of all thank you for sharing your code.
I have a problem that makes me tired :) Basically, I have a system where i do not know the amount of data that I have to receive back from the PL so i have a setup like that

`DMA mm2s -> FIFO -> SYSTEM -> FIFO -> ASK_DATA -> FIFO -> DMA s2mm
`
Now I'm doing the test by removing the system block 
`DMA mm2s ->  -> FIFO -> ASK_DATA -> FIFO -> DMA s2mm
`
However, when I send the data and try to collect back with small chunk the second time that i call ASK_DATA from the PS the DMA is blocked. I made some debugging with the debug ip core and I see that even if I'm sending only one time N packets on the first fifo I see 2N packets. In fact the DMA is sending 2 times the tlast signal and I don't know why. please see the image:

[image]

For making the transmit I'm using the one way transfer

`rc = axidma_oneway_transfer(dev, tx_channel, tx_buf, NUM_PACKETS * sizeof(u32), true);
`

have you ever inexperienced an issue like that? I'm making something wrong? Is there a specific limitation on the DMA parameters that I can use? I'm using petalinux and Vivado 2016.4

Thanks,

Simon

casalebrunet, 2017-11-07:
apparently there is a bug in the Xilinx driver provided in the petalinux 2016.3 and 2016.4 (see the forum [link](https://forums.xilinx.com/t5/Zynq-All-Programmable-SoC/Double-data-transfers-on-AXI-DMA-Engine-in-ZynqMP/m-p/805535)). I switched to the 2017.2 and it works. Thanks!

closed 2017-11-07 by casalebrunet

eric950722, 2023-04-23:
Hi, can you share how you did the one-way transfer from PL to PS using this driver? I keep receiving RX channel timeout.