bperez77/xilinx_axidma #31: hello,banchmark wrong

tangxu00, 2017-11-19:
that is the message:
Z-turn# ./axidma_benchmark 
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
        Number of DMA Transfers: 1000 transfers

axidma: axidma_chrdev.c: axidma_mmap: 282: Unable to allocate contiguous DMA memory region of size 8294400.
axidma: axidma_chrdev.c: axidma_mmap: 284: Please make sure that you specified cma=<size> on the kernel command line, and the size is large enough.
Unable to allocate receive buffer from the AXI DMA device: Cannot allocate memory
Z-turn# 

and i add cma=25M to devicetree:
bootargs = "console=ttyPS0,115200 root=/dev/mmcblk0p2 rw rootfstype=ext4
  earlyprintk rootwait earlycon cma=25M";
how?

tangxu00, 2017-11-19:
sorry, my mistake, i add cma=25M to uboot, and it worked!

bperez77, 2017-11-21:
The answer to that is maybe. The driver has support for VDMA, but I was never able to get it to work even in a simple loopback mode. As discussed in #15, I believe this may actually be because of a lack of support in the backend Xilinx driver for DMA. However, that may have changed. In issue #25, it seems that someone may have gotten AXI VDMA to work. Perhaps @yaobaishen can comment on this?

If you want some help debugging, can you send me your device tree entries for AXI VDMA. Also, can you send the output of `dmesg` immediately after you run the benchmark?

tangxu00, 2017-11-21:
thanks for your help, and i have emailed yaobaishen,but no reply.
i used zynq 7010, and the leasted linux-4.9.
that is my pl.dtsi:
`/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version:  
 * Today is: Mon Nov 20 15:56:55 2017
*/

/ {
  axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};
	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_vdma_0: dma@43000000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_mm2s_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-vdma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x43000000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,flush-fsync = <0x1>;
			xlnx,num-fstores = <0x3>;
			dma-channel@43000000 {
				compatible = "xlnx,axi-vdma-mm2s-channel";
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x18>;
				xlnx,device-id = <0x0>;
			};
			dma-channel@43000030 {
				compatible = "xlnx,axi-vdma-s2mm-channel";
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x18>;
				xlnx,device-id = <0x1>;
			};
		};
	};
};
`
and  dmesg after run the benchmark:
`Z-turn# ./axidma_benchmark -v
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
        Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
axidma: axidma_dma.c: axidma_start_transfer: 298: VDMA receive transaction timed out.
Failed to perform the AXI DMA read-write transfer: Timer expired
Z-turn# dmesg 
Booting Linux on physical CPU 0x0
Linux version 4.9.0-xilinx (osrc@osrc-virtual-machine) (gcc version 4.6.1 (Sourcery CodeBench Lite 2011.09-50) ) #1 SMP PREEMPT Sun Nov 19 19:34:06 CST 2017
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: xlnx,zynq-7000
cma: Reserved 28 MiB at 0x3e400000
Memory policy: Data cache writealloc
On node 0 totalpages: 262144
free_area_init_node: node 0, pgdat c0a31500, node_mem_map ef7f8000
  Normal zone: 1536 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 196608 pages, LIFO batch:31
[... 146 lines cut ...]
Registering SWP/SWPB emulation handler
mmc0: new high speed SDHC card at address 1234
hctosys: unable to open rtc device (rtc0)
mmcblk0: mmc0:1234 SA16G 14.6 GiB 
 mmcblk0: p1
of_cfs_init
of_cfs_init: OK
ALSA device list:
  No soundcards found.
RAMDISK: gzip image found at block 0
EXT4-fs (ram0): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (ram0): mounted filesystem without journal. Opts: (null)
VFS: Mounted root (ext4 filesystem) on device 1:0.
devtmpfs: mounted
Freeing unused kernel memory: 1024K (c0900000 - c0a00000)
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
random: sshd: uninitialized urandom read (32 bytes read)
export_store: invalid GPIO 110
axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 705: DMA: Found 0 transmit channels and 0 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 707: VDMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_start_transfer: 298: VDMA receive transaction timed out.
Z-turn# `
I am working on using vdma to capture hdmi siginal. but it never worked with linux . did your driver or library provid vdma interface ?
thank you for reply.

yaobaishen, 2017-11-21:
Sorry that I didn't work out VDMA either, so I am looking forward someone to verify it too.

tangxu00, 2017-11-22:
I will try working on it , yaobaishen reply me ,and he give up . thanks for you great work !

tangxu00, 2017-11-23:
hei , I changed my hardwere desigin in vivado , maybe the vdma worked ?

Z-turn# insmod axidma.ko 
axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 705: DMA: Found 0 transmit channels and 0 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 707: VDMA: Found 1 transmit channels and 1 receive channels.
Z-turn# ./axidma_benchmark -v
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
        Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Single transfer test successfully completed!
Beginning performance analysis of the DMA engine.

random: fast init done

DMA Timing Statistics:
        Elapsed Time: 27.71 s
        Transmit Throughput: 285.42 Mb/s
        Receive Throughput: 285.42 Mb/s
        Total Throughput: 570.83 Mb/s
Z-turn#

bperez77, 2017-11-24:
Interesting, well you're the first person to have VDMA working, so that's good. There must have been some issue with my design when I was testing.

Hmm, the output of the timing statistics doesn't make much sense. According to the bandwidth numbers, your transfer should be completing in far less time, but it's not. So, for some reason the numbers are are inconsistent, which I haven't seen before.

I do notice a line of output `random: fast init done`, which may be causing some delay? Not sure if that is the source are not. Do you get the same results after running the benchmark several times in a row?

tangxu00, 2017-11-24:
my bandwidth numbers is 24 , for RGB . 
I try several times ,"random: fast init done" appear in the first time , I dont know what it mean.always 27.71s.

bperez77, 2017-11-24:
I see, so the time is consistent. That's very odd though, because obviously 7.91 Mb / 27.71 s obviously does not come out to 285.42 Mb/s. The 285.42 Mb/s throughput number is about what I would expect, but the elapsed time is way too long. You haven't made any changes to the axidma benchmark code, correct?

Sure thing, that's pretty straightforward. So you can just use `axidma_malloc` to allocate your single frame buffer. Then, you can use `axidma_video_transfer` to setup a loop transfers, where the buffer is continuously streamed out from DRAM to your IP. Then, it's up to your application code to update the buffer as need.

Alternately, if you're planning on using Xilinx's DRM driver, you can use `axidma_register_buffer` to share the DRM driver's DMA buffer with my driver. Naturally, you need to get a handle to the DRM driver's buffer through `libdrm` first.

bperez77, 2017-11-24:
Oh one other thing I noticed is the following line:
```
axidma: loading out-of-tree module taints kernel.
```

This indicates that the kernel you're running on your board, and the one that you built the driver against do not match. This isn't the cause of the timing issue, but it can cause the driver to crash, so I'd recomment making sure you're using the kernel you built the driver against.

tangxu00, 2017-11-25:
I have seen the banchmark.c ,in " DMA Timing Statistics " function , Number of DMA Transfers is 1000 ,so 7.91Mb/27.71s *1000=285.42Mb/s.
Inorder to use your driver , I bulild linux-4.9 to my board ,I can sure that I use the it  to bulid the driver . this problem maybe  related to my config for kernel .

bperez77, 2017-11-27:
Sorry had a bit of a brain fart there, you're right about the time reported by the benchmark.

What throughput are you expecting for the transfers? I know that my driver does introduce a bit of an overhead, but you should still get near the maximum performance. Looking at table 2-3 of the [AXI VDMA User Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_vdma/v6_3/pg020_axi_vdma.pdf), I think it's near the expected. Naturally, the exact throughput depends on data width configured in your AXI VDMA IP block.

tangxu00, 2017-11-29:
I see "axidma_video_transfer" function just use tx channel for display , on the contrary , I need datas streamed out from IP to DRAM buffer , not a loop , but a oneway road from external video signal to DRAM. I have not seen any “video_read” founction , maybe you can add `AXIDMA_DMA_VIDEO_READ` in ioctl when you have time , just a suggestion .

bperez77, 2017-11-29:
Ahh I see. Actually, if you're just doing transfer one at a time, you can utilize `axidma_oneway_transfer` or `axidma_twoway_transfer`, with a VDMA channel. Of course, this will be slower, which I'm guessing is what you were referring to in your question as to why it's so slow.

Otherwise, I can send you a patch for continuous loop video read, analgous to `axidma_video-write`. Unfortunately, I'm busy the next few weeks, so I won't be able to test it fully, but I can still send you the patch.

tangxu00, 2017-11-30:
Thank you so much , that's will be great helpful !

bperez77, 2017-12-01:
Ok, I added support with c3181d8.

This code is currently untested, I only checked that it compiled against the most recent version of Xilinx's kernel. Unfortunatley, I don't have a device to test with, so could you run and test the code, and let me know if you run into any issues.

tangxu00, 2017-12-03:
[vdma_test.txt](https://github.com/bperez77/xilinx_axidma/files/1524705/vdma_test.txt)
I write a simple app use vmda loop to transfer files , but i failed .
that is dmsg:
axidma: axidma_dma.c: axidma_dma_init: 706: DMA: Found 0 transmit channels and 0 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 708: VDMA: Found 1 transmit channels and 1 receive channels.
Unhandled fault: page domain fault (0x01b) at 0xbeea3cb0
pgd = ef318000
[beea3cb0] *pgd=3de24831
Internal error: Oops - BUG: 1b [#1] PREEMPT SMP ARM
Modules linked in: axidma(O)
CPU: 0 PID: 720 Comm: vdma_test Tainted: G           O    4.9.0-xilinx #2
Hardware name: Xilinx Zynq Platform
task: ef1b8440 task.stack: ef174000
PC is at axidma_video_transfer+0xd4/0x194 [axidma]
LR is at 0x500
pc : [<bf001c08>]    lr : [<00000500>]    psr: 80000013
sp : ef175df8  ip : beea3cb0  fp : beea3c34
r10: 00000000  r9 : ef174000  r8 : 00000005
r7 : 002a3000  r6 : ef12ad80  r5 : 00000000  r4 : ef175e7c
r3 : 000002d0  r2 : 00000000  r1 : ef25aa80  r0 : ef12ad80
Flags: Nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 2f31804a  DAC: 00000051
Process vdma_test (pid: 720, stack limit = 0xef174210)
Stack: (0xef175df8 to 0xef176000)
5de0:                                                       002a3000 60000013
5e00: ef31ada8 00000001 ef25aa80 00000000 00000000 00000000 00000000 00000000
5e20: 00000000 00000001 00000001 00000000 00000022 ef1b8440 00000000 00000500
5e40: 00000003 000002d0 00000005 beea3c14 ef12ad80 00000051 80185707 bf000dc0
5e60: 00000707 c0a08648 ef3c0200 00000000 00000000 ef25a780 ee9714d0 00000000
5e80: 00000001 beea3cb0 00000500 00000003 000002d0 c01b7998 ef3b21c4 ee978c40
5ea0: ee9714d0 c01b8b28 ee9714d0 00000707 040444fb c01b8bdc ee9714d0 040444fb
5ec0: ef0c6780 c01ba354 ed85e748 00000000 ef0c6780 00000000 00000000 002a3000
5ee0: 00002000 ef0c6780 beea3c14 ef3b20f0 80185707 c01ded20 40049409 c01df680
5f00: ef0c6780 00000001 002a3000 00000003 00000000 ee978c40 000002a3 c01ba754
5f20: 00000000 00000000 00000000 b68d9000 00000001 ee978c78 ef0c6780 00000003
5f40: 002a3000 00000000 ef174000 00000000 beea3c34 c01aa3e8 00000001 00000000
5f60: 00000000 ef175f6c 00000005 ef0c6780 ef0c6780 beea3c14 80185707 00000005
5f80: ef174000 00000000 beea3c34 c01df70c 00008964 00000000 000000d8 00000036
5fa0: c0106f64 c0106da0 00008964 00000000 00000005 80185707 beea3c14 beea3c14
5fc0: 00008964 00000000 000000d8 00000036 b6f65000 00000000 b6fc3000 beea3c34
5fe0: 00000000 beea3c00 b6f91cac b6ef1c1c 60000010 00000005 00000000 00000000
[<bf001c08>] (axidma_video_transfer [axidma]) from [<bf000dc0>] (axidma_ioctl+0x804/0x9a0 [axidma])
[<bf000dc0>] (axidma_ioctl [axidma]) from [<c01ded20>] (vfs_ioctl+0x18/0x34)
[<c01ded20>] (vfs_ioctl) from [<c01df680>] (do_vfs_ioctl+0x838/0x88c)
[<c01df680>] (do_vfs_ioctl) from [<c01df70c>] (SyS_ioctl+0x38/0x54)
[<c01df70c>] (SyS_ioctl) from [<c0106da0>] (ret_fast_syscall+0x0/0x3c)
Code: e1a00006 e58d7000 e1a02005 e59d1010 (e79c3105) 
---[ end trace 94a751a56bb75147 ]---

bperez77, 2017-12-04:
Yeah, even if there's an issue with your app's memory, it shouldn't cause a segfault in the kernel. I've pushed something new, so can you pull and try again.

Also, can you post the output as a text file? It looks the formatting of it got messed up a bit.

tangxu00, 2017-12-04:
hei , I tried again . I changed your axidma_transfer.c , use 'axidma_get_vdma_tx(axidma_dev)' function to change into vdma channel , It worked well to transfer file . 
so i changed `axidma_twoway_transfer` into `axidma_oneway_transfer` ,the same issue comes . 
and axidma_display.c also failed.
[axidma_transfer.txt](https://github.com/bperez77/xilinx_axidma/files/1526473/axidma_transfer.txt)
[axidma_transfer-dmsg.txt](https://github.com/bperez77/xilinx_axidma/files/1526474/axidma_transfer-dmsg.txt)
[vdma.txt](https://github.com/bperez77/xilinx_axidma/files/1526475/vdma.txt)
[vdma-dmsg.txt](https://github.com/bperez77/xilinx_axidma/files/1526476/vdma-dmsg.txt)

bperez77, 2017-12-18:
Huh, that's really odd. I don't think I've seen any error like that before. The segfault from your logs seems to be pointing to the crash happening in the driver's video transfer function. However, even after combing over the funcntion, I can't find any obvious place where it would segfault. It's eve more odd that the two way transfer works perfectly fine for you.

Just to confirm, you are on the latest version of the driver?

dlaurentiu, 2017-12-25:
Hi Brandon,

I've just tried to use example axidma_display_image with a VDMA core and I get the same crash. Running the latest xilinx-linux on a ZC702 board.

[crash.txt](https://github.com/bperez77/xilinx_axidma/files/1586150/crash.txt)

(PS. Thank you for this project; it's a lifesaver.)

bperez77, 2017-12-30:
Oh got it, I think I might know what the issue is. I made a pretty simple mistake. In the IOCTL for AXIDMA_VIDEO_WRITE (and incidentally the read IOCTL as well), I'm not copying the array of frame buffers from user space to kernel space, which is a big no-no. What is likely happening is that memory happens to be paged out, which leads to the segfault. The reason chaging to `void *frame_buffers` works is because it's handle by the first `coipy_from_user` call in that IOCTL.

So, I need a second call to `copy_from_user` to copy the array of frame buffers. Let me make that change and then push it.

tangxu00, 2018-01-12:
sorry for later ,after long time debug , I cant get a  complete frame picture use your  vdma driver , so I begin my own driver use ioremap register of vdma ,and it easy to work .  but it is not normative to linux kernel ,so I still want to try your driver , have you update the code ?
thank you .

bperez77, 2018-01-12:
Not yet, but I was planning on getting it up this weekend. I'll update this issue once it's up.

elektrokokke, 2018-02-20:
I can confirm that this issue is caused by not calling `copy_from_user` on the frame buffer array.

My dirty fix for this was:
```diff
--- a/driver/axidma_chrdev.c
+++ b/driver/axidma_chrdev.c
@@ -345,6 +345,7 @@
     struct axidma_inout_transaction inout_trans;
     struct axidma_video_transaction video_trans;
     struct axidma_chan chan_info;
+    void *framebuffers[32];
 
     // Coerce the arguement as a userspace pointer
     arg_ptr = (void __user *)arg;
@@ -452,16 +453,13 @@
                            "AXIDMA_DMA_VIDEO_READ.\n");
                 return -EFAULT;
             }
-
-            // Verify that we can access the array of frame buffers
-            size = video_trans.num_frame_buffers *
-                   sizeof(video_trans.frame_buffers[0]);
-            if (!axidma_access_ok(video_trans.frame_buffers, size, true)) {
-                axidma_err("Unable to copy frame buffer addresses from "
-                           "userspace for AXIDMA_DMA_VIDEO_WRITE.\n");
-                return -EFAULT;
-            }
-
+            if (copy_from_user(framebuffers, video_trans.frame_buffers,
+            		video_trans.num_frame_buffers * sizeof(void*)) != 0) {
+				axidma_err("Unable to copy framebuffer pointers from userspace for "
+						   "AXIDMA_DMA_VIDEO_READ.\n");
+				return -EFAULT;
+			}
+            video_trans.frame_buffers = framebuffers;
             rc = axidma_video_transfer(dev, &video_trans, AXIDMA_READ);
             break;```

There is a lot of room for improvement on this, however...

Cheers

P.S.: Thx for the nice piece of work. For simple interfacing with VDMA cores I however begin to think that the Xilinx driver makes it much much to complicated compared to the bare-metal way...

juergenmuc, 2018-03-17:
Hello Brandon,
first of all I would like to send you a huge THANK YOU for your great work!
I tried to get VDMA working for - believe it or not - months (!) now and I finally succeeded using your code, what a help!
I managed to get VDMA working on a Zynq Zybo 7010 board, running kernel xilinx 4.4.30 and ubuntu 16.04.
I modified the "image" size of axidma_benchmark a little and got the following data rates, I'm not yet sure if this is a good and plausible value for a PL clock of 150MHz.

./axidma_benchmark -v

AXI DMA Benchmark Parameters:
    Transmit Buffer Size: 1.17 Mb
    Receive Buffer Size: 1.17 Mb
    Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Step #1
Single transfer test successfully completed!
Beginning performance analysis of the DMA engine.

DMA Timing Statistics:
    Elapsed Time: 35.04 s
    Transmit Throughput: 33.44 Mb/s
    Receive Throughput: 33.44 Mb/s
    Total Throughput: 66.88 Mb/s

Next step will be to use a custom IP as axi stream source, I will also give the axidma_transfer a try.

Please keep developing great software, your work is highly appreciated!
All the best from Munich, home of Oktoberfest  ;-)
J.

bperez77, 2018-03-19:
@tangxu00 and @elektrokokke the most recent commit should resolve this issue. Let me know if you guys encounter any additional issues.

bperez77, 2018-03-19:
@juergenmuc thanks, I appreciate it! Those numbers seem reasonable, though it's hard to say at a high-level glance. The smaller your transfer size, the less throughput you will see from the driver. This is because the system calls to initiate transfers have a relatively high overheard, so the larger the transfer, the more this cost will be amortized.

commit a0717913 (2018-03-19): Added a missing copy from user for the frame buffer arrays.

This closes #31. The video transaction IOCTLs were missing a copy from
user for the array of frame buffer addresses. While the kernel can
access userspace arrays just fine, this will cause an issue if the array
is paged out or the specified address is invalid. It is proper practice
to allocate a temporary array in kernel space, then copy the contents of
the userspace array into this.
closed 2018-03-19 by bperez77