bperez77/xilinx_axidma #33: Problem with callback function

Westwood68, 2017-12-06:
Hi Brandon,

I tried to register a callback function for the RX channel using the axidma_set_callback() function. I modified the benchmark example for that reason, but unfortunately, the callback function is never triggered. 

After some debugging I came a bit closer to the root of the problem. In the library function axidma_callback() the signal from the kernel driver is evaluated and the previously registered callback function (if any) is triggered. For some reason, the channel ID derived from the signal is always 0, while my callback function is registered for channel 1 (RX), so it will never be triggered.

I registered a second callback function for the TX channel for testing. When starting the modified benchmark example with "-n 10" parameter, the axidma_callback() function is called 10 times, but it should be 20, since we have 10 TX and 10 RX operations. It seems that no signal is emitted for the RX operation.

bperez77, 2017-12-18:
Sorry for the late reply. This is definitely one of the less-tested areas of the code. I'm assuming you're using the two-way transfer as was used in the original benchmark app? If so, then this is expcted, albeit very confusing behavior.

The software stack will only invoke your callback function if the `wait` parameter is false. Otherwise, there's no real need for a callback, as you are explicitly waiting for the transfer to complete. So, what happens when you call a two-way transfer is as follows. Regardless of what you specify for `wait`, the TX transfer will be setup as if `wait` is false, so its callback function will always be used. For the RX transfer, its callback will be used only if `wait` is false.

Now, this is defiinitely confusing behavior, so I should probably set it up so that the RX channel doesn't invoke its callback if `wait` is true.

guerricmeurice, 2018-12-05:
Hello,
At first let me thank you and all contributors for the fantastic job. Your driver is really nice and efficient. To my opinion, by far the best about this subject.

I'm running in same trouble as @Westwood68 . I modified the transfer example, using independent one-way read and write calls. I don't use wait on the different calls but I rely on the axidma_set_callback() registering function.

In my application, after calling the axidma_oneway_transfer() function, I have some operations to perform before waiting for completion. As a result, after my custom operations I put the following piece of code:
`    // we need the mutex for safety
    pthread_mutex_lock( &dma_mutex );
    /* block this thread until another thread signals cond. While
       blocked, the mutex is released, then re-aquired before this
       thread is woken up and the call returns. */ 
    pthread_cond_wait( &dma_cond, &dma_mutex );  
    // We do not pass any data from the callback, so mutex can be released immediately
    pthread_mutex_unlock( &dma_mutex );`

In the main() of my test application, I also have the following code. First, mutex/cond global instantiation and custom callback definition. Here the callback intent is to continue execution after receive of completion signal:

 
`/*----------------------------------------------------------------------------
 * global Internal Variables
 *----------------------------------------------------------------------------*/

pthread_mutex_t dma_mutex = PTHREAD_MUTEX_INITIALIZER;
pthread_cond_t dma_cond = PTHREAD_COND_INITIALIZER;

/*----------------------------------------------------------------------------
 * User callback function to be invoked upon completion of an 
 * asynchronous transfer for the specified DMA channel.
 *----------------------------------------------------------------------------*/

axidma_cb_t dma_user_transfer_complete_callback(pthread_mutex_t* dma_mutex_cb_ptr, pthread_cond_t* dma_cond_cb_ptr)
{
    // we're going to manipulate the cond, so we need the mutex
    pthread_mutex_lock( dma_mutex_cb_ptr );
    fprintf(stderr, "executing user trasnfer complete callback\n");
    // wake up the main thread (AS it is sleeping) to terminate operations
    pthread_cond_signal( dma_cond_cb_ptr ); 
    pthread_mutex_unlock( dma_mutex_cb_ptr );    

    return 0;
}`

And in the main() of my test application, I also register the custom callback function:
`axidma_set_callback(axidma_dev, trans.output_channel, dma_user_transfer_complete_callback(&dma_mutex, &dma_cond), NULL);` 

The point is that I see that custom callback is directly called, not waiting for the transaction to be complete. So this ruins the intent of the code.
If I put the one-way function in wait=true mode, and do my custom operations by hand (before 10s timeout), I indeed see that system works as intended (PL data is sent to PS and stored in a file). So I believe the rest of the system is working.

Do you have any suggestion or idea?
Or should I check for a specific signal in the custom callback?

I'm not really a Linux driver guru, so any help is welcome.
Thanks again for your great job!

-- 
Guerric

guerricmeurice, 2018-12-06:
Let me self answer this.
The callback mecanism works. My code is the problem: my dma_user_transfer_complete_callback() has the wrong signature. I don't understand why it is compiling / executing without visible error but it is the problem.
I have something that works now, let me take the time to clean the file in order to share it.
During compilation of my example, make sure you link against pthread lib otherwise you will get funny results :)

eleICoto, 2019-07-05:
@guerricmeurice Hi, right I have this problem too, can you show more detail about how to slove this problem?

guerricmeurice, 2019-07-05:
Hello @eleICoto , it was still in my todo list indeed.
Let me some time to prepare something. I expect to finish that by today (CEST time).

eleICoto, 2019-07-05:
@guerricmeurice Thanks a lot, although I can make this transfer function some part works, I still can't estimate real DMA & IP time, it seems like the DMA transfer function just send the command to DMA and abort whether the wait indicator is set to true or false

guerricmeurice, 2019-07-06:
Hello @eleICoto ,
Here is my variation of te original axidma_transfer.c file. It is called axidma_transfer-rx.c.
[axidma_transfer-rx_guerricmeurice.zip](https://github.com/bperez77/xilinx_axidma/files/3364522/axidma_transfer-rx_guerricmeurice.zip)

The main difference is the use of a non blocking call: axidma_oneway_transfer in order to be able to perform other operations (like PL register accesses, using UIO method in my example) before waiting for completion. The use of UIO is not mandatory. But use of pthread_mutex_lock, pthread_cond_wait and the axidma_set_callback on dma_user_transfer_complete_callback is.

And do not get caught! : do not forget to add -lpthread to the linker flags of the makefile (in original examples.mk, to the EXAMPLES_LINKER_FLAGS). Compiler will not complain about this and you will end up with "empty" pthread functions.

I think this piece of code could be added in the repository as it shows how to use the non blocking call.

About timeout, it is still at bit obscure to me how it is handled. I think this should be the topic of another thread.

Enjoy!

Guerric - BitandByte.io

eleICoto, 2019-07-07:
@guerricmeurice Thanks for your sharing, but I'm still a little bit confused about this sync & async transfer.
This is my block design
[image]
The process flow is like this.
First, CPU initiates a transfer to send data from ps DDR to my IP(AXIS_CNN_Top_Lite).
Then, my IP will process these data and output to DMA.
Finally, CPU initiates another transfer to receive data from IP to ps DDR.
During this process, the CPU can wait until this whole process finish and then move data send by IP out of memory buffer.
I need a wait flag mechanism to ensure the receiving process is totally finished.
So, do you have any advice on which transfer function (one way or two way) I should use and how to implement this wait flag mechanism?

eleICoto, 2019-07-07:
@guerricmeurice And thanks for your great work!

guerricmeurice, 2019-07-07:
@eleICoto If I understand well, data is flowing though AXIS_CNN. As data is coming from the PS, I don't think you need the axidma-transfer-rx file I shared.
If AXIS_CNN does not take too much time to crunch your data, then you should be able to use the original twoway transfer example (in axidma_axistransfer.c). The example assumes DMA IN and OUT are looped and (at first glance) your application looks similar in terms of data flow.
In the example, the axidma_twoway_transfer() function will return upon receiving of the specified amount of data from the PL (or if timeout is reached).

eleICoto, 2019-08-23:
@guerricmeurice Sorry for the late reply, I have been solving the CCI issues for last month, 
The problem I'm facing is that the build-in clock() function can not measure DMA and IP process time correctly, in Verilog simulation my IP needs at least 200 us to finish process data, but when I measure time like this
[image]
It shows that the DMA transfer function only takes about 40us and the most confusing part is that I can certainly relieve the data send by my IP correctly
[image]

honorpeter, 2020-01-09:
> waiting

> Hello @eleICoto ,
> Here is my variation of te original axidma_transfer.c file. It is called axidma_transfer-rx.c. [...]

@guerricmeurice 

Hello  guerricmeurice ,
At first let me thank you and all contributors for the fantastic job,  I  use your code 
axidma_transfer-rx.c to use the callback,But callback mecanism can't work, I use the app on petalinux and How to add -lpthread to the linker flags of the makefile (in original examples.mk, to the EXAMPLES_LINKER_FLAGS). Compiler will not complain about this and you will end up with "empty" pthread functions.

honorpeter, 2020-01-09:
> signature
Did  this axidma_transfer-rx_guerricmeurice.zip  have  wrong signature?

honorpeter, 2020-01-09:
> Hello @eleICoto ,
> Here is my variation of te original axidma_transfer.c file. It is called axidma_transfer-rx.c. [...]

(*axidma_cb_t)  
 Different types  as your dma_user_transfer_complete_callback?

honorpeter, 2020-01-10:
@guerricmeurice 

./axidma-benchmark                                      
AXI DMA Benchmark Parameters:                                                   
        Transmit Buffer Size: 7.91 MiB                                          
        Receive Buffer Size: 7.91 MiB                                           
        Number of DMA Transfers: 1000 transfers                                 
                                                                                
Using transmit channel 0 and receive channel 1.                                 
Single transfer test successfully completed!                                    
Beginning performance analysis of the DMA engine.                               
                                                                                
DMA Timing Statistics:                                                          
        Elapsed Time: 2.29 s                                                    
        Transmit Throughput: 3449.73 MiB/s                                      
        Total Throughput: 3449.73 MiB/s                                         
[  142.666286] axidma: axidma_dma.c: axidma_prep_transfer: 236: Unable to prepa.
[  143.677187] xilinx-vdma b0000000.dma: Cannot stop channel ffffffc87976c218: 0
Failed to perform the AXI DMA transfer: Device or resource busy     

$dmesg
[   47.119431] xilinx_axidma: loading out-of-tree module taints kernel.         
[   47.129002] axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 4 transmi.
[   47.138946] axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transm.
[  142.666286] axidma: axidma_dma.c: axidma_prep_transfer: 236: Unable to prepa.
[  143.677187] xilinx-vdma b0000000.dma: Cannot stop channel ffffffc87976c218: 0
[  158.444494] axidma: axidma_dma.c: axidma_start_transfer: 301: DMA transmit t.
[  174.572498] axidma: axidma_dma.c: axidma_start_transfer: 301: DMA transmit t.
[  187.628494] axidma: axidma_dma.c: axidma_start_transfer: 301: DMA transmit t.
[  201.452494] axidma: axidma_dma.c: axidma_start_transfer: 301: DMA transmit t.

Do you have any suggestion or idea?
Or should I check for a specific signal in the custom callback?

I'm not really a Linux driver guru, so any help is welcome.
Thanks again for your great job!

guerricmeurice, 2020-01-13:
@honorpeter Are you using DMA and VDMA at same time? The code I shared is only tested for DMA. I never investigated VDMA. Can you try by using only DMA?

honorpeter, 2020-01-19:
> @honorpeter Are you using DMA and VDMA at same time? The code I shared is only tested for DMA. I never investigated VDMA. Can you try by using only DMA?

No, and mybe this  was my device tree error. I had solved  in my new  device tree. And now it works fine. Thank you.

guerricmeurice, 2020-01-19:
Ok great. 
This was looking like a strange error indeed.

honorpeter, 2020-01-22:
> Ok great.
> This was looking like a strange error indeed.

Did you try to axidma_malloc   tx_buf1 tx_buf2 tx_buf3 tx_buf4 as below  .. More than one buf for several DMA，and trasfer successfully?

tx_buf= axidma_malloc(axidma_dev, tx_size);
    if (tx_buf == NULL) {
        perror("Unable to allocate transmit buffer from the AXI DMA device.");
        rc = -1;
        goto destroy_axidma;
    }
    rx_buf = axidma_malloc(axidma_dev, rx_size);
    if (rx_buf == NULL) {
        perror("Unable to allocate receive buffer from the AXI DMA device");
        rc = -1;
        goto free_tx_buf;
    }

tx_buf1= axidma_malloc(axidma_dev, tx_size1);
    if (tx_buf 1== NULL) {
        perror("Unable to allocate transmit buffer from the AXI DMA device.");
        rc = -1;
        goto destroy_axidma;
    }
    rx_buf1 = axidma_malloc(axidma_dev, rx_size1);
    if (rx_buf 1== NULL) {
        perror("Unable to allocate receive buffer from the AXI DMA device");
        rc = -1;
        goto free_tx_buf;
    }

loylosky, 2020-11-04:
> > @honorpeter Are you using DMA and VDMA at same time? The code I shared is only tested for DMA. I never investigated VDMA. Can you try by using only DMA?
> 
> No, and mybe this was my device tree error. I had solved in my new device tree. And now it works fine. Thank you.

Hi, I met the same problem recently. I have tried several ways to solve it but none of them worked. May I ask how you solve this problem by modifying the device tree file specifically? Thanks!