bperez77/xilinx_axidma #34: Any idea about xilinx-dma 40400000.vdma: Channel ded141d0 has errors 40 cdr 0 tdr 0

Chihhsiangchen, 2017-12-27:
Hi Brandon,

In the first, thanks for your driver. I got many help from it rather than Xilinx support....
The driver can work on Loopback environment, but I encounter some DMA decode error when I poring this driver on my own IP.
My IP using only one rx channel to receive data from ADC peripheral.
The interrupt will cause this error occurred and I didn't get any data.

Do you have any idea to help me to solve this error?

My device tree as below:
`    chosen {
        bootargs = "console=ttyPS0,115200 earlyprintk uio_pdrv_genirq.of_id=generic-uio cma=25M ";
        };  
        
    amba_pl: amba_pl {
        axi_dma_0: dma@40400000 {
            #dma-cells = <1>;                                                                                                                                                                                                                                                                                                 
            clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
            clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
            compatible = "xlnx,axi-dma-1.00.a";
            interrupt-parent = <&intc>;
            interrupts = <0 29 4>;
            reg = <0x40400000 0x10000>;
            xlnx,addrwidth = <0x20>;
            dma-channel@40400030 {
                compatible = "xlnx,axi-dma-s2mm-channel";
                dma-channels = <0x1>;
                interrupts = <0 29 4>;
                xlnx,datawidth = <0x40>;
                xlnx,device-id = <0x0>;
            };
        };
    axidma_chrdev: axidma_chrdev@0 {
        compatible = "xlnx,axidma-chrdev";
        dmas = <&axi_dma_0 0>;
        dma-names =  "rx_channel";
    };  
   };
`

Thanks
Stanley

Chihhsiangchen, 2017-12-28:
Hi Brandon,

I Added  more info for you.
1. I have verified OK in bare metal, but fail in Linux.
2. Linux dmesg , log and source code as below,

Please help me to check 

Thank you very much
Stanley

Linux dmesg and log.
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.0-xilinx-v2017.2 (hsiang@Linux) (gcc version 6.2.1 20161016 (Linaro GCC 6.2-2016.11) ) #1 SMP PREEMPT Fri Dec 15 18:32:39 CST 2017
[    0.000000] CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: xlnx,zynq-7000
[    0.000000] bootconsole [earlycon0] enabled
[    0.000000] cma: Reserved 28 MiB at 0x1e400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c0a32c00, node_mem_map ddfe9000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @ddfc1000 s25932 r8192 d23220 u57344
[    0.000000] pcpu-alloc: s25932 r8192 d23220 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyPS0,115200 earlyprintk uio_pdrv_genirq.of_id=generic-uio cma=25M 
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 475560K/524288K available (6144K kernel code, 206K rwdata, 1500K rodata, 1024K init, 233K bss, 20056K reserved, 28672K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0700000   (7136 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a33800   ( 206 kB)
[    0.000000]        .bss : 0xc0a33800 - 0xc0a6df18   ( 234 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=32, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] efuse mapped to e0800000
[    0.000000] slcr mapped to e0802000
[    0.000000] L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 8 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[    0.000000] zynq_clock_init: clkc starts at e0802100
[    0.000000] Zynq clock init
[    0.000010] sched_clock: 64 bits at 333MHz, resolution 3ns, wraps every 4398046511103ns
[    0.007860] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x4ce07af025, max_idle_ns: 440795209040 ns
[    0.018876] Switching to timer-based delay loop, resolution 3ns
[    0.024877] clocksource: ttc_clocksource: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 537538477 ns
[    0.033927] timer #0 at e080a000, irq=17
[    0.038296] Console: colour dummy device 80x30
[    0.042621] Calibrating delay loop (skipped), value calculated using timer frequency.. 666.66 BogoMIPS (lpj=3333333)
[... 296 lines cut ...]
    }

  printf("channel %d size: %d Buffer size:%d\r\n", rx_chans->data[0],rx_chans->len,BUFFER_SIZE);

  volatile int done = 0;
  axidma_set_callback(axidma_dev, rx_channel, dma_int_handler, (void*)&done);
  printf("Start Receive\n");

  int ret = axidma_oneway_transfer(axidma_dev, rx_channel, rx_buf, BUFFER_SIZE, false);
  while (!done) {
    ;
    // axidma_stop_transfer(hdma, MYCHAN, enum axidma_dir dir);
  }

  display_result(rx_buf);

free_rx_buf:

  axidma_free(axidma_dev, rx_buf, BUFFER_SIZE);

exit:

  axidma_destroy(axidma_dev);
}    
`

Chihhsiangchen, 2017-12-28:
```
#define BUFFER_SIZE (1920 * 1080 * 4)
#define TRANSFER_LEN 16

void dma_int_handler(int channel_id, void *data)
{
  if(channel_id != MYCHAN)
    printf("ERROR: unexpected channel id %d\n", channel_id);
  
  (*(int*)data) = 1; // done
}

int array_contains(int target, int *array, int length)
{
  int i;
  for (i=0; i<length; i++)
    if ( array[i] == target ) return 1;
  return 0;
}

void display_result(char *buffer)
{
  int i, *p = (int*)buffer;
  for(i=0; i<TRANSFER_LEN; i++) {
    printf("%08x ", *(p++));
    if (i % 8 == 7)
      printf("\n");
  }
}
```

void main()
{
    int rc; 
    int rx_channel;
    axidma_dev_t axidma_dev;
    const array_t *rx_chans;
    char *rx_buf;
    
    // Initialize the AXI DMA device
    axidma_dev = axidma_init();
    if (axidma_dev == NULL) {
        fprintf(stderr, "Failed to initialize the AXI DMA device.\n");
        rc = 1;
    }
    printf("axidma_benchmark: rx_size=%zu\n", BUFFER_SIZE);
    rx_buf = axidma_malloc(axidma_dev, BUFFER_SIZE);
    if (rx_buf == NULL) {
        perror("Unable to allocate receive buffer from the AXI DMA device");
        rc = -1;
        goto exit;
    }

    rx_chans = axidma_get_dma_rx(axidma_dev);
    if (rx_chans->len < 1) {
        fprintf(stderr, "Error: No receive channels were found.\n");
        rc = -ENODEV;
        goto free_rx_buf;
    }

  printf("channel %d size: %d Buffer size:%d\r\n", rx_chans->data[0],rx_chans->len,BUFFER_SIZE);

  volatile int done = 0;
  axidma_set_callback(axidma_dev, rx_channel, dma_int_handler, (void*)&done);
  printf("Start Receive\n");

  int ret = axidma_oneway_transfer(axidma_dev, rx_channel, rx_buf, BUFFER_SIZE, false);
  while (!done) {
    ;
    // axidma_stop_transfer(hdma, MYCHAN, enum axidma_dir dir);
  }

  display_result(rx_buf);

free_rx_buf:

  axidma_free(axidma_dev, rx_buf, BUFFER_SIZE);

exit:

  axidma_destroy(axidma_dev);
}    

#define BUFFER_SIZE (1920 * 1080 * 4)
#define TRANSFER_LEN 16

void dma_int_handler(int channel_id, void *data)
{
  if(channel_id != MYCHAN)
    printf("ERROR: unexpected channel id %d\n", channel_id);
  
  (*(int*)data) = 1; // done
}

int array_contains(int target, int *array, int length)
{
  int i;
  for (i=0; i<length; i++)
    if ( array[i] == target ) return 1;
  return 0;
}

 void display_result(char *buffer)
 {
   int i, *p = (int*)buffer;
   for(i=0; i<TRANSFER_LEN; i++) {
     printf("%08x ", *(p++));
    if (i % 8 == 7)
      printf("\n");
  }
}

void main()
{
    int rc; 
    int rx_channel;
    axidma_dev_t axidma_dev;
    const array_t *rx_chans;
    char *rx_buf;
    
    // Initialize the AXI DMA device
    axidma_dev = axidma_init();
    if (axidma_dev == NULL) {
        fprintf(stderr, "Failed to initialize the AXI DMA device.\n");
        rc = 1;
    }
    printf("axidma_benchmark: rx_size=%zu\n", BUFFER_SIZE);
    rx_buf = axidma_malloc(axidma_dev, BUFFER_SIZE);
    if (rx_buf == NULL) {
        perror("Unable to allocate receive buffer from the AXI DMA device");
        rc = -1;
        goto exit;
    }

    rx_chans = axidma_get_dma_rx(axidma_dev);
    if (rx_chans->len < 1) {
        fprintf(stderr, "Error: No receive channels were found.\n");
        rc = -ENODEV;
        goto free_rx_buf;
    }

  printf("channel %d size: %d Buffer size:%d\r\n", rx_chans->data[0],rx_chans->len,BUFFER_SIZE);

  volatile int done = 0;
  axidma_set_callback(axidma_dev, rx_channel, dma_int_handler, (void*)&done);
  printf("Start Receive\n");

  int ret = axidma_oneway_transfer(axidma_dev, rx_channel, rx_buf, BUFFER_SIZE, false);
  while (!done) {
    ;
    // axidma_stop_transfer(hdma, MYCHAN, enum axidma_dir dir);
  }

    display_result(rx_buf);

  free_rx_buf:

    axidma_free(axidma_dev, rx_buf, BUFFER_SIZE);

  exit:

    axidma_destroy(axidma_dev);
  }    

`

bperez77, 2017-12-30:
This is a perplexing issue. So the problem, as you know, is a DMA decode error, which means that the AXI DMA IP is receiving an invalid physical address when it is trying to perform the receive transfer. However, this (theoretically) should be impossible, because the driver explicitly checks that the address you specify was previously allocated by a call to `axidma_malloc`. On top of that, your app is pretty straightforward, and doesn't violate any of the rules.

So for some reason, the DMA address being used is invalid, but I've never seen this issue before. Is there anything new in the kernel log after you run your application? If so, can you post it? I'll need to look into this issue more.

bperez77, 2018-03-19:
Closing this issue due to inactivity.

closed 2018-03-19 by bperez77

zhangsd20, 2026-01-19:
/* request DMA channels */
        d->tx_chan = dma_request_chan(dev, "tx");
        if (IS_ERR(d->tx_chan) || d->tx_chan==NULL) {
                dev_err(dev, "failed to request tx dma channel\n");
                return PTR_ERR(d->tx_chan);
        }

        printk("tx: %p, rx: %p\n", d->tx_chan, d->rx_chan);
        /* allocate DMA coherent buffers */
        d->tx_buf_v = dma_alloc_coherent(dev, BUF_SIZE, &d->tx_buf_p, GFP_KERNEL);
        if (!d->tx_buf_v) {
                ret = -ENOMEM;
                goto err_release_dma;
        }

        tx = dmaengine_prep_slave_single(d->tx_chan, d->tx_buf_p, len,
                        DMA_MEM_TO_DEV, DMA_PREP_INTERRUPT | DMA_CTRL_ACK);
        if (!tx)
                return -EIO;

        tx->callback = tx_dma_callback;
        tx->callback_param = d;

        cookie = dmaengine_submit(tx);
        if (dma_submit_error(cookie)) {
                pr_err("tx submit failed\n");
                return -EIO;
        }
        dma_async_issue_pending(d->tx_chan);

        printk("===== tx_buf_v=%p tx_buf_p=%pad\n", d->tx_buf_v, &d->tx_buf_p);

I also encountered the same problem. Could you please tell me how to solve it?