bperez77/xilinx_axidma #35: app test

tanyuanyingjie, 2018-01-09:
Hello, I tested your axidma_benchmark application, the test is successful, you can provide me an application？I want to write an array of their own definition tx_buf, my hardware circuit is to short the two channels, This function is sent through axidma_twoway_transfer go read in rx_buf array I sent, the program written for a period of time can not be adjusted, beg you give a template program.

bperez77, 2018-01-10:
I don't have any specific application for you other than what is in the `examples` directory. Those act as the template program. If you give me some more details about what you're exactly trying to accomplish, then I can assist you further.

Also, if you're looking for documentation for the AXI DMA library, you can run:
```bash
make library_docs
```

In terms of creating your own array, it's fairly simple: just call `axidma_malloc` with the appropriate parameters. You can see what each one does in the documentation, or by reading over the example.

bperez77, 2018-01-22:
Closing this issue due to inactivity.

closed 2018-01-22 by bperez77