bperez77/xilinx_axidma #36: axidma_benchmark example fails with sizes of 16384 bytes or larger

baf2099, 2018-01-12:
@bperez77 First let me say thanks for your hard work on bridging the gap Xilinx left for us.

I'm experiencing an issue using your axidma_benchmark example in my system. I've played around with this quite a bit now and found that I can successfully run the example as long as the transmit and receive size remains under 16384 bytes, however once I attempt a transfer of 16384 bytes like so...
./axidma_benchmark -b 16384 -s 16384 the test fails resulting in the message "Failed to perform the AXI DMA read-write transfer: Timer expired"

Interestingly, once it's failed like this it doesn't seem to matter what size I use, it will fail. This seems to indicate that something in the DMA process is in a stuck state. 

Also interesting, if I just run the full 8MB test after a fresh linux boot, it doesn't result in a timeout, but rather most of the receive buffer is just unchanged as indicated by your message... 
"Warning: 99.95% of the receive buffer matches the initialization pattern.
This may mean that the receive buffer was not properly updated.
Single transfer test successfully completed!"

Here is some info on my setup...
Hardware: ZedBoard
Vivado Version: 2017.4
PetaLinux Version: 2017.3
Image of my block diagram attached
[image]

I use the PetaLinux system-user.dtsi file to add the necessary items to the device tree like this...
/include/ "system-conf.dtsi"
/ {
	axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
	    dma-names = "tx_channel", "rx_channel";
	};

	amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>;
			dma-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
		};
	};
};

Side questions that may provide some clue...
1. Does your driver care if the hardware DMA is setup for Scatter-Gather or not? I assumed it would work with or without this feature turned on.
2. Does your driver care about the datawidth of the DMA? As you can see in my case it's currently 32, but it could have been 64 or even as much as 1024 in AXI DMA v7.1

Any help on this issue would be greatly appreciated!

baf2099, 2018-01-16:
Some additional information on the issue. I've now added ILA's (Internal Logic Analyzer for those who don't know) on each of the DMA's interrupt lines as well as the AXI-Stream that is looped back. Here is what I observe...

./axidma_benchmark -b 1024 -s 1024, runs successfully and I can see both interrupts trigger as well as the AXI-Stream moving data

./axidma_benchmark -b 16384 -s 16384, fails with message "Failed to perform the AXI DMA read-write transfer: Timer expired", I do NOT see either interrupt trigger or a single beat of data on the AXI-Stream.

./axidma_benchmark, fails with message "Warning: 99.95% of the receive buffer matches the initialization pattern. This may mean that the receive buffer was not properly updated.", interestingly I DO see both interrupts trigger as well as the AXI-Stream moving data, but with 8MB of data to move I can only look at a small sampling of the AXI-Stream.

I'm not sure what conclusions can be drawn from this additional info, but it would seem that at least in some cases the DMA is not being activated at all, and in other cases it's being activated but maybe only for a small subset of the data?

bperez77, 2018-01-17:
I think you're running into a common gotcha with the driver. To start with your first two questions:

1. Yes and no. My driver is just a bridge between userspace and Xilinx's AXI DMA driver, so their driver handles the setup. Though my driver doesn't check for scatter-gather being enabled, it is implicit in certain functions. For example, you can't run any of the video transfer functions with more than one buffer if scatter-gather is not enabled.
2. No, the data width has no influence on my driver, that is all handled by your FPGA IP and the Xilinx AXI DMA driver.

Back to your original issue, I think the problem lies with the value of the **Width of Buffer Length Register** parameter. See page 76 of the [AXI DMA v7.1 Product Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf) for more details. This controls what the maximum length of the buffer any given single DMA transaction or scatter-gather entry can send or receive. By default it is set to 14 bits, so the maximum length is 2^14 = 16383, which explains the behavior you're seeing.

I recommend setting it to its maximum value of 23, so you can send up to 8 MiB of data in each transfer. If you don't, you won't be able to run the benchmark with the default command-line values. As to why you don't get a timeout for that case, I've seen it before, so I'd chalk it up to undefined behavior in the AXI DMA IP.

baf2099, 2018-01-17:
I actually figured out that my assumption about using the default DMA IP settings was bad just a few hours before you posted this, but I'm glad to see you confirm that conclusion. I'm happy to say that after updating this to its max value of 23, re-building the bit file, and packaging it up with my PetaLinux project I can now successfully run the axidma_benchmark test with any setting and get reasonable results.

In regards to your scatter-gather statement that "My driver is just a bridge between userspace and Xilinx's AXI DMA driver, so their driver handles the setup" I notice that in your driver, specifically in axidma_dma.c you have code like this in the read, write, and read-write functions...

```
// Setup the scatter-gather list for the transfer (only one entry)
sg_init_table(&sg_list, 1);
c = axidma_init_sg_entry(dev, &sg_list, 0, trans->buf, trans->buf_len);
```

looking at the sg_init_table function in scatterlist.c I see that...

```
//sg_init_table - Initialize SG table
//@sgl:	   The SG table
//@nents:	   Number of entries in table
void sg_init_table(struct scatterlist *sgl, unsigned int nents)
```

This seems to indicate that your base read, write, and read-write functions can ONLY use a single entry in a scatterlist, regardless of whether the DMA IP has scatter-gather enabled, but maybe I'm just misinterpreting how this works. Can you shed some light on what this means? If I did enable scatter-gather in the DMA IP what would actually happen?

bperez77, 2018-01-22:
So, this actually has to do more with how the Linux kernel handles DMA transactiosn with drivers than anything specific to Xilinx's DMA engine. The `sg_init_table` function does relate to scatter-gather DMAs, but it can be used even when SG mode is not enabled. Naturally the SG table can only have one entry in this case.

Interestingly, my driver is actually mostly independent of the Xilinx DMA driver. The only information it needs to know is about the VDMA configuration structure, which is required to be set before VDMA transactions. Otherwise, the code goes through the Linux kernel's generic DMA code layer, which separates users of the DMA interface from the specific drivers that implement the backends. For this, it's standard to use SG interface to setup DMA transactions, but these are compatible with simple DMA engines, that lack SG support.

If you did enable scatter-gather on the AXI DMA IP, the driver should still function normally.

bperez77, 2018-01-22:
There's more information [here](https://github.com/Xilinx/linux-xlnx/blob/master/Documentation/dmaengine/client.txt) if you're curious.

closed 2018-01-22 by bperez77