bperez77/xilinx_axidma #37: driver/axidma_chrdev.c:275:5: error: implicit declaration of function 'of_dma_configure'

w66673733, 2018-01-22:
function of_dma_configure() can not find

bperez77, 2018-01-22:
What version of Linux does Linaro 12.11 use? It has to be a Linux 4.x version in order to compatible with the driver.

w66673733, 2018-01-23:
The linux version is 4.0.0 the problem will occur.  when I use 4.6.0 it seems work,.
However, after I insmod axidma.ko, /dev/axidma is not created.

closed 2018-01-23 by w66673733

bperez77, 2018-01-26:
Interesting, yeah some vendors don't backport all the changes from 4.x versions, so you can occasionally run into issues like this.

What's the output of `dmesg` immediately after you try to `insmod` the driver. If the device is not created, it is likely because there was a failure in the driver. This could be to a wide variety of problems.