bperez77/xilinx_axidma #38: Running multiple DMA IPs

baf2099, 2018-01-23:
Can your driver be attached to and run multiple DMA IPs (concurrently or one at a time)? If not do you have any plans to extend the driver to support multiple DMAs?

bperez77, 2018-01-26:
Yes, if I'm understanding your question correctly, it's easy to attach to multiple DMA/VDMA IP blocks. You just need to make sure you have device tree entries for them, and reference them in the AXI DMA device tree node. You can get all the available DMA channels from you user application using the `axidma_get_*` library functions.

baf2099, 2018-01-26:
So I tried to attach 2 DMA's to your driver in the device tree but at first it gave me errors as device ID can only be 0 or 1 and I thought they needed to be unique, and then it seemed to only be able to register one transmit and receive channel, unless I'm just doing something incorrectly? 

```
{
	axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&axi_dma_0 0 &axi_dma_0 1 &axi_dma_1 0>;
	    dma-names = "tx_channel", "rx_channel", "rx_ch_two";
};
```
Further digging and I noticed that there is a pull request on this repo from @corna here...

[https://github.com/bperez77/xilinx_axidma/pull/32](url)

and one of the commits is for "Add support for multiple AXI DMA devices" here...

[https://github.com/corna/xilinx_axidma/commit/d5a23981e6f51cc55843ab156c7420757ec5ee9d](url)

which seems to extend the device tree to support multiple DMA IP instances. Is this needed to run multiple DMAs at all or does it just allow me to run multiple DMA's simultaneously? Do you know if these code changes are stable? Any plans to mainline these extensions?

bperez77, 2018-01-26:
What errors are you getting when you try to make the device ID something other than 0 or 1? You should be able to make the device ID any arbitrary number (unless something has changed in the kernel unbeknownst to me). 

For the `dmas` property, the 0 and 1 do not refer to the device ID of the channel. Instead, that refers to the index of the child within that device node entry. So, for `axi_dma_0`, you have two children with that node, and for `axi_dma_1` you have one. If you changed the device ID of the node within `axi_dma_1`, then you would still refer to it with `&axi_dma_1 0`.

If you could post the error line(s) from `dmesg` you're getting when the device ID is something other than 0 or 1, that would appreciated. I suspect something may have changed that I wasn't aware of, or it may be something malformatted in the device tree. Could you also attach you device tree entries as a file as well?

baf2099, 2018-01-29:
Good news & Bad news. Bad news is that I can't replicate the issue I was having so I must have had a typo, or some silly issue last time around. Good news is that I now have 2 DMA's working with your driver, but I will caveat that with the fact that I have not attempted to run them concurrently yet to see if the driver has any issues with the current kernel.

bperez77, 2018-01-30:
Got it, that makes sense. Let me know if you run into any other issues.

bperez77, 2018-03-19:
Closing this issue due to inactivity.

closed 2018-03-19 by bperez77

zhangzilin, 2018-12-13:
> Good news & Bad news. Bad news is that I can't replicate the issue I was having so I must have had a typo, or some silly issue last time around. Good news is that I now have 2 DMA's working with [...]

I am using two DMA too, using only the rx channel of each dma. My device tree is:

amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axidma_chrdev: axidma_chrdev@0 {
    			compatible = "xlnx,axidma-chrdev"
    			dmas = <&axi_dma_0 0 &axi_dma_1 1>;
    			dma-names = "rx_channel", "rx_channel_1";
			};

		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,include-sg ;
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
		};
		axi_dma_1: dma@40410000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 30 4>;
			reg = <0x40410000 0x10000>;
			xlnx,addrwidth = <0x20>;
			xlnx,include-sg ;
			dma-channel@40410030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
		};

I encountered this error while loading the driver：

Unable to handle kernel NULL pointer dereference at virtual address 00000000
pgd = ee808000
[00000000] *pgd=2f0cd831, *pte=00000000, *ppte=00000000
Internal error: Oops - BUG: 17 [#1] PREEMPT SMP ARM
Modules linked in: axidma(O+)
CPU: 0 PID: 736 Comm: insmod Tainted: G           O    4.14.0-xilinx #1
Hardware name: Xilinx Zynq Platform
task: ef0eb9c0 task.stack: ea8a6000
PC is at axidma_of_parse_dma_nodes+0x1dc/0x414 [axidma]
LR is at of_get_next_child+0x38/0x44

I hope to get your help！

Thanks.