bperez77/xilinx_axidma #39: AXI DMA not found

hawkroot-ua, 2018-02-18:
Hello, 
I'm trying to install xilinx_axidma driver. For drive coppile I using petalinux with 4.9.0-xilinx-v2017.4 kernel version. After system build I get only messages like these:

dma-pl330 f8003000.dmac: Loaded driver for PL330 DMAC-241330
dma-pl330 f8003000.dmac:        DBUFF-128x8bytes Num_Chans-8 Num_Peri-4 Num_Events-16
xilinx-vdma 40400000.dma: Xilinx AXI DMA Engine Driver Probed!!
xilinx_axidma: loading out-of-tree module taints kernel.

#lsmod
    Tainted: G
xilinx_axidma 19976 0 - Live 0x7f031000 (O)

In /proc/devices I not see axi_dma

My device-tree:

`	amba_pl {
		#address-cells = <0x1>;
		#size-cells = <0x1>;
		compatible = "simple-bus";
		ranges;
		
		axi_dma_0: dma@40400000 {
			#dma-cells = <0x1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <0x1 0xf 0x1 0xf 0x1 0xf 0x1 0xf>;
			compatible = "xlnx,axi-dma", "xlnx,axi-dma-6.03.a", "xlnx,axi-dma-1.00.a";
			interrupt-parent = <0x4>;
			interrupts = <0x0 0x1d 0x4 0x0 0x1e 0x4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x40>;
			xlnx,include-sg;

			dma-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0x0 0x1d 0x4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};

			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0x0 0x1e 0x4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
		};
	};

	axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
	    dma-names = "tx_channel", "rx_channel";
	};`

bperez77, 2018-02-28:
Apologies for the late reply. Since it's appearing with `lsmod`, it seems like the module was inserted correctly into the kernel. You should see the device at `/dev/axidma`, though I believe is should also show up in `/proc/devices`. 

What happens when you try to run the application? Does it simply fail because it cannot find `/dev/axidma`?

hawkroot-ua, 2018-02-28:
Thank you for response. The problem in device tree, it was not correct.

bperez77, 2018-03-04:
Got it, what was the issue exactly? Just in case anyone else encounters a similar problem.

russwinsper, 2018-03-15:
Brandon,
I'm a little late to the party and trying to catch up quickly but running into a similar problem. My basic question is, do you have and willing to share the Vivado project this was built against to help speed up my learning curve.?

bperez77, 2018-03-19:
Thanks @hawkroot-ua.

closed 2018-03-19 by bperez77

bperez77, 2018-03-19:
@russwinsper, unfortunately, I only have an old Vivado 2015.2 loopback design that I've been working with. I can share that with you. Otherwise, this is all you need to do to create your own loopback design for testing:
  1. Instantiate an AXI DMA IP, and run the automatic connections for it.
  2. In the properties, make sure both the read and write channels are enabled, enable the scatter-gather engine, and make sure unaligned transfers are allowed for both channels.
  3. Set the width of the buffer length register to its maximum value, 23.
  4. Connect the read channel to the write channel, specifically connecting the port `M_AXIS_MM2S` to `S_AXIS_S2MM`.

russwinsper, 2018-03-27:
Brandon,
I'm quite a bit closer but now running into "axidma: axidma_dma.c: axidma_request_channels: 649: Unable to get slave channel 0: tx_channel." Here's my device tree node:

axidma_chrdev: axidma_chrdev@0 {
   		compatible = "xlnx,axidma-chrdev";
        	dmas = <&axi_dma_0 0 &axi_dma_0 1>;
       		dma-names = "tx_channel", "rx_channel";
    };

    axi_dma_0: axidma0@a0000000 {
            #dma-cells = <1>;
            compatible = "xlnx,axi-dma", "xlnx,axi-dma-6.03.a", "xlnx,axi-dma-1.00.a";
        reg = <0x0 0xa0000000 0x0 0x10000>;
            clocks = <0x42>, <0x42>, <0x42>, <0x42>;
            clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
            xlnx,include-sg;    
            xlnx,addrwidth = <0x20>;

        dma-mm2s-channel@a0000000 {
            compatible = "xlnx,axi-dma-mm2s-channel";
            dma-channels = <1>;
            xlnx,datawidth = <64>;
            xlnx,device-id = <0>;
            interrupt-parent = <0x4>;
            interrupts = <0 29 4>;
        };

        dma-s2mm-channel@a0000030 {
            compatible = "xlnx,axi-dma-s2mm-channel";
            dma-channels = <1>;
            xlnx,datawidth = <64>;
            xlnx,device-id = <1>;
            interrupt-parent = <0x4>;
            interrupts = <0 30 4>;
        };
    };

Any suggestions would be greatly appreciated.

bperez77, 2018-03-29:
This could be potentially one of two issues. The first (and most likely issue) is that the AXI DMA device tree entries aren't being loaded by the Xilinx driver, so this driver cannot find the DMA channel. The other possibility is that the channel is being claimed by another driver, though I don't think this is the case.

Can you attach the output of the `dmesg` command immediately after you `insmod` the driver? Please attach it as a file.