bperez77/xilinx_axidma #40: VDMA: axidma_prep_transfer only configures first frame buffer address

elektrokokke, 2018-02-21:
When using the driver with a VDMA core with more than one framebuffer (e.g., with xlnx,num-fstores = <0x2>), the frame buffer addresses beyond the first are not configured.

This is the lines which seem to configure the first frame buffer address (from axidma_prep_transfer in axidma_dma.c):
```C
        dma_template.dst_start = sg_dma_address(&sg_list[0]);
        dma_template.src_start = sg_dma_address(&sg_list[0]);
```

There is no reference to the other entries of the sg_list.

The VDMA core starts running nonetheless, but frame buffer addresses beyond the first are set to 0x00000000. I don't know what the implications of this are.

Also I don't know the Xilinx VDMA driver well enough... would it work to replace the call to `dmaengine_prep_interleaved_dma` with a call to `dmaengine_prep_slave_sg`, like in the non-VDMA case?

elektrokokke, 2018-02-21:
I have to correct my previous statement: the VDMA core stops with errors if it expects more than one frame buffer.

I still had an old bitstream with only one frame buffer running on the FPGA while testing the driver with multiple frames...

bperez77, 2018-02-28:
Oh I see. Yeah it can be tricky to keep of these disparate parts of the system in sync with one another. So, is this issue resolved, or what error are you seeing now?

elektrokokke, 2018-02-28:
I opted for using only one frame buffer for now, as I do not understand the linux DMA engine code well enough yet for working out a fix...
With only one frame buffer it runs like a charm, however, in the end I will need multiple frame buffers for asynchronous read and write on the frame buffers (I intend to have a process in the PS copy the frame to a user space buffer for further processing).

bperez77, 2018-03-05:
Actually I think I might have found the issue. Your first comment is close, but not quite there. The `src_start` and `dst_start` are only needed for the first frame of the DMA; the rest are meant to be determined by the scatter-gather list. Looking at lines [219-229 of axidma_dma.c](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_dma.c#L219), I believe the issue is that I'm not setting the `src_inc`, `dest_inc`, `src_sgl`, `dst_sgl` fields to true.

According to [lines 141-170 of linux/include/dmaengine.h](https://github.com/Xilinx/linux-xlnx/blob/master/include/linux/dmaengine.h#L141) (from the mainline Xilinx Linux fork), the `src_sgl` and `dest_sgl` fields are used to specify that a scatter-gather list is used. On top of that, it looks a different datatype is used for the scatter-gather list, which is a `struct data_chunk`, instead of a `struct scatterlist`.

I'm guessing this likely happened because I didn't stay up to date with the kernel changes.

bperez77, 2018-03-05:
This may be a bit more involved to fix. I think I had a bit of conceptual misunderstanding with interleaved DMA transfers; they don't seem to be used for these scatter-gather style transfers.

Yuxin-Yu, 2024-12-02:
Hi everyone @bperez77 @elektrokokke .Have you fix this bug,I met the same problem