bperez77/xilinx_axidma #41: DMA Completion Status

baf2099, 2018-02-28:
I've come to learn that there are one of two ways an interrupt will be triggered when DMAing data from the PL to the PS...

Case 1. Data stream hits a TLAST signal and generates an interrupt based on EOF (End of Frame)

Case 2. Number of bytes requested to be transferred is reached and generates an interrupt based on IOC (Interrupt on Complete)

Optimally TLAST would always be concurrent with the last byte of the transfer that was requested so an interrupt would show both EOF and IOC if I am understanding correctly

There are some xilinx forum discussions on the subject such as this one for reference
[https://forums.xilinx.com/t5/Embedded-Processor-System-Design/Axi-DMA-S2MM-Is-it-necessary-to-assert-TLAST/td-p/743336](url)

What I would like to do is validate why we got an interrupt and ensure that a transaction got all the data that was requested. Maybe your driver does this already in some fashion, but how would I go about checking the status of a DMA transaction complete interrupt?

bperez77, 2018-03-04:
There's no explicit check in the my driver for this. However, the Xilinx DMA driver will generate a message in the kernel log buffer if there are any errors registered in the AXI DMA DMASR register. One of these errors is the `DMAIntErr`, which is bit 4 of the register. This is set when either TLAST is asserted before the expected end of the transfer, or if the length of one of the scatter-gather descriptors is 0 (see page 28 of the [AXI DMA v7.1 User Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf)).

So if you check the output of `dmesg` you'll be able to see if this error case happened. However, it may be worth adding this into the driver as well.

baf2099, 2018-03-05:
I think this would be very useful to people not only for debugging but also for normal use when it's imperative that each transaction can be verified. I'd be willing to help develop this into your driver if you could point me in the right direction. 

It seems that at the FPGA level the data we want is stored between the AXI_DMA_DMASR regsiter and at least in the case of a PL to PS transfer, the S2MM_STATUS register. The problem may be however that the Xilinx driver reads/clears these registers and then stores them elsewhere in memory. I've poked around in the Xilinx driver and while I do see the AXI_DMA_DMASR register defined/used I strangely don't see the S2MM_STATUS register being defined and equally strange the AXI DMA User Guide doesn't even define a register value for it.

Looking at the xilinx_dma.c line 1447 I see the IRQ handler function...
`static irqreturn_t xilinx_dma_irq_handler(int irq, void *data)`
...this seems to be the moment when the xilinx driver reads the DMASR register on line 1453...
`status = dma_ctrl_read(chan, XILINX_DMA_REG_DMASR);`
...and subsequently will write back to that register, looking to clear/correct errors. I don't however see how that original status value that was captured makes it to any readable variable/struct from your driver. Maybe there is another way, or place to capture this info and report it back through your driver to the user space?

Ultimately I think the goal should be to know the exact circumstances on which a transaction ended.
1. Did the transaction end on a TLAST, also known as the EOF (end of frame)
2. Did the transaction end due to satisfying the transfer size requested, also known as IOC (interrupt on complete)
3. How many bytes were actually transferred. This is important in the case that a EOF occurred before an IOC could, meaning that we request X bytes for a transfer, but only received x-n bytes due to receiving TLAST earlier then expected.
4. How many bytes are held up in the DMA. This is important in the case that a IOC occurred before an EOF could, meaning that we requested x bytes for a transfer, and recieved them, but there are still more bytes to be transferred before reaching TLAST. This one might be tough, but there is possible tie in here to the work being done by @corna with the axidma_get_residue() function

In a correctly configured system, with a successful DMA transaction I would expect to see both the IOC and EOF bits set every time, as well as the exact number of bytes I requested having been transferred into a buffer.

baf2099, 2018-03-15:
Hey @bperez77 just curious if you've had a chance to look over my comment/idea and if you have any insight for me?

bperez77, 2018-03-19:
Apologies, I did send a reply a while back, but for some reason it never went through.

In regards to the `S2MM_STATUS` register, I don't believe the driver uses this at all. As far I can tell from the user guide, this register is at the granualirty of individual scatter-gather descriptors within a larger transfer. The Xilinx driver only cares about the end result of a transaction, so it only utilizes the `DMASR` register. I believe the reason that the user guide doesn't define an address for the register is that it is not exposed as an MMIO register. Perhaps it is still viewable from signal tap, so that is why it is mentioned?

I could definitely use the help with the driver. A big reason why the development on the driver has been so slow is because I don't have access to an FPGA board at the moment. So I could definitely use the help.

I definitely agree, it would be very useful to have this sort of status for the end-user, and especially to have it returned to the user via the callback function. I think all of your goals are good and satificatory to get this information to the user.

Ultimately, the main issue I think comes from the fact that the Xilinx driver does not expose this information at all. The `DMASR` is read into a local varaible, and then the register is reset. This local variable is used to check the status, but then is lost once the IRQ handler returns. Thus, I think this leaves us with two options.

The first is to hijack the interrupts used for the AXI DMA IPs. Then, when an interrupt occurs, the DMASR register can be read and stored somewhere, and then the normal Xilinx interrupt handler can be invoked. This is of course a pretty hacky solution. Additionally, since the IRQ handler is declared as static by the Xilinx driver, this would either need to be removed with a patch, or the Makefile would have to pass the address of this function to the driver when it is compiled (or it can be read at runtime, actually). 

The much better and more maintainable solution is to work with the Xilinx kernel developers to get this information exposed to the callback function. Unfortunately, because of how the Linux DMA interface is structured, I don't think there's any way for them to pass the status register back to the callback directly. However, they should be able to pass the residue to the callback. This would require them to not pass `NULL` to `dmaengine_desc_callback_invoke` on [line 825](https://github.com/Xilinx/linux-xlnx/blob/master/drivers/dma/xilinx/xilinx_dma.c#L825), and instead pass a proper `dmaengine_result` structure with the residue. 

The other option for them would be to add a new `status` field to the `xilinx_dma_chan` structure. This field would store the value of the `DMASR` register from the last interrupt for the channel. Then, they would also need to add some sort of interface to query the saved `DMASR` register values by channel ID.