bperez77/xilinx_axidma #42: function "of_dma_configure" undefined

sk2046, 2018-03-05:
when i use: petalinux-build,error,the log say function "of_dma_configure" undefined.i use petalinux 2015.4. Thanks.

sk2046, 2018-03-05:
Hi,and where the function "of_dma_configure" is defined?

bperez77, 2018-03-05:
That function is defined on [line 88 of drivers/of/device.c](https://github.com/Xilinx/linux-xlnx/blob/master/drivers/of/device.c#L88). Note, that this is on the latest version of the Xilinx fork of Linux.

I'm not sure what version of Linux PetaLinux 2015.4 uses, but since you cannot find `of_dma_configure`, it is most likely a 3.x version of Linux. The driver only supports 4.x+ versions of Linux. The other possibly is that is a 4.x version, but an old enough version that they didn't yet backport all of the new Linux kenrel's DMA inteface.

In either case, I would recommend upgrading to a newer version of PetaLinux, unless you have a good reason for using 2015.4.

Issue #37 is a similar issue.

sk2046, 2018-03-06:
Thank you very much!I will try again.

bperez77, 2018-03-19:
Closing this issue due to inactivity.

closed 2018-03-19 by bperez77