bperez77/xilinx_axidma #43: DMA receive transaction timed out

sk2046, 2018-03-10:
Hi Brandon,

I am trying to install your driver for using with DMA on petalinux 2017.4 and run into some problems.
Here is what I have done:
```
root@zynq_petalinux:/lib/modules/4.9.0-xilinx-v2017.4/extra# insmod ./xilinx-axidma.ko 
xilinx_axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 706: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 708: VDMA: Found 0 transmit channels and 0 receive channels.
```
It is OK.
```
root@zynq_petalinux:/usr/bin# ./axidma_benchmark -i 1 -o 1
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 1.00 Mb
        Receive Buffer Size: 1.00 Mb
        Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
axidma: axidma_dma.c: axidma_start_transfer: 298: DMA receive transaction timed out.`
```
What's the problem?Thank you.

sk2046, 2018-03-10:
Sometimes the error like this:
[https://forums.xilinx.com/t5/Embedded-Linux/ZYNQ-7010-AXI-DMA-driver-Fail-TIMEOUT/td-p/833457](url)

sk2046, 2018-03-10:
When i use DMA with SG mode,the problem has been solved.

bperez77, 2018-03-19:
Interesting, I'm going to have to make a note of this. The driver should work regardless of whether or not it's in SG mode. Something may have changed in the Xilinx driver. I'm going to close this issue since it's resolved.

closed 2018-03-19 by bperez77