bperez77/xilinx_axidma #44: Transactions failures when a DDR PL is instanciated

user-moduleus, 2018-03-15:
Hi,

Fisrt of all, many thanks for sharing your driver. I have got it working pretty good with your loop-back example and with my particular Vivado design. In this last, I have a flow of data which comes in a FIFO and which is read back from this last thanks to a DMA.

Furthermore, because of my application case need a large amoung of memory, I use a external DDR memory that I control in the PL thanks to the MIG IP (like in a basic design design which is provided by Xilinx for their Mini-itx evaluation board for example - http://microzed.org/design/2056/17).

But, when this DDR PL is instanciated in my PetaLinux project, I get the following error from the driver: "xilinx-vdma 40400000.dma: Channel df035910 has errors 40, cdr 0 tdr 0" and the transaction doesn't occur. 

I used a trick to 'overcome' this issue but it isn't satisfy as I disable the MIG device directly in the device tree:
&PLDDR3_IP_MIG_PL_DDR3 {
	reg = <0x80000000 0x00000000>;
};
Linux doesn't see any more this external DDR so the transaction doesn't occur into it, but this external memory becomes useless...

Have you any suggestion?

bperez77, 2018-03-19:
Just to clarify, your design is an AXI DMA IP connected to the MIG IP (which is in turn connected to the extrenal DDR)? Or is the MIG and AXI DMA independent from one another?

The line you see from the driver indicates that the AXI DMA IP has hit a `DMAIntErr`. If this is a TX channel, then this error indicates that one of the scatter-gather buffer descriptors had its length set to 0. Otherwise, for an RX channel, this error can also occur if the amount of data received is less than what was specified by the length. You can find more information about these errors on pages 16 and 27, respectively, in the [AXI DMA v7.1 User Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf).

It's hard to say what the issue is. Have you independently verified that the MIG IP is working? If not, I would try that.

Otherwise, can you send me the following information to help in debugging?
  1. A PDF or picture of the block diagram for your design.
  2. The relevant device tree entries for the AXI DMA IP and my driver (please only incude these parts).
  3. The output of `dmesg` after you run your application.

user-moduleus, 2018-03-19:
Thanks for your answer.

To be honest, I am getting confused about which Vivado design is the best suitable for my application case. In fact, I have 0.5 GB of DDR PS, and 1 GB 'DDR PL' (controlled by MIG through the Zynq GP port). And I need to transfert the largest possible among of data from a ADC to the memory. 

For the moment, I transfert data to the DDR PS using a DMA with the following desgin:

-- current_design --
[image]

The design is not entirely here but, basically, you have an acquisition design which fill a fifo and this last one is read back thanks to a DMA. The 'DDR PL' is connected to the Zynq with a GP port so it is used by Linux just as storage media. The MIG is well configured as I can read/write data to it and over 1GB (tested with a device tree where Linux doesn't see the 'DDR PL' and a c program which mmap the memory to read/write into it). 

To ensure that the buffer allocated by the driver will lie in the DDR PS, I constrain the CMA using the Linux boot arg cma=25M@0x0. Otherwize (just with cma=25M), I got the error "xilinx-vdma 40400000.dma: Channel df035910 has errors 40, cdr 0 tdr 0" as the driver is trying to perform the transfer with a buffer allocated in the 'DDR PL'.

What do I have to do if I want to perform the transfer into the DDR PL? I think that I have to rework the Vivado design (connecting the MIG to a DMA). Does the following design is correct for that purpose?

-- example_design --
[image]

This picture is an example design (released for another board). There is two DMA IP with loopbacks, and this is 2nd one which is connected to the DDR PL. Of course, the loopback DMA connection will be replaced by my own acquisition design.

If it is, I don't understand how the driver memory allocation (or CMA) can occur in the 'DDR PL' as the MIG IP is not mapped into the Zynq PS memory. And, after the transaction, how can I work on those data from my linux c program.

bperez77, 2018-03-21:
Disclaimer: I have never worked with the MIG or any external DRAM with Xilinx FPGAs, so take my statements with respect to those with a grain of salt.

In the **curent_design**, everthing seems to be hooked OK. Since you are using a receive/S2MM DMA channel, the error is likely occuring because the amount of data being received is less than what was specified to the AXI DMA transfer function you invoked from your C application. This is the source of this error the overwhelming majority of the time. Now, this error can be ignored if you are not specifying the correct size for the transfer. However, it sounds like that's not the case here.

I'm quite confused in your **current_design** why adding the MIG would cause this error to suddenly appear. The MIG design is completely independent from the AXI DMA path, so I don't see how it would affect the AXI DMA path.

From the point of view of Linux, if you put the external DDR in address space, it just views it as part of a "contiguous" virtual address space. Linux is aware that this memory is external, but the allocator and Xilinx AXI DMA driver are not. There should be a way to restrict where DMA buffers can be allocated.

bperez77, 2018-03-21:
In terms of your end application, do you need the extra memory space for the FPGA IPs, your C application, or both? If you only need it for the FPGA IPs, then I would recommend keeping the extrenal DRAM connected only to the PL fabric. Otherwise, the best way to utilize it will depend on the specifics of your application.

For your **example_design**, I believe that would be the way to connect it. However, you would run into some issues if you tried to use the driver with that design. With that design, your PS system could setup AXI DMA transfers from the external DDR to the PL, and the PL back to external DDR. If you want setup transfers between the PL DRR and PS DDR, you would need two AXI DMA IPs: one for PL DDR to PS DDR and another for PS DDR to PL DDR.

In either case, there's one big issue you'll run into if you're trying to use the driver. The driver implicitly assumes that the addresses you're working with are all in the processor's memory space. Thus, it cannot handle addresses for external memories that are not connected to the PS, as in your **example_design**.

All in all, I think this leaves you with several potential options:
  1. Use the PL DDR only with IP, and directly interface with the MIG or AXI DMA in your hardware design.
  2. Connect the PL DDR to both the PS and PL. I believe this should be possible with AXI Interconnect or a similar IP, but honestly, I've never used them outside of block automation.
  3. Update the driver to allow for transfers to "external" address, which are ones that are not present in the processor's memory space.

If only the PL really needs the external DDR, option 1 is probably best. Otherwise, option 2 is probably your best bet.

On a side note, you'll probably want to connect the MIG to one of the HP ports. The GP ports have significatntly lower bandwidth, and are more intended for control signals and writes to MMIO registers.

user-moduleus, 2018-03-21:
Thanks for your answers.

For the 1st one, last sentence, I think that the way to restrict the memory position of DMA buffers allocation is in the Linux boot argument "cma": < cma=25M@0x0 >. In fact, 0x0 is the start address of the DDR PS.

For the 2nd one, yeah I need to both put the data from PL to the DDR PL and to rework onto those previous data into my C program. 

From your potential option, I feel that my best bet is the no 2. I need to investigate some testing designs. Below is some drafts of them.

The 1st one is a simple DMA loop-back connected to the DDR PL. Furthermore, the connection with the GP allows the Zynq PS to have a memory map range of this DDR PL. This design compiled (throught the bitstream) but I cannot really use it. Therefore, I will move to the 2nd one where my acquisition design will be emulated by a counter and the data will be pushed into the DDR PL. From my Linux C program, as the DDR PL is mapped, NORMALLY, I should be able to rework onto those data.

Do you see any issue this 2nd second design? As I didn't see any example design for this kind of Vivado project, I need to 'improvise' a little bit...

For the moment, I think my final purpose will be the 3rd drawing below. My acquisiton data will fall into the DDR PL and the DDR PS can also gather some of them to reduce the load. So, a 2nd DMA DDR PL to DDR PS has to be used.

From your final side note, I don't know how the MIG with its axi slave port can be connected the HP port (also a slave port). One more time, I parsed the web about that issue but I never find something else than the GP port. Do you know how? Because, yeah, it seems that the DDR PL accesses performances will be affected.

--- desing1 ----
[image]

--- desing2 ----
[image]

--- desing3 ----
[image]

bperez77, 2018-03-22:
I'm going to my responses inline with yours so it's easier to track the threads.

> For the 1st one, last sentence, I think that the way to restrict the memory position of DMA buffers allocation is in the Linux boot argument "cma": < cma=25M@0x0 >. In fact, 0x0 is the start address of the DDR PS.

I wasn't sure if the address constraint on the `cma` command line argument was a physical or virtual address. Double checking the documentation, it is indeed a physical address, so that workaround should be sufficient.

>Do you see any issue this 2nd second design? As I didn't see any example design for this kind of Vivado project, I need to 'improvise' a little bit...

Improvising with Xilinx stuff always leads to interesting results. In all seriousness, I don't see any major issues with the second design. As I mentioned before, I have no experience with external DDR on Xilinx, so take what I'm saying with a grain of salt. 

The only small issue I see is that the only path from your acquisition IP to the PS DDR is through the external DDR, but that seems to be handled by your 3rd design.

> For the moment, I think my final purpose will be the 3rd drawing below. My acquisiton data will fall into the DDR PL and the DDR PS can also gather some of them to reduce the load. So, a 2nd DMA DDR PL to DDR PS has to be used.

I'm not sure that your third design as pictured is possible with the AXI DMA IP. I don't think you could drive the `M_AXIS_MM2S` port like that, and there's only one S2MM port in a single IP. Some alternatives are to use multichannel DMA (though this is not supported by my driver at the moment), or have your acquistion IP be responsible for deciding where its output data goes.

> From your final side note, I don't know how the MIG with its axi slave port can be connected the HP port (also a slave port). One more time, I parsed the web about that issue but I never find [...]

Good point, I forgot that HP ports can only act as slaves. There's got to be some way to do it, but I'm not really sure how so. The only thing I could think of is if you could find a way to interface the AXI DMA IP with the MIG, then you could use the AXI DMA IP to drive transfers to the external DDR. Of course this kind of puts you back to square one, so that's probably not the right way to do it.

user-moduleus, 2018-03-23:
I'm also going to put my responses inline with yours.

> I wasn't sure if the address constraint on the cma command line argument was a physical or virtual address. Double checking the documentation, it is indeed a physical address, so that workaround should be sufficient.

From http://redsymbol.net/linux-kernel-boot-parameters/, it seems to be a physical address. But, as I am a Linux embedded beginner, maybe I miss something.

> I'm not sure that your third design as pictured is possible with the AXI DMA IP. I don't think you could drive the M_AXIS_MM2S port like that, and there's only one S2MM port in a single IP. Some [...]

Yes, this drawing is wrong and a solution can be by using a MUX to target either the DDR PS or the DDR PL. This is represented by the following drawing.  But, I'm not sure about the compatibility between the MIG S_AXI interface and the S_AXI_HP0 pinterface. It seems that the HP one uses more control signal than the other.

[image]

Have you planned to implement the multichannel support?

--

Now, back to the design2 from my previous comment, I have a few questions. First, from your driver and library, I understood that DMA buffer is allocate by the axidma_malloc function which calls your mmap character driver function. On another hand, in the design, the DDR PL is mapped at the physical address 0x80000000. How can I specify this address to your driver? Is it done at boot time by the device tree? My device tree node is: 
	axidma_chrdev: axidma_chrdev@0 {
		compatible = "xlnx,axidma-chrdev";
		dmas = <&axi_dma_0 0>;
		dma-names = "rx_channel";
	};

Does it seems that if I specify it like:
	axidma_chrdev: axidma_chrdev@0x80000000 {
		compatible = "xlnx,axidma-chrdev";
		dmas = <&axi_dma_0 0>;
		dma-names = "rx_channel";
	};
The buffer will be allocated in the DDR PL?

Or, do I have to use the axidma_register_buffer function? I see that for using this function, I have to write a little driver which will perform the allocation and pass it to your driver. I'm not very familiar about the Linux driver. Does a simple mmap to the magical device /dev/mem can work? Or, do I have to start from your axidma_mmap driver function and implemente another one onto another device character?

bperez77, 2018-03-24:
One follow up to this, from issue #46. One big issue with using `/dev/mem` is that you cannot set the memory allocated with it to uncached. As the FPGA does not have a cache and the CPU does, they are not cache-coherent, and this could potentially lead to major problems in your application. So I think the only option in your case is to write the custom driver.

You should be able to take the mmap-related parts of my driver and use them as a template for yours.

bperez77, 2018-05-30:
Closing this issue due to inactivity.

closed 2018-05-30 by bperez77