bperez77/xilinx_axidma #47: Not finding tx_channel in device tree

russwinsper, 2018-03-27:
Brandon,
I seem to be having a problem finding the channels in the device tree. Below is the device tree node which is really just a copy of yours with a couple of changes. Can you suggest anything?

Brandon,
I'm quite a bit closer but now running into "axidma: axidma_dma.c: axidma_request_channels: 649: Unable to get slave channel 0: tx_channel." Here's my device tree node:

axidma_chrdev: axidma_chrdev@0 {
compatible = "xlnx,axidma-chrdev";
dmas = <&axi_dma_0 0 &axi_dma_0 1>;
dma-names = "tx_channel", "rx_channel";
};

axi_dma_0: axidma0@a0000000 {
        #dma-cells = <1>;
        compatible = "xlnx,axi-dma", "xlnx,axi-dma-6.03.a", "xlnx,axi-dma-1.00.a";
    reg = <0x0 0xa0000000 0x0 0x10000>;
        clocks = <0x42>, <0x42>, <0x42>, <0x42>;
        clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
        xlnx,include-sg;    
        xlnx,addrwidth = <0x20>;

    dma-mm2s-channel@a0000000 {
        compatible = "xlnx,axi-dma-mm2s-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <0>;
        interrupt-parent = <0x4>;
        interrupts = <0 29 4>;
    };

    dma-s2mm-channel@a0000030 {
        compatible = "xlnx,axi-dma-s2mm-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <1>;
        interrupt-parent = <0x4>;
        interrupts = <0 30 4>;
    };
};

russwinsper, 2018-03-27:
I'm using Petalinux 2017.2, does that make a difference?

bperez77, 2018-03-29:
This could be potentially one of two issues. The first (and most likely) potential issue is that the AXI DMA device tree entries aren't being loaded by the Xilinx driver, so this driver cannot find the DMA channel. The other possibility is that the channel is being claimed by another driver, though I don't think this is the case.

Can you attach the output of the dmesg command immediately after you insmod the driver? Please attach it as a file.

I've also responded to you in #39. Let's continue the discussion in this issue.

russwinsper, 2018-03-29:
Brandon,
Attached is the dmesg log from boot until after loading the driver.
[dmesg.txt](https://github.com/bperez77/xilinx_axidma/files/1861127/dmesg.txt)

russwinsper, 2018-03-29:
Oops accidentally hit the close button

russwinsper, 2018-03-30:
Brandon,
Yes the device tree is being built by petalinux and the block design looks correct, I rebuilt it, but I still see the same error with the s_axi_lite_aclk.

bperez77, 2018-03-30:
Interesting, I'm surprised that there's an error then, it sounds like the issue may lie with PetaLinux. Can you attach the full device tree as a file?

russwinsper, 2018-03-30:
Here's the full device tree.
[system.txt](https://github.com/bperez77/xilinx_axidma/files/1864969/system.txt)

russwinsper, 2018-03-30:
It's system.dts saved as text.

bperez77, 2018-03-31:
Do you happen to have the original device tree? When you go from *.dtb to *.dts, the symbolic names in the original device tree are not preserved.

On initial inspection, it looks the the `clkc` node doesn't seem to be have the right entries. It's pretty hard to say why though. I'm not too familiar with PetaLinux, but my understanding is that the generated device tree should just work. Perhaps you are not generating it off the correct design?

russwinsper, 2018-04-03:
Brandon
We solved this, it was indeed a incomplete clkc node generated by petalinux.

Thanks for the help.

closed 2018-04-03 by russwinsper

bperez77, 2018-04-03:
Great, glad to hear and no problem. If you don't mind, could you share what the original source of the issue was and how you fixed it? Just so that if other people run into a similar issue, they can use that as reference.