bperez77/xilinx_axidma #48: single dma data loss

soarsf, 2018-04-07:
Hi,
I'm trying to transfer data stream with ZUC102. PS receives data from PL fifo using DMA, then transfers it through wifi. Incoming data rate is lower than 30MBps，DMA clock runs at 300MBps. 

[image]

In order to periodically inform wifi thread that data id ready, we had to use single DMA. PL burst length is 256 bytes, fifo length is 32 KBytes.

We noticed that between each 2 transfers, there was always a data loss of 0x4C bytes, even if we did nothing else but DMA transfer, no matter what size DMA buffer is.

Debugging showed that PL fifo was never full. What could possibly have caused data loss? Thanks

bperez77, 2018-04-08:
A big red flag in your design is that the `s_axis_s2mm_tlast` signal is not connected to anything. The TLAST signal is used to indicate the end of a packet/transaction, and this is what triggers the interrupt to the PS. I believe the end of a packet is handled when the specified size is reached, if TLAST is grounded to use.

The error you're seeing is a `DmaIntErr` (see page 28 of the [AXI DMA v7.1 User Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf)). There are a few conditions for when this error can occur, but the most likely error for your case is that the S2MM stream stopped early/short. This occurs if TLAST is asserted before the expected size of the receive transfer. This means that less data was transfered than was expected.

To prevent this error, you must specify the exact size of data that you are expecting to receive. If the length of a transfer cannot be known ahead of time, then you set the size to a reasonable upper bound and safely ignore this error.

What is the size of the valid data that you're getting back? Do you also only get back one valid FIFO entry in the transfer, or more? I suspect the missing TLAST signal is the source of your issues.

soarsf, 2018-04-09:
Thank you for the reply. 
I'm trying to transfer coded video stream, so there isn't a TLAST signal. Data is written into FIFO continuously and with variable intervals. Block size could vary from  dozens  to hundreds of bytes.
All I want is a cyclic DMA with notification after each buffer is full. I suppose I should try alternative methods, the SG DMA could be my sulution.
Would you kindly answer my following questions?
1. In SG Mode(not cyclic), is callback triggered only after the whole chain was completed, or after each descriptor?
2. If callback is triggered only after the whole chain was completed, will there be callbacks in cyclic mode?
3. in AXI DMA v7.1 User Guide, cyclic DMA mode is described as an enhancement of SG DMA, but in xilinx driver(xilinx_dma.c), the functions xilinx_dma_prep_slave_sg and xilinx_dma_prep_dma_cyclic have totaly different parameters. In order to achieve cyclic mode, can I still work on your code or just start over?

bperez77, 2018-04-09:
Could you elaborate more on the data loss? Are you actually losing data, or is it just that you're getting less data than expected back in your DMA transactions?

bperez77, 2018-04-09:
[Here](https://github.com/bperez77/xilinx_axidma/commit/d9c90ca3ea1be4f53fcd0d5b378501e0d5d2229b) is the commit where I removed support for cyclic buffer descriptors, to give you an idea of what would need to be added back in. 

Also, if you're willing to switch to VDMA, that has support for cyclic transactions in the driver. However, I'm not sure it would adaptable to your use case.

soarsf, 2018-04-09:
Thank you very much! That would help a lot.
As for the data loss, the test procedure was to start a single transfer, copy data in DMA buffer somewhere else, then start another transfer.
 FIFO size was 32 KB, single DMA buffer size 16KB. Since PL writes coded video data into the FIFO in a TLV pattern with a counter, I can count how many bytes are lost.
The result was that data in every 64KB buffer was good, between buffers there was always a loss of no more than 100 bytes.
When I increased DMA buffer size to 256KB, the result turned out the same, still no more than 100 bytes lost. 
FIFO was unlikely to be full during the copy time.

soarsf, 2018-04-11:
It WAS the  TLAST signal just as you said. After it was enabled, no more data lost.
Thank you so much for your help.

closed 2018-04-11 by soarsf

soarsf, 2018-04-18:
Cyclic mode, which meets out needs, works well enough after we did some minor changes on your code.
There weren't any issues worthy of discussing.
Thanks again!

46cv8, 2020-12-31:
@soarsf I realize this is an old issue that was closed but is there any chance you could commit the changes you made to a forked copy of the repository? Or if you already did but not under the uesrname soarsf could you provide a link to your forked repository?

Also @bperez77 , if I understand correctly based on the comment for the commit one needs to revert to use the cyclic mode, one requires a custom linux kernel to make it work. It seems that this kernel and the build instructions used to live at the following github but are no longer accessible there.
https://github.com/bperez77/zynq_linux
Are they available at another url? I would really like to use cycle mode with a large 600 to 800 mb buffer.

Also @bperez77  I am unsure how the library uses scatter gather mode under the hood when operating in cyclic mode.
In order to use scatter gather doesn't it depend on where and how the M_AXI_SG port was connected. Does it require the M_AXI_SG port to be connected to the DRAM interconnect through the same HP0 connection as the M_AXI_MM2S port? Or is it also be possible to use it with BRAM accessible via a GP port and general_uio? I apologize for my lack of familiarity with the library currently, I'm just trying to understand where/how I should connect my M_AXI_SG port if I hope I hope to use your library along with cyclic mode and SG block descriptors. Thanks!