bperez77/xilinx_axidma #49: Offset of a user space virtual address is ignored

Baum55, 2018-04-09:
I call the function "axidma_oneway_transfer()" from the lib in my application and I wanted to add an offset to the buffer.
I have 2 threads. 1 thread who reads the data from the FPGA and another thread who process the data.
To make sure that the processed data are not overridden, I wanted to add an offset to the allocated buffer.
First I read to buf, then to buf + offset, then to buf + 2*offset and so on.

For example:
``
char* buf = (char*)axidma_malloc(axidma_dev, 100);
``
``
axidma_oneway_transfer(axidma_dev, rx_chans->data[0], buf + 10, 10, true);
``

But the data are always written to the base address of buf and not buf + offset (in the example: buf + 10).
Is this the desired behavior?

bperez77, 2018-04-11:
Yep, that's definitely a bug in the driver. I forgot to add in the offset from the base address that you pass in with the function. I should have fix up fairly quickly.

Baum55, 2018-04-11:
That's great to hear. Thank you.

bperez77, 2018-04-11:
Ok, I just pushed a change that should fix it. Can you let me know if that does the trick? I unfortunately don't have a test setup to verify the change.

closed 2018-04-11 by bperez77

Baum55, 2018-04-11:
No, it does not work. You add the offset only in the second case to the return addresse.
I have a issue with your commit "Changed the include for `send_sig_info`. "

I will test it intensive and give you the changes with a pull request.

bperez77, 2018-04-12:
Ahh crap, that's what I get for making a quick change like that. I'll look over the PR.

commit 3389d0f4 (2018-04-11): Fixed an issue with offsets into DMA buffers being ignored.

This closes #49. When translating a user virtual address to its
equivalent DMA address, the offset of the virtual address in the
allocated DMA buffer was being ignored. That is, the base address of the
buffer was always being returned, instead of the address corresponding
to the address specified by the user. This was leading to transactions
writing to the incorrect locations in buffers.