bperez77/xilinx_axidma #50: AXI DMA IP changes in Vivado 2018.1

baf2099, 2018-04-09:
Xilinx officially released Vivado 2018.1 today and they have made some changes which may effect your driver as can be seen here...

https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf

The main item of note is that the IP core now supports up to 64MB transfers!

It looks to me like the Xilinx Driver has not yet been updated (at least on github) to handle this increase in max transfer size as line 165 shows...
`#define XILINX_DMA_MAX_TRANS_LEN_MAX	23`
...and it should be 26

https://github.com/Xilinx/linux-xlnx/blob/master/drivers/dma/xilinx/xilinx_dma.c

bperez77, 2018-04-11:
Nice! That's a pretty good step up in max transfer size. Though since they haven't updated the max in the driver, actually trying to use the max size would lead to some pretty insidious bugs.

baf2099, 2018-05-09:
The Xilinx Driver has now been updated just to let you (everyone) know.

closed 2018-05-19 by bperez77