bperez77/xilinx_axidma #52: VDMA loopback does not work on MPSOC

chonwhite, 2018-04-11:
anybody get VDMA loopback working on MPSOC(ZCU102)?
when I tried I got a segmentation fault, and kernel NULL pointer exception.
25.093002] xilinx_axidma: loading out-of-tree module taints kernel.
[   25.100086] axidma: axidma_dma.c: axidma_dma_init: 705: DMA: Found 0 transmit channels and 0 receive channels.
[   25.110113] axidma: axidma_dma.c: axidma_dma_init: 707: VDMA: Found 1 transmit channels and 1 receive channels.
 64.747284] axidma: axidma_dma.c: axidma_start_transfer: 297: VDMA receive transaction timed out.
[   64.756174] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   64.764253] pgd = ffffffc05635e000
[   64.767605] [00000000] *pgd=0000000055f68003
[   64.771689] , *pud=0000000055f68003
[   64.777066] , *pmd=0000000000000000
[   64.779006] 
[   64.782438] Internal error: Oops: 86000006 [#1] SMP
[   64.789254] Modules linked in: xilinx_axidma(O) uio_pdrv_genirq
[   64.797107] CPU: 0 PID: 2068 Comm: benchmark Tainted: G           O    4.9.0-xilinx-v2017.2 #1
[   64.807778] Hardware name: ZynqMP ZCU102 RevB (DT)
[   64.814596] task: ffffffc055e48480 task.stack: ffffffc055f90000
[   64.822542] PC is at 0x0
[   64.827073] LR is at 0x0
[   64.831606] pc : [<0000000000000000>] lr : [<0000000000000000>] pstate: a0000145
[   64.841091] sp : ffffffc055f93d80
[   64.846500] x29: 0000000000001e00 x28: ffffffc055f90000 
[   64.853903] x27: ffffff8008942000 x26: 000000000000001d 
[   64.861329] x25: 0000000000000123 x24: 0000000000000015 
[   64.868775] x23: 0000007ff2e3bd40 x22: 0000007ff2e3bd40 
[   64.876238] x21: ffffffc055f93dd0 x20: 0000007ff2e3bd40 
[   64.883662] x19: ffffffc0571ab900 x18: 0000000000000010 
[   64.891057] x17: 0000000000000000 x16: 0000000000000000 
[   64.898451] x15: 0000000000000006 x14: 69746361736e6172 
[   64.905818] x13: 7420657669656365 x12: ffffff80089b4060 
[   64.913179] x11: ffffffc05598e650 x10: 000000000000018b 
[   64.920522] x9 : 00000000ffffffff x8 : 0000000000000003 
[   64.927833] x7 : 0000000000000004 x6 : 0000000059c01000 
[   64.935101] x5 : ffffffc055c09780 x4 : ffffffc0561afc40 
[   64.942360] x3 : ffffffc0561afbe8 x2 : 0000000000000004 
[   64.949600] x1 : 0000000000000140 x0 : 00000000ffffffc2 
[   64.956815] 
[   64.960212] Process benchmark (pid: 2068, stack limit = 0xffffffc055f90020)
[   64.969136] Stack: (0xffffffc055f93d80 to 0xffffffc055f94000)
[   64.976820] 3d80: ffffffc055f93e00 ffffff800819dc14 ffffffc056ed7800 ffffffc05596cd58
[   64.986592] 3da0: ffffffc056ed7800 0000007ff2e3bd40 0000007ff2e3bd40 0000000000000015
[   64.996385] 3dc0: 0000007f9d5f8000 ffffffc055f93ec0 0000000000000001 0000007f9d5f9000
[   65.006197] 3de0: 00000000007e9000 0000000000000001 0000007f9ce10000 00000000007e9000
[   65.015993] 3e00: ffffffc055f93e80 ffffff800819e304 ffffffc056ed7800 0000000000000003
[   65.025768] 3e20: ffffffc056ed7800 0000000080305706 0000007ff2e3bd40 0000000000401010
[   65.035566] 3e40: ffffffc055f93e80 ffffff800818c68c ffffffc0558a8700 ffffffc0558a8700
[   65.045371] 3e60: 000000002adf0010 000000000000001c ffffffc055f93e80 ffffff800819e2e4
[   65.055175] 3e80: 0000000000000000 ffffff8008082ef0 0000000000000000 00000000007e9000
[   65.064974] 3ea0: ffffffffffffffff 0000007f9dea014c 0000000060000000 0000000000000000
[   65.074763] 3ec0: 0000000000000003 0000000080305706 0000007ff2e3bd40 00000000007e9000
[   65.084552] 3ee0: 0000000000000001 0000007f9ce10000 00000000007e9000 0000000000000001
[   65.094336] 3f00: 000000000000001d 0000000000000000 0000000000000001 000000002adf1090
[   65.104072] 3f20: 000000002adf10b8 0000000000000002 0000000000000000 000000000000028e
[   65.113740] 3f40: 0000000000413ab8 0000007f9dea0140 0000000000040900 00000000007e9000
[   65.123390] 3f60: 00000000007e9000 0000000000000001 0000000000413af0 0000000000000000
[   65.133012] 3f80: 0000007f9ce10000 00000000007e9000 0000000000000000 0000000000000002
[   65.142584] 3fa0: 0000007f9d5f9000 0000007ff2e3bd20 000000000040211c 0000007ff2e3bd20

bperez77, 2018-04-12:
Interesting, a NULL pointer exception is unusual. The source of the issue seems to come from the timeout.

In order to help you resolve this, can you please provide the following:
  1. The output of the `dmesg` command after you run the application and this exception occurs (please attach this as a file).
  2. The relevant entries in your device tree for this driver and the AXI VDMA nodes.

chonwhite, 2018-04-12:
Thank you for your replay, I'm currently compiling drivers as modules and benchmark as apps using petalinux17.2.
I had to change system-user.dtsi into system-user.txt ;

below attachments are device-tree and dmesg log files.
[dmesg.log](https://github.com/bperez77/xilinx_axidma/files/1902890/dmesg.log)
[system-user.txt](https://github.com/bperez77/xilinx_axidma/files/1902899/system-user.txt)

bperez77, 2018-04-14:
It looks like you only have the end of the `dmesg` output, can you attach the entire output?

I couldn't clean anything from the kernel log, but there are two things that are interesting in your device tree. The first is what appears to be two different nodes for the driver:
```
axidma_chrdev: axidma_chrdev@0 {
        compatible = "xlnx,axidma-chrdev";
        dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
        dma-names = "tx_channel", "rx_channel";
};

vdma_char_dev: vdma_char_dev@0 {
        compatible = "baidu,video-dma";
        dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
        dma-names = "tx_channel", "rx_channel";
};
```

This shouldn't be an issue, I think, because the `compatible` string for the second one means that it won't trigger a probe of the driver. Also, you likely would've seen an earlier failure if this was the case.

The second thing I see is the MMIO address for your VDMA IP. I'm not sure what device you're using, but I'm used to seeing a slightly different address:
```
reg = <0x0 0xa0000000 0x0 0x10000>;
```

Can you verify that this is the correct address in your block design? I'll look more into the driver source code and see if I can see why this is happening.

chonwhite, 2018-04-16:
`vdma_char_dev: vdma_char_dev@0 {
        compatible = "baidu,video-dma";
        dmas = <&axi_vdma_0 0 &axi_vdma_0 1>;
        dma-names = "tx_channel", "rx_channel";
};`
this device tree is for Xilinx vdmatest.

`reg = <0x0 0xa0000000 0x0 0x10000>;`
is the vdma address for MPSOC devices,which is different from Zynq7000 devices.

EKjeldsen2, 2018-05-17:
I am having this exact same segmentation fault issue with your VDMA driver on the ZCU102 MPSOC as well.  Has a solution been found?

bperez77, 2018-05-21:
I don't think the issue for this problem was ever found. If there's any new information you have and could provide, that would be greatly appreciated. I wasn't able to repro this issue since I don't hav ethe same setup.

chonwhite, 2018-06-08:
I ended up using VIPP and v4l2 to capture images from memory.

bperez77, 2018-06-22:
Closing this issue due to inactivity.

closed 2018-06-22 by bperez77