bperez77/xilinx_axidma #53: Is scatter gather used in (vanilla) DMA mode?

formidable01, 2018-05-08:
We are using AXI DMA IP with SG enabled and looking at the code, it seems that SG is only available on the interface if you use VDMA. Is that correct? Is there a different way to use SG functionality in DMA mode that is less obvious (to me)? Thanks for your help in clarifying this.

bperez77, 2018-05-19:
You can use the AXI DMA IP with SG mode enabled. However, this is not exposed on the interface functions in the driver for AXI DMA; you are correct that you can only send one buffer at a time.

How much data? You can increase the maximum transfer size for each SG entry with the buffer length register in the AXI DMA IP in the block design.

Alternately, the driver used to suppor the cyclic mode for AXI DMA. #48 has a description on how to re-enable it. With some modifications, you can adapt this to work with non-cyclic SG transfers. I can provide some pointers for this if you decide to do that.

formidable01, 2018-05-31:
Part of what I trying to understand is whether I will see any benefit from using SG mode vs just sending many single buffers. Will the throughput be higher in SG mode? I'm not sure. I can call axidma_oneway_transfer in async mode (i.e., with wait = false) and pass it a new single buffer each time and it seems that it should be "just as good" as doing a similar operation with SG (i.e., adding a new buffer to the SG chain - and managing that chain of buffers, too). Is there a limit to the number of buffers I can queue up using  axidma_oneway_transfer?

formidable01, 2018-06-07:
I did a little experimentation and found that the actual usable queue size is around 180. That is, if I call axidma_oneway_transfer to do a TX operation with wait=false and don't have the RX running to consume that data then the DMA hangs when the queued TX buffers exceeds about 180. So that's something. 

I still need to see about SG. We will be sending relatively small buffers (about 2K or so) but potentially a lot of them. So there's going to be a cost for each transaction.  As a backup, if you could provide further pointers for implementing SG as you suggested that would be helpful.

bperez77, 2018-06-08:
Interesting, I wonder why it doesn't go up to the maximum; perhaps some of the DMA descriptors are partitioned to specific channels?

Yeah those buffers are decently small, the system call overhead may end up dominating your execution time. Reverting [this commit](https://github.com/bperez77/xilinx_axidma/commit/d9c90ca3ea1be4f53fcd0d5b378501e0d5d2229b) will re-enable cyclic mode for AXI DMA. This isn't a true SG, but will allow you to use a single buffer as a scratch pad to continuous send/receive data.

To get a true SG mode, you would have to make modifications across the entire stack. I'll describe it below:

  1. Either update [axidma_oneway_transfer](https://github.com/bperez77/xilinx_axidma/commit/d9c90ca3ea1be4f53fcd0d5b378501e0d5d2229b) to take in a list of buffers and lengths, or add a new library call similar to it that does the same.

   2. Update or add a new IOCTL in [axidma_ioctl](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_chrdev.c#L437) (the IOCTL pictured there is the `AXIDMA_DMA_WRITE` ioctl) that accepts a list of buffers and lengths. You can use the [AXIDMA_DMA_VIDEO_READ](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_chrdev.c#L456) IOCTL as an example.

  3. Update the [axidma_read_transfer](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_dma.c#L359) (and/or [axidma_write_transfer](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_dma.c#L409)) function or add a new transfer function that sets up the scatter-gather list appropriately for multiple entries. Again, the video transfer functions provide a good example of how to do this; see [axidma_video_transfer](https://github.com/bperez77/xilinx_axidma/blob/master/driver/axidma_dma.c#L582) for a good example.

With these changes, you shouldn't have to make any other changes to the rest of the code, though these changes are non-trivial.

ThomasMGilman, 2022-09-02:
Hi @bperez77 , I understand this thread has been closed for some time. I'm trying to better understand how the changes listed in this forum will enable cyclic SG dma transfers? Upon digging in the SG setup, it doesn't appear as though the Linux dma driver functions for scatter gather setup are called. Are these functions necessary too for setting up a cyclic SG list? It appears as though with the current changes I have made to this driver, by performing 'Kmalloc', and 'sg_set_buf' inside 'axidma_init_sg_entry' the callback is called once and the sglist doesn't proceed to the next sglist entry. Does this need to be performed manually in the axidma_dma_callback to push the sglist to the next entry, or should the linux drivers increment to the next entry? Do I also need to change the user's pace code to initiate the dma device in the kernel instead using 'dma_alloc_coherent'?

Forgive me if I am failing to understand; I am still relatively new to DMA, and SG setup and would like to better understand the SG setup.