bperez77/xilinx_axidma #54: insmod failing in Petalinux 2018.1

baf2099, 2018-05-11:
After upgrading to Vivado 2018.1 and Petalinux 2018.1 I built a new project with 2 PL to PS only DMAs. I downloaded your latest driver code from the repository and rebuilt the kernel module along with the rest of Petalinux. Once the device was booted the only action I took was to run...

`insmod /lib/modules/4.14.0-xilinx-v2018.1/extra/xilinx-axidma.ko `

...this immediately resulted in a segmentation fault and the dmesg output looks as follows...

`mmc0: new high speed SDHC card at address 0007
mmcblk0: mmc0:0007 SD04G 3.71 GiB 
 mmcblk0: p1
udevd[750]: starting version 3.2.2
udevd[751]: starting eudev-3.2.2
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
macb e000b000.ethernet eth0: link up (1000/Full)
IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
xilinx_axidma: loading out-of-tree module taints kernel.
Unable to handle kernel NULL pointer dereference at virtual address 00000000
pgd = dbae8000
[00000000] *pgd=1c2b8831, *pte=00000000, *ppte=00000000
Internal error: Oops - BUG: 17 [#1] PREEMPT SMP ARM
Modules linked in: xilinx_axidma(O+) uio_pdrv_genirq
CPU: 1 PID: 1196 Comm: insmod Tainted: G           O    4.14.0-xilinx-v2018.1 #3
Hardware name: Xilinx Zynq Platform
task: dc27ecc0 task.stack: dc0bc000
PC is at axidma_of_parse_dma_nodes+0x1fc/0x488 [xilinx_axidma]
LR is at of_find_property+0x40/0x4c
pc : [<bf00a40c>]    lr : [<c04f761c>]    psr: 600d0013
sp : dc0bdcd0  ip : 00000000  fp : dc7dc624
r10: dacef114  r9 : dacef100  r8 : dc7dceb0
r7 : 00000014  r6 : 00000000  r5 : 00000001  r4 : dad9fcc0
r3 : 00000000  r2 : 00000000  r1 : 600d0013  r0 : 00000000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 1bae804a  DAC: 00000051
Process insmod (pid: 1196, stack limit = 0xdc0bc210)
Stack: (0xdc0bdcd0 to 0xdc0be000)
dcc0:                                     dc0bdce8 00000001 00000001 00000001
dce0: bf00d018 00000000 dc7dc624 00000001 00000001 00000000 00000002 00000002
dd00: 00000002 c04f822c 00000000 dc172200 bf00d018 bf00c6f0 00000000 dc7dceb0
dd20: c7a3b964 c7a3b964 c7a3b960 00000001 00000009 dad9fcc0 00000002 bf00d018
dd40: 00000000 00000014 dc172200 00000002 c01768d0 bf009f1c dc172200 dad9fcc0
dd60: 0000002d dad9fcc0 dc172200 bf00d018 00000000 bf00d018 00000000 bf008074
dd80: bf008028 dc172210 bf00d018 c03b15fc dc172210 c0a55cd4 c0a55cd8 c03b0140
dda0: dc172210 dc172244 bf00d018 c0a18bb0 00000000 00000001 dacef264 c03b0280
ddc0: 00000000 bf00d018 c03b0204 c03ae990 dc072f58 dc177f34 bf00d018 dbb2e680
dde0: 00000000 c03af7b8 bf00b52e bf00b52f 00000000 bf00d018 dc0bdf40 dac87f00
de00: bf010000 c03b0a04 ffffe000 dc0bdf40 dac87f00 c0101a54 00000000 00000001
de20: 00000000 a00d0013 00000000 bf00d0c8 c0a3abc4 c0a3abc4 dad9f240 dac4501c
de40: 014000c0 c01cd348 dcb36e60 00000000 c0a3ad00 c0a3ad00 0001acf3 a00d0013
de60: bf00d080 dc0bdf40 dac87f00 00000001 dacef240 dacef264 c01768d0 c017a818
de80: bf00d080 dac87f00 bf00d080 dc0bdf40 bf00d0c8 c0179ca8 bf00d08c 00007fff
dea0: bf00d080 c0176a18 e09bc000 c0176940 c0a0e68c bf011000 e09bc000 bf00d22c
dec0: ffffffff 000b9150 014002c0 00000009 00000000 c01c1fd0 dac87f00 00000000
dee0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
df00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 000080ec
df20: 000c128c 00000000 e09c40ec 000b9150 ffffe000 00000051 00000000 c0179ec8
df40: e09bec49 e09bc000 000080ec e09c3afc e09c3970 e09c2414 00006000 00006500
df60: 00000000 00000000 00000000 000027a4 00000023 00000024 0000001b 0000001f
df80: 00000011 00000000 000080ec bef90e83 bef90e7c 00000080 c01071e4 dc0bc000
dfa0: 00000000 c0107020 000080ec bef90e83 000b91a0 000080ec 000b9150 00000049
dfc0: 000080ec bef90e83 bef90e7c 00000080 000001b9 000b66c0 bef90d84 00000000
dfe0: bef90bc0 bef90bb0 00020ebc b6e75e50 600d0010 000b91a0 00000000 00000000
[<bf00a40c>] (axidma_of_parse_dma_nodes [xilinx_axidma]) from [<bf009f1c>] (axidma_dma_init+0xac/0x1bc [xilinx_axidma])
[<bf009f1c>] (axidma_dma_init [xilinx_axidma]) from [<bf008074>] (axidma_probe+0x4c/0xc4 [xilinx_axidma])
[<bf008074>] (axidma_probe [xilinx_axidma]) from [<c03b15fc>] (platform_drv_probe+0x50/0x9c)
[<c03b15fc>] (platform_drv_probe) from [<c03b0140>] (driver_probe_device+0x1ec/0x2b0)
[<c03b0140>] (driver_probe_device) from [<c03b0280>] (__driver_attach+0x7c/0xa8)
[<c03b0280>] (__driver_attach) from [<c03ae990>] (bus_for_each_dev+0x7c/0x8c)
[<c03ae990>] (bus_for_each_dev) from [<c03af7b8>] (bus_add_driver+0x16c/0x1d4)
[<c03af7b8>] (bus_add_driver) from [<c03b0a04>] (driver_register+0xa0/0xe0)
[<c03b0a04>] (driver_register) from [<c0101a54>] (do_one_initcall+0xfc/0x11c)
[<c0101a54>] (do_one_initcall) from [<c017a818>] (do_init_module+0x54/0x1c4)
[<c017a818>] (do_init_module) from [<c0179ca8>] (load_module+0x1d98/0x1e9c)
[<c0179ca8>] (load_module) from [<c0179ec8>] (SyS_init_module+0x11c/0x12c)
[<c0179ec8>] (SyS_init_module) from [<c0107020>] (ret_fast_syscall+0x0/0x48)
Code: e1a00006 eb53b475 e3500000 1a000006 (e5963000) 
---[ end trace 588ea7ea39cf59c0 ]---
root@avnet-digilent-zedboard-2018_1:~# insmod /lib/modules/4.14.0-xilinx-v2018.1/extra/xilinx-axidma.ko ^C
root@avnet-digilent-zedboard-2018_1:~# 
root@avnet-digilent-zedboard-2018_1:~# ^C   
root@avnet-digilent-zedboard-2018_1:~# 
root@avnet-digilent-zedboard-2018_1:~# 
root@avnet-digilent-zedboard-2018_1:~# 
root@avnet-digilent-zedboard-2018_1:~# 
root@avnet-digilent-zedboard-2018_1:~# packet_write_wait: Connection to 192.168.1.167 port 22: Broken pipe
brandon@brandon-zynq:~/petaproject-2018-1$ petalinux-build --project req-tester[INFO] building project
[INFO] sourcing bitbake
`

As I mentioned in a previous post I know that xilinx upgraded their dma driver for a few reasons, like support of 64MB transfers, but I thought that if I just maintained the 8MB max that things would be ok. 

Maybe this is a compatibility issue with the new driver? or maybe this is a deeper issue? I know the circumstances are very different but could it possibly be related to the fault seen in #45 ?

baf2099, 2018-05-15:
Found the issue. Turns out that it was a mal-formatted device tree which caused the code to index a non-existent child node in the device tree dma entry. Specifically the axi dma character driver entry had looked like this...

```
axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&e_dma_hier_axi_e_dma 0 &u_dma_hier_axi_u_dma 1>; //bad line
	    dma-names = "eth_rx_img_ch", "usb_rx_img_ch";
	};
```
when it should have looked like this...
```
axidma_chrdev: axidma_chrdev@0 {
	    compatible = "xlnx,axidma-chrdev";
	    dmas = <&e_dma_hier_axi_e_dma 0 &u_dma_hier_axi_u_dma 0>; //corrected
	    dma-names = "eth_rx_img_ch", "usb_rx_img_ch";
	};
```
@bperez77 If I can make a recommendation to help others realize they've made this mistake and to prevent segmentation faults due to NULL pointer deference I would modify the axidma_of.c to validate that the child node exists before trying to use it. Maybe issue a kernel print that says it failed to find the specified child node and to check the device tree for errors?

```
  // Go to the child node that we're parsing
    dma_chan_node = of_get_next_child(dma_node, NULL);
    if (channel == 1) {
        dma_chan_node = of_get_next_child(dma_node, dma_chan_node);
    }

//ADD NEW CODE HERE TO CHECK IF DMA_CHAN_NODE IS NULL!!!

    // Read out the channel's unique device id, and put it in the structure
    if (of_find_property(dma_chan_node, "xlnx,device-id", NULL) == NULL) {
        axidma_node_err(dma_chan_node, "DMA channel is missing the "
                        "'xlnx,device-id' property.\n");
        return -EINVAL;
    }
```

closed 2018-05-30 by bperez77

eleICoto, 2019-05-09:
@bperez77 Hi it seems like you are using two dma in your design
would you please give more infomation about your design?
now i'm facing Unable to get slave channel 0: rx_channel. problem
I have modified my devicetree like this 
[image]
but insmod xxx.ko still failed

commit b1415734 (2018-05-30): Fixed a missing check for NULL in device tree parsing.

This closes #54. When parsing the child nodes of an AXI DMA node out of
the device tree, the code was missing a check for NULL, which can happen
if the node does not exist. For example, if the user mixed up the 0 and
1 index for their device tree node, this could trigger this error
condition. Without the check, the driver will crash on its probe with a
page fault.