bperez77/xilinx_axidma #57: Channels 0 and 1 in the 'dmas' list have the same channel id

rdb9879, 2018-05-18:
I'm trying to get the kernel driver working in Petalinux. I have a Vivado block diagram that has just an AXI DMA IP instantiated with the transmit/received looped back (just for testing, for now).

I've followed the Petalinux related thread: https://github.com/bperez77/xilinx_axidma/issues/24, but I am still getting "Channels 0 and 1 in the 'dmas' list have the same channel id" when I run the command "insmod ./xilinx-axidma.ko"

It looks like the solution to this in that thread was modifying system-user.dtsi to have different channel numbers (0 for one channel, 1 for the other) :

```
    axidma_chrdev: axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <&axi_dma_0 0 &axi_dma_0 1>;
    dma-names = "tx_channel", "rx_channel";
};
```

I even decompiled system.dtb into system.dts to verify that the changes took place. Here is what I see:

```
axidma_chrdev@0 {
    compatible = "xlnx,axidma-chrdev";
    dmas = <0x7 0x0 0x7 0x1>;
    dma-names = "tx_channel", "rx_channel";
```

Is there any other reason why I'd be receiving this message?

bperez77, 2018-05-21:
I think I see what the issue is. I believe that you're getting confused with channel IDs and the index number. The channel ID is not specified in this driver's device tree node (the `dmas` property). The second number that you think is the channel ID is actually the index. This index is the number that identifies the child of the device tree node that is being referred to.

For example, in the following device tree:

```
axi_dma_0: axidma0@40400000 {
    #dma-cells = <1>;
    compatible = "xlnx,axi-dma", "xlnx,axi-dma-6.03.a", "xlnx,axi-dma-1.00.a";
    reg = <0x40400000 0x10000>;
    clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
    clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";    
    xlnx,include-sg;    
    xlnx,addrwidth = <32>;

    dma-mm2s-channel@40400000 {
        compatible = "xlnx,axi-dma-mm2s-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <0>;
        interrupt-parent = <&intc>;
        interrupts = <0 29 4>;
    };
    
    dma-s2mm-channel@40400000 {
        compatible = "xlnx,axi-dma-s2mm-channel";
        dma-channels = <1>;
        xlnx,datawidth = <64>;
        xlnx,device-id = <1>;
        interrupt-parent = <&intc>;
        interrupts = <0 30 4>;
    };
};
```

`&axi_dma_0 0` refers to the node `dma-mm2s-channel@40400000`, while `&axi_dma_0 1` refers to the node `dma-s2mm-channel@40400000`. 

The value that you need to change is the `xlnx,device-id` property of the AXI DMA/VDMA device tree nodes. This value must be unique across all device tree nodes (but doesn't necessarily have to be contiguous). So, to solve your issue, you need to change one of the `xlnx,device-id` properties to 1 (or any number other than 0).

rdb9879, 2018-05-21:
Ah I understand now. I can see the `axi_dma_0:` structure in pl.dtsi, and indeed both of the `xlnx,device-id` are set to zero. What is the "appropriate" way to modify this field using yocto/bitbake? Is this another field I should add to system-user.dtsi? I am hesitant to modify pl.dtsi directly since this was automatically generated by the petalinux build tools (I believe this was generated when I imported the *.hdf file), and may get overwritten at some point.

bperez77, 2018-05-21:
I'm not sure what the appropriate way to change that field is; I haven't worked much with Yocto/BitBake. I would imagine that they would set the `xlnx,device-id` field to be unique, but that seems to not be the case.

I think adding it to the file that isn't auto-generated would be the better way to go. You should be able to refer to the device tree node in the `pl.dtsi` file and override the value for the device ID.

tpeshkin, 2018-05-22:
Hi
Previously had the same issue with device-id set to 0 for both channels. So I just copied whole amba_pl section (together with my axi_dma inside it) to system-user.dtsi and then changed device-id of next channel to appropriate value, after that succeeded to proceed with driver.
I hope this will help you al well.

bperez77, 2018-05-25:
Thanks for the suggestion @tpeshkin. For now, I'll add a reference to this issue to the README, @rdb9879.

jotorious, 2018-08-02:
I've just started this morning trying to use this module/library and I'm pretty much walking through alot of the issues that others have had to fight down.

I'm not sure if it adds value, but what I think one can do in regards to the device-tree being wrong is add a reference into the bottom of system-user.dtsi, which would only contain the change. It still requires the hardcoded address fields for each of the dma channels, but it looks cleaner. As an example:

```
/include/ "system-conf.dtsi"
/{

}

&amba_pl {
  dma@a0000000 {
    dma-channel@a0000000{
      xlnx,device-id = <0x0>;
    };
    dma-channel@a0000030{
      xlnx,device-id = <0x1>;
    };
  };
};
```

the &amba_pl basically tells the dtc to search for nodes that start with the label amba_pl

SimonLeeCHN, 2019-05-15:
when using the petalinux to build the linux on zynq , we'd better write all the changes to the files in project-spec/meta-user
so aim at how to use this project in petalinux , the example to modified devicetree would look like this:

`/include/ "system-conf.dtsi"
/{
};
 
&amba_pl{
    axidma_chrdev: axidma_chrdev@0 {
            compatible = "xlnx,axidma-chrdev";
            dmas = <&axi_dma_0 0 &axi_dma_0 1>;
            dma-names = "tx_channel", "rx_channel";
    };
};
 
&axi_dma_0{
    dma-channel@40400000 {
        xlnx,device-id = <0x0>;
    };
    dma-channel@40400030 {
        xlnx,device-id = <0x1>;
    };
};
`

Vanau, 2020-11-25:
> when using the petalinux to build the linux on zynq , we'd better write all the changes to the files in project-spec/meta-user
> so aim at how to use this project in petalinux , the example to [...]

hi, I understand what you say,but I still have some questions. When I use 2 DMAs, and one has 1 receive channel,the other has 1 transmit channel and 1receive channel. According to your example,we would find dts after changed
`
	amba_pl: amba_pl {
		axi_dma_0: dma@40400000 {
			dma-channel@40400030 {
				xlnx,device-id = <0x2>;
			};
		};
		axi_dma_1: dma@40410000 {
			dma-channel@40410000 {
				xlnx,device-id = <0x0>;
			};
			dma-channel@40410030 {
				xlnx,device-id = <0x1>;
			};
		};
	};
		axidma_chrdev@0{
			compatible = "xlnx,axidma-chrdev";
			dmas = <&axi_dma_0 0 &axi_dma_1 0 &axi_dma_1 1>;
			dma-names = "rx_channel_test", "tx_channel", "rx_channel";
		};
`
Would it work in this way?

Vanau, 2020-11-25:
> when using the petalinux to build the linux on zynq , we'd better write all the changes to the files in project-spec/meta-user
> so aim at how to use this project in petalinux , the example to [...]

hi, I understand what you say,but I still have some questions. When I use 2 DMAs, and one has 1 receive channel,the other has 1 transmit channel and 1receive channel. According to your example,we would find dts after changed
`amba_pl: amba_pl { axi_dma_0: dma@40400000 { dma-channel@40400030 { xlnx,device-id = <0x2>; }; }; axi_dma_1: dma@40410000 { dma-channel@40410000 { xlnx,device-id = <0x0>; }; dma-channel@40410030 { xlnx,device-id = <0x1>; }; }; }; axidma_chrdev@0{ compatible = "xlnx,axidma-chrdev"; dmas = <&axi_dma_0 0 &axi_dma_1 0 &axi_dma_1 1>; dma-names = "rx_channel_test", "tx_channel", "rx_channel"; };`

Would it work in this way?