bperez77/xilinx_axidma #61: Usage of axidma transfer functions

tthhbb, 2018-06-08:
Hi!
I want to use your driver for a simple openCV application with a filter in hardware. For the image transfer between PL and PS I use a VDMA block with transmit and receive channel. The benchmark example works fine and I started to adapt your axidma_transfer example to send and receive a frame of my openCV software implementation. Using the axidma_twoway_transfer function (and VDMA looped back) I receive the correct matrix again. Using two seperate oneway transfers one after the other for transmit and receive I receive some corrupted lines between correct ones (Possibly the transmit is faster than the consuming of data at the input of VDMA?)

So after trying around a lot the last days I have a few questions.
1) Does the twoway transfer function directly put received data from the input buffer to the output buffer (And therefore is only for testing reasons and any HW filters in between would just be ignored)?
2) If so, is it the right approach to use oneway transfer for sending to PL, let the PL execute it's filters and then send it back to the PS using another oneway transfer? 
3) How can I make sure that two seperate oneway transfers are in sync? And later on also with a custom IP block?
4) How do I correctly use blocking and non-blocking mode? I found out that if the transmit operation is in waiting mode it times out. And from the receiving operation I get corrupted lines no matter if I use blocking mode or not.
5) Or should I better get rid of this _frame after frame_ idea and use the video transfer function you provide? (As far as I understood only that way I get a tlast signal after each line of the image, which could be important for my filter i can imagine)

I hope I was clear enough with my questions and thank you for your time!

bperez77, 2018-06-12:
I believe you're having a bit of a conceptual miunderstanding with how the hardware and the driver interact. My responses are inline below:

> 1. Does the twoway transfer function directly put received data from the input buffer to the output buffer (And therefore is only for testing reasons and any HW filters in between would just be ignored)?

The driver is not directly responsible for the transfer of any data. It merely acts as control and sets up an AXI (V)DMA transactiont. For the transmit, it initiates a CPU to FPGA transfer that goes out to the hardware. At the same time it also sets up an FPGA to CPU transfer that *waits until data is ready* from the hardware and accepts data. In both cases, the FPGA acts as the master in the transaction, so the hardware directly controls the data flow and contents.

> 2. If so, is it the right approach to use oneway transfer for sending to PL, let the PL execute it's filters and then send it back to the PS using another oneway transfer?

No, because of the answer to (1.). I think this is where some of the misunderstanding is coming from. The CPU and FPGA are asynchronous with respect to one another, and the control happens through the driver and the AXI (V)DMA IP.

> 3. How can I make sure that two seperate oneway transfers are in sync? And later on also with a custom IP block?

In short, you don't need to. As described in (1.), the FPGA and by extension your HW filters act as the master in the transaction, so they control when data is accepted from the CPU and sent back to the CPU. If the oneway transfers are data dependent, then they will implicitly be in-sync.

> 4. How do I correctly use blocking and non-blocking mode? I found out that if the transmit operation is in waiting mode it times out. And from the receiving operation I get corrupted lines no matter if I use blocking mode or not.

There's no "correct" way to use these per-se, it is dependent on the behavior you want to have. The transmit operation in wait mode will time out if it requires that data is sent back to the CPU, because the call with wait is blocking, so you cannot issue any additional transactions. For example, if you had the loopback example, and you did a blocking transmit transaction, this would also timeout, because it requires that a receive transaction is setup to receive the data being looped back to the CPU; there is a data dependency between the transmit and receive channels.

> Or should I better get rid of this frame after frame idea and use the video transfer function you provide? (As far as I understood only that way I get a tlast signal after each line of the image, which could be important for my filter i can imagine)

This also very much depends on the application, If you find yourself continuously doing the exact same transactions, then you should probably utilize the cyclic/continous mode. This is especially true if the data you're transferring is really video data (e.g. out to a display or from a camera).

If you provide with more specifics, I can help you dig into this data corruption issue.

tthhbb, 2018-06-12:
Thanks a lot for your answers! Especially, your hint in the first answer, that actually the hardware controls the data flow, got me on the right track to solve the issues and let me understand the whole procedure better.

The "corrupted" lines in the image I got back from the FPGA resulted from the frame settings within the trans structure (width, height, depth) that are hard coded in the axidma_rw_transfer function (I have seen your FIXME comments there). I adapted them to my resolution and pixel depth and now it also works fine with my custom filter. Regardless of a blocking or non-blocking receive transfer as it should be. I also understand now why there was no problem with the loopback design, as there was no hardware in between that required the correct control bits.

So thank you again for answering my really basic questions and your work on this understandable and easy to implement driver.

bperez77, 2018-06-12:
Ahh crap, sorry about that, I forgot about fixing that issue. If you use the `axidma_video_transfer` function, it will properly account for the parameters of the video stream.

I'll create a separate issue for that and get a fix up.

bperez77, 2018-06-14:
I've created an issue for this, any follow up discussion can happen there. I'm going to close this issue.

closed 2018-06-14 by bperez77