bperez77/xilinx_axidma #62: Generalize the Transfer Function to Properly Support VDMA

bperez77, 2018-06-14:
Currently, the transfer function does not fully support VDMA; it currently hard codes the frame size to 1920x1080x4. In order to properly generalize this functionality, a struct should be added that can be used to set the frame size from the library call.

This is a follow up from issue #61.

bperez77, 2018-06-22:
@tthhbb Can you verify that this changes work for the issue you had in #61, when you get a chance?

closed 2018-06-22 by bperez77

tthhbb, 2018-06-28:
Unfortunately I cannot try it anymore, because the work was for a university project and I already had to return the Zybo. But as the only problem was the hard coded frame size I suppose your changes solve the issue.

commit d3f0966d (2018-06-22): Extend the transfer function to properly support VDMA.

This resolves #62. The transfer function has been extended to properly
and fully support AXI VMDA. Before, the part of the driver that handled
single VDMA transfers (not looping video transfers) was hardcoded to a
1920x1080x4 frame. Now, the transfer function accepts a frame structure
that specified the height, width, and depth (pixels) of the image.

This require full cross-stack changes, so the driver, library, and
example applications have been updated. Additionally, this was used as
an opprotunity to refactor some of the driver code,  [... cut]