bperez77/xilinx_axidma #63: aarch64 : Unable to set the DMA coherent mask.

sqf09f, 2018-07-01:
Hi!
I was using this driver on an MPSOC board. For PS side it was xlnx-linux 4.9.0 + ubuntu 16.04, and for PL side it was just a axidma with loopback.

When I was trying to insmod I received following:
```
[  810.260658] axidma: axidma_dma.c: axidma_dma_init: 710: DMA: Found 1 transmit channels and 1 receive channels.
[  810.260664] axidma: axidma_dma.c: axidma_dma_init: 712: VDMA: Found 0 transmit channels and 0 receive channels.
[  810.260886] axidma: axidma_chrdev.c: axidma_chrdev_init: 596: Unable to set the DMA coherent mask.
```
The device file was not created.

If I removed following lines in axidma_chrdev.c :
```
    dma_mask = DMA_BIT_MASK(8 * sizeof(dma_addr_t));
    if (dma_set_coherent_mask(dev->device, dma_mask)) {
        axidma_err("Unable to set the DMA coherent mask.\n");
        goto device_cleanup;
    }
```
Then the character device would be created.
But when I run benchmark, it looks like following:
```
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 MiB
        Receive Buffer Size: 7.91 MiB
        Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Warning: 99.95% of the receive buffer matches the initialization pattern.
This may mean that the receive buffer was not properly updated.
Single transfer test successfully completed!
Beginning performance analysis of the DMA engine.

DMA Timing Statistics:
        Elapsed Time: 0.04 s
        Transmit Throughput: 203345.20 MiB/s
        Receive Throughput: 203345.20 MiB/s
        Total Throughput: 406690.41 MiB/s
```
I'm a newbie to driver stuff. Could you please give a help with this issue?
Thanks,
Clark

sqf09f, 2018-07-01:
I managed to solve the second issue (99.95% of the receive buffer matches the initialization pattern) by removing the axi stream data FIFO I put in between axidma s2mm and mm2s.
But for the first issue I still have no idea. It seems that the driver works fine now without those dma_set_coherent_mask lines. But I still very appreciate if you can help to explain why I got that error (Unable to set the DMA coherent mask.)

Thanks a lot!

sqf09f, 2018-07-02:
Also, it seems that for MPSOC, when the sg engine on AXIDMA IP is turned off, the driver would report timeout for every transaction. Only when there is an SG port on AXIDMA IP core the driver would work.

I've attached 2 pdf files, the first on "sg.pdf" works fine with the driver, the second one "no_sg.pdf" won't work with the driver.
[sg.pdf](https://github.com/bperez77/xilinx_axidma/files/2156979/sg.pdf)
[no_sg.pdf](https://github.com/bperez77/xilinx_axidma/files/2156980/no_sg.pdf)

bperez77, 2018-07-07:
For your second issue, this could be an issue with the Xilinx AXI DMA IP. Could you try that same test with the Xilinx DMA Test driver? I don't have access to an MPSoC board, so I can't replicate your experiment. See the Xilinx Wiki page [here](http://www.wiki.xilinx.com/DMA+Drivers+-+Soft+IPs#Test%20Procedure--AXI%20DMA) for how to enable the driver. Make sure that you comment out my driver's node in the device tree.

bperez77, 2018-07-07:
I'm assuming that the kernel you're running is from the `xlnx_rebase_v4.9` branch. It looks like the only way setting the mask can fail is if DMA is not supported (see [here](https://github.com/Xilinx/linux-xlnx/blob/xlnx_rebase_v4.9/include/linux/dma-mapping.h#L569)). 

I think I know what the issue is. The device I'm passing in comes from the character device that I created for my driver. Naturally, this does not actually support any DMA operations. Instead, I should be passing in the platform device that is created when the driver is probed. I'll push that change up shortly.

bperez77, 2018-07-07:
Ok, I just pushed up a fix for it. I verified it worked on my Zybo board (32-bit architecture). Can you pull and verify the fix works for you?

closed 2018-07-07 by bperez77

sqf09f, 2018-07-09:
Thanks for the update. I just validated that it works with no issue for MPSOC aarch64.

commit c7705419 (2018-07-07): Fixed an issue with setting the DMA coherent mask.

This resolves #63. The wrong device was being passed to the function.
Instead of using the character device's generic device, the platform
device's generic device should have been used. Also, the setting of the
mask was moved into the DMA initialization function.