bperez77/xilinx_axidma #64: Question: Minimum Tx Transfer Size

jimknoll1, 2018-07-11:
Is there a minimum amount of data transferred via MM2S?  I have axidma_malloc'ed a 216 byte (54 32-bit words) tx buffer and called axidma_twoway_transfer with the size set to 216.  When I observe a transfer with ILA, I see 1024 bytes transferred.  The first 216 bytes are the data I intended to transfer followed by 808 bytes of 0.

Thanks,

Jim

bperez77, 2018-07-13:
I took a quick look over the backend Xilinx DMA driver and the AXI DMA user guide, and there doesn't seem to be any enforced minimum for the transfer. My driver also doesn't enforce any minimum size.

Can you send me a block diagram of where you have the ILA connection? Also, is this just a loopback design, or do you have custom IP downstream on the MM2S AXIS interface?

jimknoll1, 2018-07-13:
Thanks for writing me back.  I'm dumb.  I was not running updated software
that was sending the correct packet size.  The software was sending a 1 kB
packet.

The driver works great, but we do notice when sending a packet of 1024
32-bit words, the dma does not assert valid for 60 clocks after
transferring the first 512 words.  It does complete the other 512 words
correctly, but it just seems weird.

We don't notice the same behavior from a standalone no OS app.

Thanks again for responding and for all the great work you dinner on the
driver.

Jim

bperez77, 2018-07-16:
No worries, it's hard to keep everything straight when you have such a complex system like this.

That's interesting and quite weird. I've never noticed or heard of that behavior before (though I've never checked with ILA either). Since this behavior doesn't happen from the embedded app, I suspect it must be something specific with Linux. My initial though would be that the transaction is being split up into two separate transfers, but I think that the minimum width of buffer length register parameter value is larger than 10. 

Are you familiar with how to run the AXI DMA test driver? I can give you some pointers if not,. This should help isolate if the issue is with the Xilinx backend DMA driver, or if it's with my driver.

One other thing to note is that the system call overhead is pretty high, so it will likely tank the performance of small transfer sizes like the 1 KB buffers that you're using (unless you're transfering a ton of them).

bperez77, 2018-07-16:
Actually it looks like the minimum width of the buffer length register can be 8, so it could be possible the transaction is automatically being split into an SG transaction (see page 24 of the [Xilinx AXI DMA v7.1 User Guide](https://www.xilinx.com/support/documentation/ip_documentation/axi_dma/v7_1/pg021_axi_dma.pdf). Just out of curiosity, what value did you set this parameter to?

jimknoll1, 2018-07-17:
[image]
Attached is a screenshot from ILA.  I have the DMA buffer length register width set to 14.  Unfotunately, since we got our project working, I have lost my ZedBoard so I can't help with troubleshooting.
[image]

closed 2018-07-17 by jimknoll1