bperez77/xilinx_axidma #67: Question about threading

jotorious, 2018-09-11:
I am writing an application which uses multiple threads, to transport data in from a TCP socket over to the DMA and back out a TCP socket. 

One thread writes data into the axidma writebuf buffer via a socket recv(). Another thread reads data from the axidma readbuf via send(). And then another thread calls axidma_twoway_transfer. (I believe) I have properly synchronized the threads. I have a state machine that only allows socket data to be received into the writebuffer when it's empty, and then only allows the call to axidma_twoway_transfer when said writebuffer is full. The same on the socket sending side. I have 2 writebuffers, and 2 read buffers, and the point of this scheme is to pipeline the behaviors so that my application runs as fast as the slowest of these 3 threads (recv, dma, send).

When I run my application, it segfaults in what appears right now to be the axidma_twoway_transfer call. I read the documentation, and while I saw everything about single process, and both sides have to occur synchronously, I didn't get the impression that what I'm doing above would fail. I'm trying to understand if my approach as described is a nonstarter

jotorious, 2018-09-12:
Delete this question. I had failed to initialize the device pointer passed to my dma thread, This approach works.

closed 2018-09-12 by jotorious

bperez77, 2018-09-15:
Great, glad to hear it worked it. Yeah it can be tricky to get all the pieces in place.

Multithreading isn't explicitly supported by the driver, but (at least for actual transfers) it should be handled by the backend Xilinx AXI DMA driver.