bperez77/xilinx_axidma #70: ACP port support

user-mod, 2018-10-25:
Hi,

I have a simple question regarding your driver: do it support data transfer through the ACP port?

I tried with a zynq ultrascale+ device and I get the following log:
[  126.148496] xilinx-vdma a0000000.dma: Channel ffffffc87b1c0e18 has errors 200, cdr 704c0000 tdr 704c0180
[  127.148486] xilinx-vdma a0000000.dma: Cannot start channel ffffffc87b1c0c18: 10009
[  137.196167] axidma: axidma_dma.c: axidma_start_transfer: 301: DMA receive transaction timed out.
Failed to perform the AXI DMA read-write transfer: Timer expired
DMA read write transaction failed.

I also tried with the HPC (high performance coherent) port and it works nicely.

woodmeister123, 2018-11-14:
I didn't try the ACP port, the HPC port worked fine in non-coherent mode without any mods to the driver. Small change required to make it work cache coherent.

user-mod, 2018-11-15:
I tried again this morning and by enabling kernel logs at the highest level.

For a scatter gather hw, I get the previous logs and without scatter gather, I get:
[  190.360801] xilinx-vdma a0000000.dma: Channel ffffffc87b191418 has errors 20, cdr 0 tdr 0

I'm using your axidma_transfer user space example program and, in this last hw configuration, I get an output file correctly sized but filled with "@@@@@".

user-mod, 2018-11-20:
Here, you can find the logs for both the benchmark application & the dmesg:
[axidma_benchmark.log](https://github.com/bperez77/xilinx_axidma/files/2598862/axidma_benchmark.log)
[dmesg.log](https://github.com/bperez77/xilinx_axidma/files/2598863/dmesg.log)