bperez77/xilinx_axidma #71: Multiple receive buffer queue

woodmeister123, 2018-10-29:
Should it be possible to queue up multiple receive buffers using the one way transfer command in async mode? Currently when I do this all of the callbacks return at the same time once data starts flowing.

bperez77, 2018-11-11:
This should be possible. My driver doesn't directly handle queueing of asynchronous requests, but the backend AXI DMA driver supports having more than one outstanding requests across any given channel. The maximum number of these is controlled by the maximum number of DMA pool descriptors in the Linux kernel.

An interrupt should be delivered once for each request, which eventually will manifest as your callback. Do you mean that you're only seeing one interrupt in total? There may be something misconfigured in the driver if that's the case.

woodmeister123, 2018-11-13:
The callback was being triggered the correct number of times, but I'm not sure if this was all somehow being triggered from the same interrupt.

My design is streaming based with no capability to add back-pressure, so I think really to get this working when the system is under load I need to enable one of the SG or cyclic modes.

bperez77, 2018-11-20:
Oh I see, are you saying you need additional information in the interrupt? Like some sort of inrementing index. Yeah SG mode will ensure you can handle a higher load.

woodmeister123, 2018-11-23:
I've now managed to implement cyclic mode which does what I want much better. The xilinx driver supports it natively since they've merged things so there were only modifications needed to this driver. I'll try and submit a patch when I get time.

closed 2018-11-23 by woodmeister123

honorpeter, 2020-01-09:
> The callback was being triggered the correct number of times, but I'm not sure if this was all somehow being triggered from the same interrupt.
>
> My design is streaming based with no capability to [...]

@woodmeister123 I use one DMA to receive  data from PL . But the callback was not being triggered at all . How to use the  callback ? Could you share some code about how to use the callback?Thank you .