bperez77/xilinx_axidma #73: Axi_CDMA (Central Direct Memory Access) device driver

92jisuk, 2018-11-14:
Hi. I am happy to see you in your github.

I found the device driver of axi_dma in your github.
I ported DMA driver on the ZED board using your code.
It was working very well.

But now, I am trying to port the CDMA driver on my board.
I must add and change some codes for porting the CDMA driver.
But I have no idea for it...

Could you share the CDMA device driver codes if you have....?

Thank you.

92jisuk, 2018-11-15:
Hi. I searched about CDMA driver in Google.
And I found the cdmatest.c code what looks like device driver of CDMA.
Here is the github site: https://github.com/Xilinx/linux-xlnx/blob/master/drivers/dma/xilinx/cdmatest.c

Could you check this is the correct code for axi_cdma device driver?
If right, What should I do using this code...?

Thank you.

bperez77, 2018-11-20:
Certainly, I don't have the code right now, but I should be able to implement it in a few days. I'm currently away from my FPGA board, so the code won't be heavily tested, but I think it should be straight-forward.

92jisuk, 2018-11-21:
Hi Brandon.
 
Thank you for consideration about my issue.
I can implement using AxiDMA instead of AxiCDMA eventhough it is inconvenient for me.
 
If making the cdma device driver is not for your job, I don't want to steal your time...
I hope to really  appreciate for your mind.
 
Thank you Brandon!
 
From Jisu.

92jisuk, 2018-12-05:
Hi, Brandon.
I had said that you don't need to effort for my question.
But I need your help...

I get a "testcdma.c" code from:
https://github.com/Xilinx/linux-xlnx/blob/master/drivers/dma/xilinx/cdmatest.c
and tried to compile this file using petalinux.
I failed to compile it with some errors and I am wondering this file is exact device driver file or not.

Could you please help me to create CDMA device driver?
Thank you.

From Jisu