bperez77/xilinx_axidma #78: Several AXI DMA controllers

GOOD-Stuff, 2018-12-06:
Hello!
How would be works the driver with several AXI DMA controllers? 
And would be conflicts in system with AXI Ethernet? 

With regards,
Gustov Vladimir.

majicjq, 2018-12-08:
hello,Gustov！
I use AXI-DMA(rx&tx) and VDMA(rx&tx).writting device tree as:
	axivdma_chrdev: axivdma_chrdev@0 {/* github */
    		compatible = "xlnx,axidma-chrdev";
    		dmas = <&axi_vdma_0 0 
			&axi_vdma_0 1
			&axi_dma_0 0
			&axi_dma_0 1>;
    		dma-names = "vdma_tx_channel", "vdma_rx_channel", "dma_tx_channel", "dma_rx_channel";
		xlnx,num-fstores = <0x3>;
	};
And I have to change "xlnx,device-id" in &axi_vdma_0 and &axi_dma_0.

GOOD-Stuff, 2018-12-10:
Hi, majicjq!
Thanks for reply!

closed 2018-12-10 by GOOD-Stuff