bperez77/xilinx_axidma #83: axidma_video_transfer results in unallocated DMA Buffer Error

dnicholes, 2019-01-30:
I am trying to take video from a camera and use the VDMA IP to move frames to the PS memory using this xilinx_axidma driver. The software seems pretty straightforward. Here's what I am doing:

```
    int rx_channel;
    axidma_dev_t axidma_dev;
    const array_t *rx_chans;
    char *rx_buf;

    // Initialize the AXI DMA device
    axidma_dev = axidma_init();
    if (axidma_dev == NULL) {
        fprintf(stderr, "Failed to initialize the AXI DMA device.\n");
        rc = 1;
    }
    printf("axidma_benchmark: rx_size=%u\n", BUFFER_SIZE);
    rx_buf = (char*)axidma_malloc(axidma_dev, BUFFER_SIZE);
    if (rx_buf == NULL) {
        perror("Unable to allocate receive buffer from the AXI DMA device");

        axidma_destroy(axidma_dev);

        return -1;
    }
    else {
    	printf("Allocated VDMA RX Buffer at: 0x%" PRIXPTR "\n", (uintptr_t)rx_buf);
    }

    rx_chans = axidma_get_vdma_rx(axidma_dev);
    if (rx_chans->len < 1) {
        fprintf(stderr, "Error: No receive channels were found.\n");

        axidma_free(axidma_dev, rx_buf, BUFFER_SIZE);

        axidma_destroy(axidma_dev);

        return -ENODEV;
    }

  printf("channel %d size: %d Buffer size:%d\r\n", rx_chans->data[0],rx_chans->len,BUFFER_SIZE);

  rx_channel = rx_chans->data[0];
  volatile int done = 0;
  axidma_set_callback(axidma_dev, rx_channel, dma_int_handler, (void*)&done);
  printf("Start Receive 4\n");

  //axidma_oneway_transfer(axidma_dev, rx_channel, rx_buf, BUFFER_SIZE, true);
  axidma_video_transfer(axidma_dev, rx_channel, IMG_WIDTH, IMG_HEIGHT, IMG_DEPTH, (void**)&rx_buf, 3);
  while (!done) {
    axidma_stop_transfer(axidma_dev, rx_channel);
  }

  display_result(rx_buf);

  axidma_free(axidma_dev, rx_buf, BUFFER_SIZE);

  axidma_destroy(axidma_dev);
```

This fails with the error: 
`[   30.214203] axidma: axidma_dma.c: axidma_init_sg_entry: 114: Requested transfer address 0000007f9412a000 does not fall within a previously allocated DMA buffer.`

Any insights into this failure would be much appreciated!

dantepayne, 2024-01-26:
I meet the same problem,have you fix it ?
More Information:
If I set VDMA IP frame buffers to 3 then run this code 
`axidma_video_transfer(dma.dev,dma.trans.input_channel,dma.trans.video.rx_frame.width,dma.trans.video.rx_frame.height,dma.trans.video.rx_frame.depth,(void **)&dma.trans.video.frame_buffers,3);`
Then I meet the same problem like you.
If I set VDMA IP frame buffers to 1 then run this code
`axidma_video_transfer(dma.dev,dma.trans.input_channel,dma.trans.video.rx_frame.width,dma.trans.video.rx_frame.height,dma.trans.video.rx_frame.depth,(void **)&dma.trans.video.frame_buffers,1);`
Every time, I malloc "*frame_buffers" like this:
`dma.trans.video.frame_size = dma.trans.video.rx_frame.width*dma.trans.video.rx_frame.height*dma.trans.video.rx_frame.depth;
dma.trans.video.frame = axidma_malloc(dma.dev,dma.trans.video.frame_size)`
 Kernel stack is corrupted in: axidma_prep_transfer+0x1c8/0x29c [axidma_modules]
Here is My FPGA Design:
[image]

wangzhenhua92429, 2024-01-26:
你发给我的邮件已收到！                                ——王振华

La-Nausee, 2024-04-17:
> I meet the same problem,have you fix it ? More Information: Every time, I malloc "*frame_buffers" like this: `dma.trans.video.frame_size = dma.trans.video.rx_frame.width*dma.trans.video.rx_frame.height*dma.trans.video.rx_frame.depth; dma.trans.video.frame = axidma_malloc(dma.dev,dma.trans.video.frame_size)` If I set VDMA IP frame buffers to 3 then run this code `axidma_video_transfer(dma.dev,dma.trans.input_channel,dma.trans.video.rx_frame.width,dma.trans.video.rx_frame.height,dma.trans.video.rx_frame.depth,(void **)&dma.trans.video.frame_buffers,3);` Then I meet the same problem like you. If I set VDMA IP frame buffers to 1 then run this code `axidma_video_transfer(dma.dev,dma.trans.input_channel,dma.trans.video.rx_frame.width,dma.trans.video.rx_frame.height,dma.trans.video.rx_frame.depth,(void **)&dma.trans.video.frame_buffers,1);`
> 
> Kernel stack is corrupted in: axidma_prep_transfer+0x1c8/0x29c [axidma_modules] Here is My FPGA Design: ![ov5640](https://private-user-images.githubusercontent.com/100111039/299865615-9f565b9f-b24b-434d-9126-c8eee5bb5291.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMjgwMTQsIm5iZiI6MTcxMzMyNzcxNCwicGF0aCI6Ii8xMDAxMTEwMzkvMjk5ODY1NjE1LTlmNTY1YjlmLWIyNGItNDM0ZC05MTI2LWM4ZWVlNWJiNTI5MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QwNDIxNTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZmQ0MjUzODg3NzFiNGZjN2E3MjdhZTcwM2E5OTAxYWMzMTBmYTI0YjU2MTg0ZWNkNmVkMTE0ZTExYmNkYWJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TtT4-62BzX3vgBGiFGDpJiIv7Lxs3vXHmm5TKLFiJXY)

Hi, bro. Have you solved this problem?