bperez77/xilinx_axidma #84: I did just a short test, but get a read timeout, see below. Do I need to change the example code somehow?

maikonadams, 2019-02-03:
I did just a short test, but get a read timeout, see below. Do I need to change the example code somehow?
```
root@te0715_linux:~# modprobe xilinx_axidma
xilinx_axidma: loading out-of-tree module taints kernel.
axidma: axidma_dma.c: axidma_dma_init: 706: DMA: Found 0 transmit channels and 0 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 708: VDMA: Found 1 transmit channels and 1 receive channels.
root@te0715_linux:~# ./axidma_benchmark -v -n 100
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
        Number of DMA Transfers: 100 transfers

Using transmit channel 0 and receive channel 1.
xilinx-vdma 43000000.dma: Frame Store Configuration is not enabled at the
xilinx-vdma 43000000.dma: H/w level enable Debug info 13 or 6 at the h/w level
xilinx-vdma 43000000.dma: OR Submit the frames upto h/w Capable

xilinx-vdma 43000000.dma: Frame Store Configuration is not enabled at the
xilinx-vdma 43000000.dma: H/w level enable Debug info 13 or 6 at the h/w level
xilinx-vdma 43000000.dma: OR Submit the frames upto h/w Capable

axidma: axidma_dma.c: axidma_start_transfer: 298: VDMA receive transaction timed out.
Failed to perform the AXI DMA read-write transfer: Timer expired
root@te0715_linux:~# 
```

_Originally posted by @Westwood68 in https://github.com/bperez77/xilinx_axidma/issues/24#issuecomment-348455892_

how did you get to compile the benchmark to petalinux ?

suikammd, 2019-05-05:
Hey, will kindly tell me how to make this driver work on petalinux?
I followed the author's guide in [Added how to build the driver with PetaLinux.](https://github.com/bperez77/xilinx_axidma/issues/24#issuecomment-335252145), but I don't know how to make the examples work in petalinux?

tpeshkin, 2019-05-06:
Do you have an issue with building driver or examples, or there are errors when executing example app (share your log)?

eleICoto, 2019-05-12:
@tpeshkin Sorry for bothering, would you please post your makefile and xx.bb you use for building example？I know how to build app in peatlinux, but don't know how to edit makefile

tpeshkin, 2019-05-13:
Hi, my .bb and Makefile for axidma-benchmark example

[files.zip](https://github.com/bperez77/xilinx_axidma/files/3171980/files.zip)

benpaodewali, 2019-05-29:
> Hey, will you kindly tell me how to make this driver work in petalinux?
> I followed the author's guide in [Added how to build the driver with PetaLinux.](https://github.com/bperez77/xilinx_axidma/issues/24#issuecomment-335252145), but I don't know how to make the examples work in petalinux?

你可以直接在目录下运行：make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm examples
在这之前需要先在petalinux目录下source一下那个setting.sh，配置这个终端的环境，然后跳到这个目录运行上面这句，就可以得到应用程序的可执行文件了

suikammd, 2019-05-30:
Hi，这个执行是没有问题的。我现在的问题是，如何在板子上验证这个文件正确执行了（我用的是xilinx zcu104）？不过我发现在insmod这个driver的时候没有任何输出。

eleICoto, 2019-05-30:
@suikammd after you insmod driver, type dmesg to check the debug message

suikammd, 2019-05-30:
I check the message, but no new message show up. 
Maybe because of the wrong device tree file?
Another question is I'm not sure whether it works on petalinux qemu.

eleICoto, 2019-05-30:
@suikammd  can you post your dmesg message after you insmod?

suikammd, 2019-05-31:
It's exactly the same as before I insmod the driver.