bperez77/xilinx_axidma #85: Unable to prepare the dma engine for the DMA transmit buffer.

maikonadams, 2019-02-06:
I am getting the following messages when I run the testbench,
do you guys have any idea what could it be ?

axidma: axidma_dma.c: axidma_prep_transfer: 236: Unable to prepare the dma engine for the DMA transmit buffer.
cma: cma_release(page effd1000)
cma: cma_release(page effc1000)
cma: cma_alloc(cma c184d420, count 2025, align 8)
cma: cma_alloc(): returned effc1000
cma: cma_alloc(cma c184d420, count 2025, align 8)
cma: cma_alloc(): returned effd1000

maikonadams, 2019-02-06:
I'm using direct transfer only , and I see the function is dmaengine_prep_slave_sg , maybe that's y.

maikonadams, 2019-02-11:
I am using Petalinux , and this is my system-user.dtsi :
/include/ "system-conf.dtsi"
/ {
amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axidma_chrdev: axidma_chrdev@0 {
    			compatible = "xlnx,axidma-chrdev";
    			dmas = <&axi_dma_0 0 &axi_dma_0 1>;
    			dma-names = "tx_channel", "rx_channel";
    		};
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-names = "mm2s_introut", "s2mm_introut";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x40>;
			xlnx,sg-length-width = <0xe>;
			dma-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x40>;
				xlnx,device-id = <0x0>;
			};
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x40>;
				xlnx,device-id = <0x1>;
			};
		};
		
	};
};

lsmod
    Tainted: G  
axidma 28672 0 - Live 0xbf008000 (O)
uio_pdrv_genirq 16384 0 - Live 0xbf000000
 when I install :

axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 1 transmit channels and 1 receive channels.
axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit channels and 0 receive channels.

I feel I am close ... but

kingofworldch, 2019-03-21:
Hello，I meet this problem，too。How to solve this problem for you？ I am confused！

sethk2, 2019-04-08:
Most likely you've already solved this or moved on, but for me I needed to change the "Width of Buffer Length Register" for the AXI Direct Memory Access IP in Vivado. I think by default it's set to something like 14 and bumping it to 26 allowed the benchmark test to run.

[image]

yrink, 2019-06-28:
Hello, 
iam getting the same Error messages as above. 
Ima using a System, which looks like my attached file. 
[image]
With the AXIS_Switches i wanna control the routing to my Test IP Cores. This should be fine so far, because my Transaction stops just after 236 Transfers. I guess my problem is in the Signals of my Test IPs. Does someone see my issue in this Code?

typedef ap_axis <32,1,1,1> AXI_T;
typedef hls::stream<AXI_T> STREAM_T;

void test2(STREAM_T &A, STREAM_T &B){

#pragma HLS INTERFACE axis port=A
#pragma HLS INTERFACE axis port=B
#pragma HLS INTERFACE ap_ctrl_none port=return

    AXI_T tmpA, tmpB;

    int LEN=5000;

    for(int i=0; i<LEN; i++){

#pragma HLS PIPELINE

        A >> tmpA;

        tmpB.data=tmpA.data+tmpA.data;
        if(i == LEN-1){
			tmpB.last = 1;
		}
        tmpB.keep = 0xf;
		B << tmpB;

    }
   }

I assert the TLAST Signal on the last Transfer, so actually it should be fine. 

My DMA Settings are:
Width of buffer: 26 Bits-> Should be more than enough, becaus i only transfer 5000 32 Bit values
Address Width: 64 Bits
Memory Map Data Width: 64
Stream Data Width: 32
Burst Size:256 
Allow Unaligned Transfers and enable Scatter gather are checked.

loylosky, 2020-11-04:
> Most likely you've already solved this or moved on, but for me I needed to change the "Width of Buffer Length Register" for the AXI Direct Memory Access IP in Vivado. I think by default it's set to something like 14 and bumping it to 26 allowed the benchmark test to run.
> 
> [image]

Hi, may I ask does this solution solve the issue? I met the same problem recently. I try to modify my device tree file for the first time, but it didn't work. I would really appreciate it if you could provide some advice. Thanks!

loylosky, 2020-11-04:
> I am using Petalinux , and this is my system-user.dtsi :
> /include/ "system-conf.dtsi"
> / {
> amba_pl: amba_pl {
> #address-cells = <1>;
> #size-cells = <1>;
> compatible = "simple-bus";
> ranges ; [...]

Hi, I got the same problem as you. I have tried several methods but none of them worked. May I ask have you solved this issue later? Thanks!