bperez77/xilinx_axidma #87: cannot create duplicate filename'/class/axidma'

visen-yanghx, 2019-02-21:
Hello, friend:

Now I also do axiDMA linux driver add, but I use zu9 , encountered some problems,  thank you!!!

My source code is https://github.com/bperez77/xilinx_axidma; however, the following modifications have been made:

1. Driver

A. Modifying Device Tree Files

Add the device tree file content correctly according to the actual PL hardware design. (My PL side uses three AXI-DMA, which is basically the same as your PL design, I can make sure there is no problem.)

B. Driver files added to source code

Create an extern folder under the / kernel / drivers directory to place the driver files. Copy all the files under the driver directory to the extern folder. Modify the Makefile file under the drivers directory and add the following contents:

Obj - $(CONFIG_EXTERN) += extern/

Modify the Kconfig file in the drivers directory and add the following:

Source "drivers/extern/Kconfig"

C. Configuration Kernel

Enter commands in the kernel source directory:

# make ARCH = arm menuconfig

Check the following kernel configuration options

Device Drivers - - >

Skycaster external driver support - ->

Skycaster external driver support

[*] enable debug message

<*> axidma driver

My device tree is

Axi_dma_0: dma@80010000{

# dma-cells = < 1 >;

Clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_s2mm_aclk";

Clocks = <&clk 71>, <&clk 71>, <&clk 71>, <&clk 71>, <&misc_clk_0>;

Compatible = xlnx, axi-dma-1.00.a;

Interrupt-names = mm2s_introut, s2mm_introut;

Interrupt-parent = <&gic>;

Interrupts = 0 89 40 904 >;

Reg = 0x0 0x80010000 x0 0x10000 >;

Xlnx, addrwidth = < 0x20 >;

Xlnx, sg-length-width = < 0x10 >;

Dma-channel@80010000{

Compatible = xlnx, axi-dma-mm2s-channel;

Dma-channels = < 0x1 >;

Interrupts = 0 894 >;

Xlnx, datawidth = < 0x100 >;

Xlnx, device-id = < 0x0 >;

}

Dma-channel@80010030{

Compatible = xlnx, axi-dma-s2mm-channel;

Dma-channels = < 0x1 >;

Interrupts = 0 904 >;

Xlnx, datawidth = < 0x100 >;

Xlnx, device-id = < 0x1 >;

}

}

Axi_dma_1: dma@80020000{

# dma-cells = < 1 >;

Clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk", "m_axi_s2mm_aclk";

Clocks = <&clk 71>, <&clk 71>, <&clk 71>, <&clk 71>, <&misc_clk_0>;

Compatible = xlnx, axi-dma-1.00.a;

Interrupt-names = mm2s_introut, s2mm_introut;

Interrupt-parent = <&gic>;

Interrupts = 0 91 40 92 4 >;

Reg = < 0x00x80020000x00x10000 >;

Xlnx, addrwidth = < 0x20 >;

Xlnx, sg-length-width = < 0x10 >;

Dma-channel@80020000{

Compatible = xlnx, axi-dma-mm2s-channel;

Dma-channels = < 0x1 >;

Interrupts = 0 914 >;

Xlnx, datawidth = < 0x100 >;

Xlnx, device-id = < 0x0 >;

}

Dma-channel@80020030{

Compatible = xlnx, axi-dma-s2mm-channel;

Dma-channels = < 0x1 >;

Interrupts = 0 924 >;

Xlnx, datawidth = < 0x100 >;

Xlnx, device-id = < 0x1 >;

}

}

Axidma_chrdev: axidma_chrdev@0{

Compatible = xlnx, axidma-chrdev;

DMAS = < & axi_dma_0 0x0 & axi_dma_0 0x1>;

Dma-names = "axidma0_tx_channel" and "axidma0_rx_channel";

Dma-id = < 0x0 >;

Chrdev-name = axidma0;

Minor-num = < 0x0 >;

Num-devices = < 0x1 >;

}

Axidma_chrdev@1{

Compatible = xlnx, axidma-chrdev;

DMAS = < & axi_dma_1 0x0 & axi_dma_1 0x1>;

Dma-names = "axidma1_tx_channel" and "axidma1_rx_channel";

Dma-id = < 0x1 >;

Chrdev-name = axidma1;

Minor-num = < 0x1 >;

Num-devices = < 0x1 >;

}

However, the compilation succeeded and the startup error was as follows:

[2.861032] axidma: axidma_dma.c: axidma_dma_init: 708: DMA: Found 1 transmit channels and 1 receive channels.

[2.870852] axidma: axidma_dma.c: axidma_dma_init: 710: VDMA: Found 0 transmit channels and 0 receive channels.

[2.881047] axidma: axidma_chrdev.c: axidma_chrdev_init: 593: add a character device - - - yanghx. (This is the print added by the kernel)

[2.889977] axidma: axidma_dma.c: axidma_dma_init: 708: DMA: Found 1 transmit channels and 1 receive channels.

[2.899784] axidma: axidma_dma.c: axidma_dma_init: 710: VDMA: Found 0 transmit channels and 0 receive channels.

[2.909826] sysfs: cannot create duplicate filename'/class/axidma'

[2.915839] mmc0: new HS200 MMC card at address 0001

[2.916170] mmcblk0: mmc0:0001 Q2J55L 7.09 GiB

[2.916289] mmcblk0 boot 0: mmc0:0001 Q2J55L partition 1 16.0 MiB

[2.916402] mmcblk0boot 1: mmc0:0001 Q2J55L partition 216.0 MiB

[2.916511] mmcblk0rpmb: mmc0:000

wouterjoris, 2019-04-16:
I think you are seeing this because you copied, as I did, the full example device tree snippet into your system-top.dts file while it is autogenerated at the same time into pl.dtsi.

shuai132, 2019-06-30:
The code `compatible = "xlnx,axidma-chrdev";` in your device tree should be one and only one. Please check it!