bperez77/xilinx_axidma #90: How to build xilinx_axidma-master with petalinux without the error of oe_runmake failed

feng007liu, 2019-03-10:
I created a new modules with petalinux, and copy all of the C source files and header files under the xilinx_axidma-master driver directory, and the header file include/axidma_ioctl.h to the driver source directory, finally, remove the auto-generated file xilinx-axidma.c. However, when I try to build the drvier, there were some mistakes :

Since I have delete the automatically generated xilinx-axidma.c, a tips about "unable to get checksum for xilinx-axidma SRC_URI entry ....." appeared, and I delete the SRC_URI line in the xilinx-axidma.bb, and got the error as shown below:
2.ERROR:xilinx-axidma-1.0-r0 do_compile: oe_runmake failed

I am using the Ubuntu 16.04.2，and petalinux 2017.4

This problem is very troubling tome, please advice if someone have a little spare time :)
Thank you very much!

GOOD-Stuff, 2019-04-01:
Hello. 
I don't work with Petalinux, but you can try to separately compiled this driver (set KBUILD_DIR to your kernels source) and library and then send it (via scp or copied it on SD card) to worker system.

honorpeter, 2019-06-16:
@feng007liu  I aslo has this problem, any one could help ?

honorpeter, 2019-06-16:
@bperez77  I'm using the driver on zcu102 with petalinux 2018.2 and  I have the same  problem  like  https://github.com/bperez77/xilinx_axidma/issues/90#issue-419143157 please advice if someone have a little spare time,thank you.

chinamaoge, 2019-07-28:
I aslo has this problem, any one could help ? I'll keep an eye on the question！

JulieLambert, 2020-07-20:
I also had errors building the xilinx-axidma driver with petalinux 2019.2.
Adding `#include <linux/of_device.h>` to axi_dma.c solved the problem.