bperez77/xilinx_axidma #91: Inappropriate ioctl for device

pmccumber, 2019-03-29:
We seem to have a driver installed correctly.  It is a read-only axi dma.

_[  976.113508] axidma: loading out-of-tree module taints kernel.
[  976.120585] axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 0 transmit channels and 1 receive channels.
[  976.130519] axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit channels and 0 receive channels._

But when I try to invoke the driver I get this message.  I boiled down the transaction to the simplest of forms but here is the line and the invocation with the error:

_int rc = axidma_oneway_transfer(mDmaDev, DMA_RX_CHANNEL, mpDmaBuf, DMA_BUFFER_SIZE, true);

root@olympus20181217:~# ./axidma_hw
Failed to perform the AXI DMA transfer: Inappropriate ioctl for device
RX DMA failed! (-1)_

What is it I'm missing here?

GOOD-Stuff, 2019-04-01:
Hi! 
Are you correct compile library (libaxidma.so) and copy it into /lib? 
Also you can try set ILA around your AXI DMA IP and look to incoming data, if DMA doesn't up s2mm interrupt, may be you have some errors in PL project.

pmccumber, 2019-04-02:
Thanks for the response.  

I explicitly set the LD_LIBRARY_PATH to where I put the libaxidma.so.  

And we are setting ILA around the AXI DMA IP.  That's a good idea.  

From a sw perspective, it seems pretty simple.  The drive installs.  The dmesg looks good.  The API is pretty simple.  I just get that Invalid ioctl.  I'll post what we see.

GOOD-Stuff, 2019-04-03:
Hello.

> I explicitly set the LD_LIBRARY_PATH to where I put the libaxidma.so.

No, I mean copy it into /lib directory on Zynq board.

pmccumber, 2019-04-08:
Is anyone using this driver with an Ultrascale+ on a more recent version of Petalinux?

pmccumber, 2019-04-10:
The library is getting executed.  

We scoped the axidma logic in the pl and transactions occur when the driver is installed.  When we try to use the entry point, axidma_oneway_transfer() the pl sees no transactions and I get the "Inappropriate ioctl for device".

I'm using an Ultrascale+.  I'm compiling with:
make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 library

Doesn't seem to be hitting the driver correctly.

pmccumber, 2019-04-10:
Solution found.  I had updated the driver tree in Petalinux with appropriate source, including axidma_ioctl.h and failed to update the examples/library tree.   Yeesh.