bperez77/xilinx_axidma #92: petalinux build error

oussayran, 2019-04-02:
Hello,

i created a block design on vivado 2018.3 of Video test pattern generator (v8.0), frame buffer and zynq U+. 

i am trying to build petalinux using linux-xlnx tags ( xlnx_rebase_v4.14) and its last modification was in november 2018, which means the driver of Video test pattern generator (v8.0) was not modified inside the previous driver for v7.0. it just includes the driver for 5.0 and 7.0 . So it is obvious it would not be probed and that what i am got.

recently, a modifiation of the driver 8.0 was added in the branch master and not in tags. also i tried to build petalinux using the master branch but when i wrote petalinux-build , i got this error :

Ready $ petalinux-build -c kernel
[INFO] building kernel
[INFO] sourcing bitbake
INFO: bitbake virtual/kernel
WARNING: Host distribution "rhel-7.6" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |####################################################################################################################################################################| Time: 0:00:00
Loaded 3459 entries from dependency cache.
Parsing recipes: 100% |##################################################################################################################################################################| Time: 0:00:04
Parsing of 2569 .bb files complete (2533 cached, 36 parsed). 3461 targets, 137 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |###############################################################################################################################################################| Time: 0:00:20
Checking sstate mirror object availability: 100% |#######################################################################################################################################| Time: 0:00:26
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-xlnx-4.14-xilinx-v2018.3+git999-r0 do_populate_lic: QA Issue: linux-xlnx: The LIC_FILES_CHKSUM does not match for file://COPYING;md5=d7810fab7487fb0aad327b76f1be7cd7
linux-xlnx: The new md5 checksum is bbea815ee2795b2f4230826c0c6b8814
linux-xlnx: Here is the selected license text:
vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
The Linux Kernel is provided under:

SPDX-License-Identifier: GPL-2.0 WITH Linux-syscall-note

Being under the terms of the GNU General Public License version 2 only,
according with:

LICENSES/preferred/GPL-2.0

With an explicit syscall exception, as stated at:

LICENSES/exceptions/Linux-syscall-note

In addition, other licenses may also apply. Please see:

Documentation/process/license-rules.rst

for more details.
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
linux-xlnx: Check if the license information has changed in /local2/is221877/user_face/dev/xilinx/mohamad/se120/linux/master/linux-xlnx/COPYING to verify that the LICENSE value "GPLv2" remains valid [license-checksum]
ERROR: linux-xlnx-4.14-xilinx-v2018.3+git999-r0 do_populate_lic: Fatal QA errors found, failing task.
ERROR: linux-xlnx-4.14-xilinx-v2018.3+git999-r0 do_populate_lic: Function failed: populate_lic_qa_checksum
ERROR: Logfile of failure stored in: /local2/is221877/user_face/dev/xilinx/mohamad/se120/linux/master/tpg_frbw_plnx/build/tmp/work/plnx_zynqmp-xilinx-linux/linux-xlnx/4.14-xilinx-v2018.3+git999-r0/temp/log.do_populate_lic.21776
ERROR: Task (/opt/pkg/petalinux/2018.3/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2018.3.bb:do_populate_lic) failed with exit code '1'
NOTE: Tasks Summary: Attempted 133 tasks of which 113 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
/opt/pkg/petalinux/2018.3/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2018.3.bb:do_populate_lic
Summary: There was 1 WARNING message shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build kernel

 

it seems that it is a md5 checksum problem.

 

ANY SUGGESTIONS?

Regards,

Mohamad

honorpeter, 2019-06-16:
@oussayran How to do  when you  use petalinux with the driver files?

oussayran, 2019-06-16:
This issue was solved in 2019.1

closed 2019-06-16 by oussayran

honorpeter, 2019-06-16:
@oussayran  but I  only have 2018.2,and all my projects's version is 2018.2. if you have a little spare time,could you help me.

oussayran, 2019-06-16:
I didn't understand, what's your problem exactly!?

honorpeter, 2019-06-16:
@oussayran  I  have find out the solved  issue   https://github.com/bperez77/xilinx_axidma/issues/24 . 
  I will try  it with  linux-xlnx-4.14-xilinx-v2018.2   ,I  can't  change the  version.

kimanha, 2019-06-20:
@honorpeter Can you show me your Makefile and  xilinx-axidma.bb ?

I fallowed the same way but found build error.
I am using the petalinux 2019.1

Step 1:
bpmlab2 ~/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux $ petalinux-create -t modules -n xilinx-axidma --enable
INFO: Create modules: xilinx-axidma
INFO: New modules successfully created in /home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/project-spec/meta-user/recipes-modules/xilinx-axidma
INFO: Enabling created component...
INFO: sourcing bitbake
INFO: oldconfig rootfs
INFO: xilinx-axidma has been enabled

Step 2: Copy source code to petalinux module location
es/xilinx-axidma/files $ ls -l
total 132
-rw-rw-r-- 1 bpmlab2 bpmlab2  4001 Jun 19 21:49 axi_dma.c
-rw-rw-r-- 1 bpmlab2 bpmlab2 22907 Jun 19 21:49 axidma_chrdev.c
-rw-rw-r-- 1 bpmlab2 bpmlab2 25247 Jun 19 21:49 axidma_dma.c
-rw-rw-r-- 1 bpmlab2 bpmlab2  5884 Jun 19 21:49 axidma.h
-rw-rw-r-- 1 bpmlab2 bpmlab2 16773 Jun 19 21:49 axidma_ioctl.h
-rw-rw-r-- 1 bpmlab2 bpmlab2  9297 Jun 19 21:49 axidma_of.c
-rw-rw-r-- 1 bpmlab2 bpmlab2 18006 Jun 19 23:10 COPYING
-rw-rw-r-- 1 bpmlab2 bpmlab2   127 Jun 19 23:14 Makefile
-rw-rw-r-- 1 bpmlab2 bpmlab2  4941 Jun 19 23:10 xilinx-axidma.c

es/xilinx-axidma/files $ cat Makefile
DRIVER_NAME = xilinx-axidma
$(DRIVER_NAME)-objs = axi_dma.o axidma_chrdev.o axidma_dma.o axidma_of.o
obj-m := $(DRIVER_NAME).o

Step 3: build
bpmlab2 ~/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux $ petalinux-build -c rootfs/xilinx-axidma
[INFO] building rootfs/xilinx-axidma
[INFO] sourcing bitbake
[INFO] generating user layers
INFO: bitbake rootfs/xilinx-axidma
WARNING: Host distribution "linuxmint-18.3" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################################| Time: 0:00:01
Loaded 3811 entries from dependency cache.
Parsing recipes: 100% |##############################################################################| Time: 0:00:03
Parsing of 2778 .bb files complete (2774 cached, 4 parsed). 3813 targets, 163 skipped, 0 masked, 0 errors.
ERROR: Nothing PROVIDES 'rootfs/xilinx-axidma'. Close matches:
  xilinx-axidma

Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build rootfs/xilinx-axidma

kimanha, 2019-06-20:
build errors:

bpmlab2 ~/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux $ petalinux-build
[INFO] building project
[INFO] sourcing bitbake
[INFO] generating user layers
INFO: bitbake petalinux-user-image
WARNING: Host distribution "linuxmint-18.3" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################################| Time: 0:00:01
Loaded 3812 entries from dependency cache.
Parsing recipes: 100% |##############################################################################| Time: 0:00:02
Parsing of 2778 .bb files complete (2777 cached, 1 parsed). 3813 targets, 163 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |###########################################################################| Time: 0:00:04
Checking sstate mirror object availability: 100% |###################################################| Time: 0:00:04
Sstate summary: Wanted 134 Found 20 Missed 228 Current 753 (14% match, 87% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: xilinx-axidma-1.0-r0 do_compile: oe_runmake failed
ERROR: xilinx-axidma-1.0-r0 do_compile: Function failed: do_compile (log file is located at /home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/log.do_compile.13370)
ERROR: Logfile of failure stored in: /home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/log.do_compile.13370
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 KERNEL_SRC=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work-shared/plnx-zynq7/kernel-source KERNEL_PATH=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work-shared/plnx-zynq7/kernel-source KERNEL_VERSION=4.19.0-xilinx-v2019.1 CC=arm-xilinx-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fdebug-prefix-map=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0=/usr/src/debug/xilinx-axidma/1.0-r0 -fdebug-prefix-map=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/recipe-sysroot= -fdebug-prefix-map=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work-shared/plnx-zynq7/kernel-source=/usr/src/kernel LD=arm-xilinx-linux-gnueabi-ld.bfd   AR=arm-xilinx-linux-gnueabi-ar  O=/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work-shared/plnx-zynq7/kernel-build-artifacts KBUILD_EXTRA_SYMBOLS=
| make: *** No targets.  Stop.
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/xilinx-axidma/1.0-r0/temp/log.do_compile.13370)
ERROR: Task (/home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/project-spec/meta-user/recipes-modules/xilinx-axidma/xilinx-axidma.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3161 tasks of which 2928 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/bpmlab2/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux/project-spec/meta-user/recipes-modules/xilinx-axidma/xilinx-axidma.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project

linbaiwpi, 2019-09-08:
@kimanha I am right now working on the dma driver targeting to the same board. I notice that you did not delete the xilinx-axidma.c file auto generated after Step 2.

loylosky, 2020-11-03:
> build errors:
>
> bpmlab2 ~/xilinx/kiman/zc706/axi4_stream_vivado_prj.sdk/zc06_axi4_stream_plinux $ petalinux-build
> [INFO] building project
> [INFO] sourcing bitbake
> [INFO] generating user layers
> INFO: [...]

I got the same error as yours. May I ask have you solved this problem now? Thanks a lot!!!

wangzhenhua92429, 2023-05-15:
你发给我的邮件已收到！                                ——王振华