bperez77/xilinx_axidma #93: Axidma_benchmark: Assertion `find_channel(dev, tx_channel)->dir == AXIDMA_WRITE' failed.

wouterjoris, 2019-04-02:
Hi all ,

I've been testing with the axidma module and everything seems OK until I try to test the setup with the example application, then I get this output result: 

```
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 Mb
        Receive Buffer Size: 7.91 Mb
        Number of DMA Transfers: 1000 transfers

Using transmit channel 1 and receive channel 0.
axidma_benchmark.elf: library/libaxidma.c:481: axidma_twoway_transfer: Assertion find_channel(dev, tx_channel)->dir == AXIDMA_WRITE' failed.
```
Does someone have an idea what could be wrong here?

pmccumber, 2019-04-08:
Which part and version of Petalinux are you using?

wouterjoris, 2019-04-09:
HI @pmccumber 

Thanks for your help. I'm not using Petalinux but I build the OSL manually with checkout tag xilinx_v2018.2

wouterjoris, 2019-04-16:
Update for future reference: I noticed the axidma_twoway_transfer had different arguments defined in the C file and header file I missed this at first because the lib was pre-compiled before I ran the benchmark application.  

I must have checkout different tags or branch somehow. I only don't know how this happened. 

Nevertheless, I got past this error by correcting the source files.

closed 2019-04-16 by wouterjoris