bperez77/xilinx_axidma #94: 2018.2 Petalinux AXIDMA Config options

pmccumber, 2019-04-03:
The notes suggest enabling these:

CONFIG_XILINX_AXIDMA=y
CONFIG_XILINX_AXIVDMA=y

I did not see these in the config file.   What do these options do in the kernel?  I intend to simply add them.

pmccumber, 2019-04-04:
If these options are gone in 2018.2, how does that affect this driver?

xmentos, 2019-04-05:
You could use `petalinux-config -c kernel` to activate these

pmccumber, 2019-04-08:
Yes.  That's what I did.  Those options still are not present in the version we are using (2018.2).   There are analogous ones but was curious how it might affect the driver.

tpeshkin, 2019-04-08:
Seems these config options are obsolete, to allow DMA transfers enable following option:

> menuconfig XILINX_DMA_ENGINES
>         bool "Xilinx DMA Engines"
>         help
>           Enable support for the Xilinx DMA controllers. It supports three DMA
>           engines: Axi Central DMA (memory to memory transfer), Axi DMA (memory and
>           device transfer), and Axi VDMA (memory and video device transfer).
>

closed 2019-04-09 by pmccumber

honorpeter, 2019-06-18:
> menuconfig XILINX_DMA_ENGINES
@tpeshkin But how to do and in where place ?

tpeshkin, 2019-06-18:
petalinux-config -c kernel
Device Drivers ---> 
    [*] DMA Engine support --->
        [*] Xilinx DMA Engines