bperez77/xilinx_axidma #97: problem with "dma-coherent" in the device tree

sk2046, 2019-05-02:
Hello,
I found that read access to the Rx memory allocated by the driver is slow too.I have already referred to the issue:[https://github.com/bperez77/xilinx_axidma/issues/69](url)
In my app,i dis like this:
`uint8_t *dma_rx_buf;
dma_rx_buf = axidma_malloc(axidma_dev, 32768);
sendto(Socket_UDP,dma_rx_buf,16388,0,(struct sockaddr*)&UDP_addr_dst,sizeof(UDP_addr_dst));`
When I use function sendto(), I read dma_rx_buf very slowly.I modify the DT like this:
`/include/ "system-conf.dtsi"
/ {
      	axidma_chrdev: axidma_chrdev@0 {
	compatible = "xlnx,axidma-chrdev";
	dmas = <&axi_dma_0 0 &axi_dma_0 1>;
	dma-names = "tx_channel", "rx_channel";
        };
        amba_pl: amba_pl {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "simple-bus";
		ranges ;
		axi_dma_0: dma@40400000 {
			#dma-cells = <1>;
			clock-names = "s_axi_lite_aclk", "m_axi_sg_aclk", "m_axi_mm2s_aclk", "m_axi_s2mm_aclk";
			clocks = <&clkc 15>, <&clkc 15>, <&clkc 15>, <&clkc 15>;
			compatible = "xlnx,axi-dma-1.00.a";
			interrupt-parent = <&intc>;
			interrupts = <0 29 4 0 30 4>;
			reg = <0x40400000 0x10000>;
			xlnx,addrwidth = <0x20>;
			dma-coherent;
			dma-channel@40400000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 29 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x0>;
			};
			dma-channel@40400030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 30 4>;
				xlnx,datawidth = <0x20>;
				xlnx,device-id = <0x1>;
			};
		};
	};
};`
Then when I use the function axidma_oneway_transfer() to receive data,a mistake has occurred:
`xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051500 tdr 3f051500
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051580 tdr 3f051580
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051600 tdr 3f051600
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051680 tdr 3f051680
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051700 tdr 3f051700
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051780 tdr 3f051780
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051800 tdr 3f051800
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051880 tdr 3f051880
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051900 tdr 3f051900
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051980 tdr 3f051980
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051a00 tdr 3f051a00
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051a80 tdr 3f051a80
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051b00 tdr 3f051b00
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051b80 tdr 3f051b80
xilinx-vdma 40400000.dma: Channel ef2edc10 has errors 10, cdr 3f051c00 tdr 3f051c00`
If i do not use "dma-coherent;" in DT, there is no erro. How can i improve the speed that access to the Rx memory dma_rx_buf？
When i test like this:
`char udp_tx_buf[65536];
udp_txSize = sendto(Socket_UDP,udp_tx_buf,16388,0,(struct sockaddr*)&UDP_addr_dst,sizeof(UDP_addr_dst));`
The read speed is good.Is there any way to solve this problem?

eleICoto, 2019-06-06:
Hi could you please provide more detail? and format your code...

stone-sjj, 2021-03-22:
HELLO，how do you solve the problem?Thank you!

Yjxslhz, 2024-10-23:
hi,how do you solve the problem?Thank you!