bperez77/xilinx_axidma #98: Problem with two dma in design

eleICoto, 2019-05-09:
I'm using two dma in my design
my system-user.dtsi in petalinux is like this 
[image]
my pl.dtsi in petalinux is like this 
[image]
the problem i'm facing is like this 
[image]
it seems like module cant find slave channel
[image]
my bd is like this
[image]
besides i'm using ZCU104

GOOD-Stuff, 2019-05-31:
Hi!
What if you try to set in pl.dtsi something like this:
```
amba_pl: amba_pl@0 {
                ....
                axidma_chrdev: axidma_chrdev@0 {
                         compatible = "xlnx,axidma-chrdev";
                         dmas = <&ps2pl 0 &pl2ps 1>; // note to the 1 after pl2ps, it is the xlnx,device-id 
                         dma-names = "tx_channel", "rx_channel";
                 };
		ps2pl: dma@a0000000 {
			#dma-cells = <1>;
			clock-names = ... // your values
                         ......
			dma-channel@a0000000 {
				compatible = "xlnx,axi-dma-mm2s-channel";
				dma-channels = <0x1>;
				interrupts = <0 34 4>; // set here your values
                                 ...
				xlnx,device-id = <0x0>;
				xlnx,include-dre ;
			};
};
               pl2ps: dma@a0001000 {
			#dma-cells = <1>;
			clock-names = ... // your values
                         ......
			dma-channel@a0001030 {
				compatible = "xlnx,axi-dma-s2mm-channel";
				dma-channels = <0x1>;
				interrupts = <0 33 4>;
                                 ...
				xlnx,device-id = <0x1>;
				xlnx,include-dre ;
			};
		};
};
```

eleICoto, 2019-06-05:
Thank you for your replay, I actually fix this, the DMA address width in zynq ultrascale family chip must be set to 40bit, or it will go wrong.

closed 2019-06-05 by eleICoto

suikammd, 2019-06-10:
Hey, where to modify the  DMA address width?
Could you share your new dtsi file?

eleICoto, 2019-06-10:
@suikammd in vivado block design
[image]

suikammd, 2019-06-10:
@eleICoto Thanks, but after insmod the driver, nothing happens. 
Is your device tree you attached above works?

eleICoto, 2019-06-10:
@ Can you upload your block design? And specific DMA setting?
I have tested one DMA with two channel & two DMA with one channel
All these setting works fine
Right now I'm trying to speed up buffer access time

suikammd, 2019-06-11:
@eleICoto Hello, this is my block design. 
[image]
But just to insmod the driver should have nothing to do with block design?
P.S. Do you use petalinux? or Can you share me with your process of using this driver?
Thanks a lot!

eleICoto, 2019-06-12:
@suikammd  Yes i'm using petalinux 
I think there is something to do with block design, just like i posted before

eleICoto, 2019-06-12:
@suikammd And I'm wondering what board are you using?
And, if possible, can you list your workflow when generate.bit and building petalinux

suikammd, 2019-06-15:
@eleICoto Sorry for later reply.
1. Generate Bitstream to get .bit and Export Hardware to package .bit into HDF
2. petalinux-config --get-hw-description ./
3. Image Packaging Configurations -> Root filesystem type -> SD card 
    Uncheck DTG Settings->Kernel Bootargs->generate boot args automatically, set bootrags to `earlycon clk_ignore_unused earlyprintk root=/dev/mmcblk0p2 rw rootwait
 `
4. petalinux-build
Maybe there is something wrong with my device tree, I'm using zcu104, same as you.
Thanks a lot!

eleICoto, 2019-06-15:
[image]
[image]

eleICoto, 2019-06-15:
@suikammd 
[image]
Open this file, check whether the DMA setting is enable

suikammd, 2019-06-15:
Thanks @eleICoto ,I checked this file, 
`CONFIG_MTD=y
CONFIG_MTD_OF_PARTS=y
CONFIG_SERIAL_OF_PLATFORM=y
CONFIG_NET=y
CONFIG_PACKET=y
CONFIG_UNIX=y
CONFIG_INET=y
CONFIG_GPIOLIB=y
CONFIG_GPIO_XILINX=y
CONFIG_GPIO_ZYNQ=y
CONFIG_I2C=y
CONFIG_I2C_CADENCE=y
CONFIG_MMC=y
CONFIG_MMC_BLOCK=y
CONFIG_MMC_SDHCI=y
CONFIG_MMC_SDHCI_OF_XILINX_PS=y
CONFIG_MMC_SDHCI_PLTFM=y
CONFIG_MTD=y
CONFIG_MTD_M25P80=y
CONFIG_MTD_OF_PARTS=y
CONFIG_NET_CORE=y
CONFIG_RTC_CLASS=y
CONFIG_SERIAL_OF_PLATFORM=y
CONFIG_SERIAL_XILINX_PS_UART=y
CONFIG_SERIAL_XILINX_PS_UART_CONSOLE=y
CONFIG_USB=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_EHCI_XUSBPS=y
CONFIG_USB_GADGET_XUSBPS=y
CONFIG_USB_OTG=y
CONFIG_USB_SUPPORT=y
CONFIG_USB_XUSBP=y
CONFIG_USB_XUSBPS_DR_OF=y
CONFIG_USB_XUSBPS_OTG=y
CONFIG_WATCHDOG=y
CONFIG_XILINX_PS_WATCHDOG=y
`
It's not enabled, should I just add `CONFIG_DMA=y`?

eleICoto, 2019-06-15:
@suikammd So did you enter peltalinux-config -c kernel and enable those dma setting mention in this page?
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842337/DMA+Drivers+-+Soft+IPs

eleICoto, 2019-06-15:
@suikammd according to my experience, save tage in petalinux-config -c kernel cant automatic set .cfg path you have to enter it by hand

eleICoto, 2019-06-15:
@suikammd And, by the way, according to your description, I guess your ZCU104 doesn't have network access?

suikammd, 2019-06-15:
@eleICoto yes, but network access a necessity?
I'll try to config the kernel and enable dma in the cfg file.
Thanks very much

suikammd, 2019-06-15:
@eleICoto 
I add those lines in cfg file.
<pre>
CONFIG_CMA=y
CONFIG_DMA_CMA=y
CONFIG_XILINX_DMAENGINES=y
CONFIG_XILINX_AXIDMA=y
CONFIG_XILINX_AXIVDMA=y
CONFIG_DMA_SHARED_BUFFER=y
</pre>

eleICoto, 2019-06-15:
@suikammd That depends on you, but for sure there is something lack in your device tree setting, follow this page https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842279/Build+Device+Tree+Blob
Generate a Board file Device Tree Source (.dts/.dtsi) files on command line using HSM/HSI. 
And I'm not sure if there will be problem when you just add setting in .cfg file, I would recommand using petalinux-config -c kernel

suikammd, 2019-06-15:
@eleICoto I do enable dma using petalinux-config -c kernel, but it does not show up in the cfg file you metioned.

eleICoto, 2019-06-15:
@suikammd Yes, after you finfish enable process, you need to save you config, in the bottom of menuconfig, there is a save tag, enter it and modify .cfg path

suikammd, 2019-06-17:
@eleICoto yes, I config the kernel, saved the settings but those settings still not show up in cfg file, so I just add those `CONFIG_*` I mentioned above.
When I tried to insmod the driver, the dmesg changes nothing and no new device appeared. But `modprobe -l` shows I successfully insmod the driver while the driver actually not exists.
It must be something wrong with the device tree. If you are not busy, could you share me how you test DMA with two channels?
Thanks!

eleICoto, 2019-06-17:
@suikammd Alright 
[plnx_kernel.zip](https://github.com/bperez77/xilinx_axidma/files/3295807/plnx_kernel.zip)
this is the config file i'm using 
And could you up load your dmesg? just run dmesg > log.log and upload log.log
And your pl.dtsi system-user.dtsi

suikammd, 2019-06-17:
@eleICoto Thanks, three files all uploaded.
[peta.zip](https://github.com/bperez77/xilinx_axidma/files/3295990/peta.zip)

eleICoto, 2019-06-17:
@suikammd It seems like you have change your bd to two dma design, but you have not change your dma
address width to 40

eleICoto, 2019-06-17:
@suikammd And there is something wrong with your axidma_chrdev 
[image]
if you use two dma with only one channel both value should be 0
[image]
And it seems like you set the dma-names in opposite if your design is to send data from ps ddr to pl ddr

eleICoto, 2019-06-17:
@suikammd log.log is the log file after you insmod?

suikammd, 2019-06-17:
@eleICoto The dma address width is set to 40. Block design was done by my lab mate. In fact, I'm major in networks ==.
> @suikammd log.log is the log file after you insmod?

Yes, so it must be something wrong with device tree files. I'm very confused about how to modify them.

eleICoto, 2019-06-17:
@suikammd ahhhh 40 in decimal...... not hex

eleICoto, 2019-06-17:
[image]
[image]

suikammd, 2019-06-17:
@eleICoto My lab mate says he set 40 in decimal...

eleICoto, 2019-06-17:
@suikammd according to device tree file you  send to me 
[image]
[image]
both addrwidth is set to 40 hex 64 dec

honorpeter, 2019-06-17:
@suikammd Do you succeed? I follow the eleICoto said,but  I got this log when I insmod  and  run dmesg

 root@xdmatest:~# dmesg

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.0-xilinx-v2018.2 (oe-user@oe-host) (gcc version 9
[    0.000000] Boot CPU: AArch64 Processor [410fd034]
[    0.000000] Machine model: ZynqMP ZCU102 Rev1.0
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')
[    0.000000] bootconsole [cdns0] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x000000006fc00000
[    0.000000] On node 0 totalpages: 1048320
[    0.000000]   DMA zone: 7165 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524032 pages, LIFO batch:31
[    0.000000]   Normal zone: 7168 pages used for memmap
[    0.000000]   Normal zone: 524288 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] percpu: Embedded 21 pages/cpu @ffffffc87ff63000 s46488 r8192 d313366
[    0.000000] pcpu-alloc: s46488 r8192 d31336 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0                                       
[    0.000000] CPU features: enabling workaround for ARM erratum 845719            
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1033987      
[    0.000000] Kernel command line: earlycon clk_ignore_unused root=/dev/mmcblk0p2t
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)                
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)  
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)    
[    0.000000] software IO TLB [mem 0x6bc00000-0x6fc00000] (64MB) mapped at [fffff]
[    0.000000] Memory: 3784592K/4193280K available (9916K kernel code, 644K rwdata)
[    0.000000] Virtual kernel memory layout:                                       
[    0.000000]     modules : 0xffffff8000000000 - 0xffffff8008000000   (   128 MB) 
[    0.000000]     vmalloc : 0xffffff8008000000 - 0xffffffbebfff0000   (   250 GB) 
[    0.000000]       .text : 0xffffff8008080000 - 0xffffff8008a30000   (  9920 KB) 
[    0.000000]     .rodata : 0xffffff8008a30000 - 0xffffff8008d40000   (  3136 KB) 
[    0.000000]       .init : 0xffffff8008d40000 - 0xffffff8008dc0000   (   512 KB) 
[    0.000000]       .data : 0xffffff8008dc0000 - 0xffffff8008e61200   (   645 KB) 
[    0.000000]        .bss : 0xffffff8008e61200 - 0xffffff800907f2b0   (  2169 KB) 
[    0.000000]     fixed   : 0xffffffbefe7fd000 - 0xffffffbefec00000   (  4108 KB) 
[    0.000000]     PCI I/O : 0xffffffbefee00000 - 0xffffffbeffe00000   (    16 MB) 
[    0.000000]     vmemmap : 0xffffffbf00000000 - 0xffffffc000000000   (     4 GB )
[    0.000000]               0xffffffbf00000000 - 0xffffffbf1dc00000   (   476 MB )
[    0.000000]     memory  : 0xffffffc000000000 - 0xffffffc880000000   ( 34816 MB) 
[    0.000000] Hierarchical RCU implementation.                                    
[    0.000000]  RCU event tracing is enabled.                                      
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.               
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4        
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0                      
[    0.000000] GIC: Adjusting CPU interface base to 0x00000000f902f000             
[    0.000000] GIC: Using split EOI/Deactivate mode                                
[    0.000000] arch_timer: cp15 timer(s) running at 99.99MHz (phys).               
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0s
[    0.000003] sched_clock: 56 bits at 99MHz, resolution 10ns, wraps every 4398046s
[    0.008314] Console: colour dummy device 80x25                                  
[    0.012568] console [tty0] enabled                                              
[    0.015934] bootconsole [cdns0] disabled                                        
[    0.019845] Calibrating delay loop (skipped), value calculated using timer freq)
[    0.019859] pid_max: default: 32768 minimum: 301                                
[    0.019962] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes)        
[    0.019980] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes)   
[    0.020559] ASID allocator initialised with 65536 entries                       
[    0.020606] Hierarchical SRCU implementation.                                   
[    0.020936] EFI services will not be available.                                 
[    0.020961] zynqmp_plat_init Platform Management API v1.0                       
[    0.020969] zynqmp_plat_init Trustzone version v1.0                             
[... 338 lines cut ...]
[    4.895113] pca954x 1-0075: registered 8 multiplexed busses for I2C switch pca98
[    4.902686] cdns-i2c ff030000.i2c: 400 kHz mmio ff030000 irq 34                 
[    4.909760] input: gpio-keys as /devices/platform/gpio-keys/input/input0        
[    4.916570] rtc_zynqmp ffa60000.rtc: setting system clock to 2019-06-17 05:27:4)
[... cut ...]
[    6.864404] pps pps0: new PPS source ptp0                                       
[    6.868363] macb ff0e0000.ethernet: gem-ptp-timer ptp clock registered.         
[    6.874996] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready                  
[   54.067511] xilinx_axidma: loading out-of-tree module taints kernel.            
[   54.075778] axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 1 transmit c.
[   54.085733] axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit .

eleICoto, 2019-06-17:
@honorpeter it seems like you have succeeded someway but i have no idea how this kind message showed up
[image]

suikammd, 2019-06-17:
@honorpeter I couldn't even insmod the driver...Could you tell me what you do before you insmod the driver.

eleICoto, 2019-06-17:
@suikammd if you can, ask your lab mate to provide you the bitstream in sdk project.

honorpeter, 2019-06-17:
@eleICoto Than you . but [ 54.067511] xilinx_axidma: **loading out-of-tree** module taints kernel.   why?

honorpeter, 2019-06-17:
> @honorpeter I couldn't even insmod the driver...Could you tell me what you do before you insmod the driver.
@eleICoto    You can do like https://github.com/bperez77/xilinx_axidma/issues/24#issuecomment-335252145.

honorpeter, 2019-06-17:
@bperez77  when I follow you step and insmod the driver on zcu102 , I find an log  and Benchmark is ok,may be  as warning. but  is  it a  matter?

Starting udev
[    5.396416] udevd[1879]: starting version 3.2.2
[    5.438209] udevd[1880]: starting eudev-3.2.2
[    5.480848] [drm] Cannot find any crtc or size

root@xdmatest:~# insmod /lib/modules/4.14.0-xilinx-v2018.2/extra/xilinx-axidma.ko 
[   45.791759] xilinx_axidma: loading out-of-tree module taints kernel.
[   45.799444] axidma: axidma_dma.c: axidma_dma_init: 718: DMA: Found 1 transmit channels and 1 receive channels.
[   45.809396] axidma: axidma_dma.c: axidma_dma_init: 720: VDMA: Found 0 transmit channels and 0 receive channels.
./axidma-benchmark
AXI DMA Benchmark Parameters:
        Transmit Buffer Size: 7.91 MiB
        Receive Buffer Size: 7.91 MiB
        Number of DMA Transfers: 1000 transfers

Using transmit channel 0 and receive channel 1.
Single transfer test successfully completed!
Beginning performance analysis of the DMA engine.

DMA Timing Statistics:
        Elapsed Time: 20.97 s
        Transmit Throughput: 377.20 MiB/s
        Receive Throughput: 377.20 MiB/s
        Total Throughput: 754.39 MiB/s
root@xdmatest:/usr/bin# 

But test is ok,  how can I do to modify the  petalinux config to avoid the issue.xilinx_axidma: loading out-of-tree module taints kernel.

eleICoto, 2019-06-17:
@honorpeter where is your .ko file located at? Did this step familiar with you? petalinux-config rootfs 
module -> enable dma module

eleICoto, 2019-06-17:
@honorpeter 
[image]
[image]

suikammd, 2019-06-17:
@honorpeter I just followed what the author said. 
Before I exec `petalinux-build`, I build the module `xilinx_axidma` successfully. 
But it seems that there must be something wrong with the device tree...

@eleICoto He already gave me the bitstream file.

honorpeter, 2019-06-17:
@suikammd  He may give you hdf. I think you should create a new project by petalinux and followed what the author said you will succeed.
    You should build the petalinux  with the hdf  and then change the  system-user.dtsi  according the pl.dtsi.

honorpeter, 2019-06-17:
> @honorpeter
> [image]
> [image]

Only do this can't avoid the  "loading out-of-tree module taints kernel" warning. I just did .

eleICoto, 2019-06-17:
@honorpeter Would you please upload your kernel config file as I mentioned before, I think there might be some config about module sig that has not been enable.

honorpeter, 2019-06-18:
@eleICoto  As https://github.com/bperez77/xilinx_axidma/issues/94#issuecomment-480879355
   but how to do and to set  in  where place .

honorpeter, 2019-06-18:
In petalinux 2018.2 petalinux-config -c kernel  → Device Drivers → DMA Engine support  ->hlep

CONFIG_DMADEVICES:     
                                                                                                                        │  
  │                                                                                                                                                │  
  │ DMA engines can do asynchronous data transfers without                                                                                         │  
  │ involving the host CPU.  Currently, this framework can be                                                                                      │  
  │ used to offload memory copies in the network stack and                                                                                         │  
  │ RAID operations in the MD driver.  This menu only presents                                                                                     │  
  │ DMA Device drivers supported by the configured arch, it may                                                                                    │  
  │ be empty in some cases.                                                                                                                        │  
  │                                                                                                                                                │  
  │ Symbol: DMADEVICES [=y]                                                                                                                        │  
  │ Type  : boolean                                                                                                                                │  
  │ Prompt: DMA Engine support                                                                                                                     │  
  │   Location:                                                                                                                                    │  
  │     -> Device Drivers                                                                                                                          │  
  │   Defined at drivers/dma/Kconfig:5                                                                                                             │  
  │   Depends on: HAS_DMA [=y]                                                                                                                     │  
  │   Selected by: SND_SOC_SH4_SIU [=n] && SOUND [=y] && !UML && SND [=y] && SND_SOC [=y] && (SUPERH || ARCH_SHMOBILE [=n] || COMPILE_TEST [=n])\  │  
  │  && (SUPERH || ARCH_SHMOBILE [=n]) && HAVE_CLK [=y] || CRYPTO_DEV_SP_CCP [=n] && CRYPTO [=y] && CRYPTO_HW [=n] && CRYPTO_DEV_CCP [=n] && \     │  
  │ CRYPTO_DEV_CCP_DD [=n]                                                                                                                         │  
  │

honorpeter, 2019-06-18:
@eleICoto 
Arrow keys navigate the menu.  <Enter> selects submenus ---> (or empty submenus ----).  Highlighted letters are hotkeys.  Pressing <Y>        │  
  │  includes, <N> excludes, <M> modularizes features.  Press <Esc><Esc> to exit, <?> for Help, </> for Search.  Legend: [*] built-in  [ ]         │  
  │  excluded  <M> module  < > module capable                                                                                                      │  
  │                                                                                                                                                │  
  │ ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐ │  
  │ │                                   --- Xilinx DMA Engines                                                                                   │ │  
  │ │                                   < >   DMA Test client for AXI DMA                                                                        │ │  
  │ │                                   < >   DMA Test client for VDMA                                                                           │ │  
  │ │                                   < >   DMA Test client for CDMA                                                                           │ │  
  │ │                                   -*-   Xilinx DPDMA Engine                                                                                │ │  
  │ │                                   [ ]     Xilinx DPDMA debugfs                                                                             │ │  
  │ │                                   -*-   Xilinx Framebuffer                                                                                 │ │  
  │ │                                                                                                                                            │ │  
  │ │                                                                                                                                            │ │  
  │ │                                                                                                                                            │ │  
  │ │                                                                                                                                            │ │  
  │ │                                                                                                                                            │ │  
  │ │                                                                                                                 
IS   DMA Test client for AXI DMA  and  DMA Test client for VDMA   should  be selected?

suikammd, 2019-06-18:
> @suikammd He may give you hdf. I think you should create a new project by petalinux and followed what the author said you will succeed.
> You should build the petalinux with the hdf and then change the system-user.dtsi according the pl.dtsi.

@honorpeter you mean follow this [issue 24](https://github.com/bperez77/xilinx_axidma/issues/24)? 
Could you share me how you build petalinux?
What I do is the following:
1.  build xilinx_axidma
2. modify device tree as the zip file I uploaded above
3. petalinux-build
4. sd card boot -> insert module nothing happens with no error show up

eleICoto, 2019-06-18:
@honorpeter Yes just upload this file ./project-spec/meta-plnx-generated/recipes-kernel/linux/configs/plnx_kernel.cfg

honorpeter, 2019-06-18:
@eleICoto 
In
./build/tmp/work/plnx_zynqmp-xilinx-linux/linux-xlnx/4.14-xilinx-v2018.2+gitAUTOINC+ad4cd988ba-r0/xilinx-kmeta/bsp/xilinx/soc/drivers-drm.cfg  I find
# CMA
CONFIG_CMA=y
CONFIG_DMA_CMA=y
CONFIG_CMA_SIZE_MBYTES=128
CONFIG_CMA_SIZE_SEL_MBYTES=y
CONFIG_CMA_ALIGNMENT=8

In
./build/tmp/work/plnx_zynqmp-xilinx-linux/linux-xlnx/4.14-xilinx-v2018.2+gitAUTOINC+ad4cd988ba-r0/xilinx-kmeta/bsp/xilinx/soc/drivers-zynqmp.cfg
I find

# DMA
CONFIG_DMA_ENGINE=y
CONFIG_XILINX_ZYNQMP_DMA=y

May be it's all ok.But can't see any about DMA in ./project-spec/meta-plnx-generated/recipes-kernel/linux/configs/plnx_kernel.cfg.

honorpeter, 2019-06-18:
> > @suikammd He may give you hdf. I think you should create a new project by petalinux and followed what the author said you will succeed.
> > You should build the petalinux with the hdf and then change the system-user.dtsi change the system-user.dtsi according the pl.dtsi.

> @honorpeter you mean follow this [issue 24](https://github.com/bperez77/xilinx_axidma/issues/24)?
> Could you share me how you build petalinux?
> What I do is the following:
>
>     1. build [...]
1.first creat petalinux project
2. then add the hdf to you project
3.petalinux-config --get-hw-description=./
4.petalinux-build
5.create module 
6.create app
7.change the system-user.dtsi according the pl.dtsi.
8.petalinux-build

honorpeter, 2019-06-18:
@eleICoto 

This is  ./project-spec/meta-plnx-generated/recipes-kernel/linux/configs/plnx_kernel.cfg

CONFIG_MTD=y
CONFIG_MTD_OF_PARTS=y
CONFIG_SERIAL_OF_PLATFORM=y
CONFIG_NET=y
CONFIG_PACKET=y
CONFIG_UNIX=y
CONFIG_INET=y
CONFIG_GPIOLIB=y
CONFIG_GPIO_XILINX=y
CONFIG_GPIO_ZYNQ=y
CONFIG_I2C=y
CONFIG_I2C_CADENCE=y
CONFIG_MMC=y
CONFIG_MMC_BLOCK=y
CONFIG_MMC_SDHCI=y
CONFIG_MMC_SDHCI_OF_XILINX_PS=y
CONFIG_MMC_SDHCI_PLTFM=y
CONFIG_MTD=y
CONFIG_MTD_M25P80=y
CONFIG_MTD_OF_PARTS=y
CONFIG_NET_CORE=y
CONFIG_PCI=y
CONFIG_PCIE_XILINX_NWL=y
CONFIG_PCI_MSI=y
CONFIG_RTC_CLASS=y
CONFIG_SERIAL_OF_PLATFORM=y
CONFIG_SERIAL_XILINX_PS_UART=y
CONFIG_SERIAL_XILINX_PS_UART_CONSOLE=y
CONFIG_USB=y
CONFIG_USB_EHCI_HCD=y
CONFIG_USB_EHCI_XUSBPS=y
CONFIG_USB_GADGET_XUSBPS=y
CONFIG_USB_OTG=y
CONFIG_USB_SUPPORT=y
CONFIG_USB_XUSBP=y
CONFIG_USB_XUSBPS_DR_OF=y
CONFIG_USB_XUSBPS_OTG=y
CONFIG_WATCHDOG=y
CONFIG_XILINX_PS_WATCHDOG=y
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
~                                                                                                                                                      
"./project-spec/meta-plnx-generated/recipes-kernel/linux/configs/plnx_kernel.cfg" 40L, 805C                                          1,0-1        全部

can't  see any about  dma and axi dma ,only see https://github.com/bperez77/xilinx_axidma/issues/98#issuecomment-502925454

suikammd, 2019-06-19:
@honorpeter Thanks!
Did you test with the author's example successfully?

honorpeter, 2019-06-19:
@suikammd Yes I had test the  example successfully . You should copy all in the pl.dtsi   to  the  system-conf.dtsi  and change the  channel's  Id .like https://github.com/bperez77/xilinx_axidma/issues/98#issue-442109819

suikammd, 2019-06-19:
@honorpeter Hi, thanks for reply.
After follow your suggestion, I met this wired error
`| Error: /home/ubuntu/board/xilinx-zcu104-2018.2/build/../components/plnx_workspace/device-tree/device-tree/system-top.dts:30.123-31.1 syntax error
`
this is the system-top.dts file:
<pre>
/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version:
 * Today is: Wed Jun 19 14:16:49 2019
 */

/dts-v1/;
/include/ "zynqmp.dtsi"
/include/ "zynqmp-clk-ccf.dtsi"
/include/ "zcu104-revc.dtsi"
/include/ "pl.dtsi"
/include/ "pcw.dtsi"
/ {
        chosen {
                bootargs = "earlycon clk_ignore_unused";
                stdout-path = "serial0:115200n8";
        };
        aliases {
                ethernet0 = &gem3;
                i2c0 = &i2c1;
                serial0 = &uart0;
                serial1 = &uart1;
                spi0 = &qspi;
        };
        memory {
                device_type = "memory";
                reg = <0x0 0x0 0x0 0x7ff00000>;
        };
 };
#include "system-user.dtsi"
</pre>
Do you know how to solve it?

honorpeter, 2019-06-19:
@suikammd  You should learn  https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841738/Getting+Started
May be  #include "system-user.dtsi"？  You can delete all ,the petalinux will genarate auto

suikammd, 2019-06-19:
@honorpeter  Thanks, it works